intel/compute-runtime #230: Abort was called at 22 line in file: ../neo/runtime/gmm_helper/gmm_interface.cpp

jeffhammond, 2019-11-13:
I installed NEO per https://github.com/intel/compute-runtime/blob/master/documentation/Neo_in_distributions.md#ubuntu-1904-1910 on my Hades Canyon NUC running Linux 5.0.0-32-generic.

It fails:
```
jrhammon@jrhammon-nuc:~/PRK/Cxx11$ gdb clinfo
...
(gdb) run
Starting program: /usr/bin/clinfo 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7ffff69c2700 (LWP 16398)]
Abort was called at 22 line in file:
../neo/runtime/gmm_helper/gmm_interface.cpp

Thread 1 "clinfo" received signal SIGABRT, Aborted.
__GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:50
50	../sysdeps/unix/sysv/linux/raise.c: No such file or directory.
(gdb) bt
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:50
#1  0x00007ffff7bdb535 in __GI_abort () at abort.c:79
#2  0x00007ffff76d2529 in NEO::abortExecution() ()
   from /opt/intel/inteloneapi/compiler/1.0.0/lib/oclgpu/intel-opencl/libigdrcl.so
#3  0x00007ffff76d2601 in NEO::abortUnrecoverable(int, char const*) ()
   from /opt/intel/inteloneapi/compiler/1.0.0/lib/oclgpu/intel-opencl/libigdrcl.so
#4  0x00007ffff772f3cb in NEO::GmmHelper::loadLib() ()
   from /opt/intel/inteloneapi/compiler/1.0.0/lib/oclgpu/intel-opencl/libigdrcl.so
#5  0x00007ffff772ed26 in NEO::GmmHelper::initContext(PLATFORM_STR const*, NEO::FeatureTable const*, NEO::WorkaroundTable const*, GT_SYSTEM_INFO const*) ()
   from /opt/intel/inteloneapi/compiler/1.0.0/lib/oclgpu/intel-opencl/libigdrcl.so
#6  0x00007ffff772f031 in NEO::GmmHelper::GmmHelper(NEO::HardwareInfo const*) ()
   from /opt/intel/inteloneapi/compiler/1.0.0/lib/oclgpu/intel-opencl/libigdrcl.so
#7  0x00007ffff771b0b3 in NEO::ExecutionEnvironment::initGmm() ()
   from /opt/intel/inteloneapi/compiler/1.0.0/lib/oclgpu/intel-opencl/libigdrcl.so
#8  0x00007ffff76eaf8d in NEO::Device::createDeviceImpl() ()
   from /opt/intel/inteloneapi/compiler/1.0.0/lib/oclgpu/intel-opencl/libigdrcl.so
#9  0x00007ffff76d559a in NEO::Platform::initialize() ()
   from /opt/intel/inteloneapi/compiler/1.0.0/lib/oclgpu/intel-opencl/libigdrcl.so
#10 0x00007ffff76b3e2b in clGetPlatformIDs ()
   from /opt/intel/inteloneapi/compiler/1.0.0/lib/oclgpu/intel-opencl/libigdrcl.so
#11 0x00007ffff7dadb98 in ?? () from /usr/lib/x86_64-linux-gnu/libOpenCL.so.1
#12 0x00007ffff7dae0d3 in clGetPlatformIDs () from /usr/lib/x86_64-linux-gnu/libOpenCL.so.1
#13 0x000055555555b478 in ?? ()
#14 0x00007ffff7bdcb6b in __libc_start_main (main=0x55555555b2c0, argc=1, argv=0x7fffffffbd98, 
    init=<optimized out>, fini=<optimized out>, rtld_fini=<optimized out>, 
    stack_end=0x7fffffffbd88) at ../csu/libc-start.c:308
#15 0x000055555555b7ba in ?? ()
```

bfliflet, 2019-11-13:
Hi Jeff,
Hades Canyon NUC has AMD graphics and is not supported by the Neo driver.  I would have expected our driver to have aborted earlier and we should look at a cleaner error response.

-Brandon

jeffhammond, 2019-11-13:
Kaby Lake-G contains Gen 9 and that has been working just fine for me all year.  This is a recent problem.  I was running Intel NEO OpenCL with SYCL latest under Ubuntu 18 for at least six months.  I had to upgrade to Ubuntu 19 because support for 18 expired.

jeffhammond, 2019-11-13:
This not the same as https://github.com/intel/compute-runtime/issues/99 until the Ubuntu 19 packages (https://github.com/intel/compute-runtime/blob/master/documentation/Neo_in_distributions.md#ubuntu-1904-1910) are busted.

```
jrhammon@jrhammon-nuc:~/ISYCL$ sudo apt-get install intel-opencl-icd
Reading package lists... Done
Building dependency tree       
Reading state information... Done
intel-opencl-icd is already the newest version (19.43.14583-1~ppa1~disco1).
0 upgraded, 0 newly installed, 0 to remove and 10 not upgraded.
jrhammon@jrhammon-nuc:~/ISYCL$ clinfo
Abort was called at 22 line in file:
../neo/runtime/gmm_helper/gmm_interface.cpp
Aborted (core dumped)
```

jeffhammond, 2019-11-13:
Okay, it is the same as #99 but only because it looked in a completely stupid place for libigdgmm.so and found the .9 remnant.  The path to that library was nowhere in my environment, so I have no idea why it was found.  Please make NEO more robust and check for obsolete shared library versions.  Also, I don't know what search path logic is used that finds obsolete libraries in /opt/intel/inteloneapi/1.0.0 rather than /usr/lib/x86_64-linux-gnu but that is very bad for UX.

bfliflet, 2019-11-13:
My bad Jeff.  Didn't realize you were using the iGfx on KBL.  Glad you figured out a solution.  We'll look at the load path.  Last I recall, it wasn't anything special.

jeffhammond, 2019-11-13:
Yeah, it's insane that I spent the better part of two days debugging this.

pszi1ard, 2020-01-14:
I have this exact same issue issue and it does not seem to be caused by an old libigdgmm.so; I've checked all .so files that `LD_DEBUG=files clinfo` prints, but none of them seem to have stale versions around.

Any ideas where else should I be looking?

Should I open a new issue?

JacekDanecki, 2020-01-14:
@pszi1ard What Linux distribution are you using? What are the versions of Neo and gmmlib? Where did you get binaries for Neo and gmmlib from?
Can you provide logs from below commands:
```
LD_DEBUG=libs clinfo
ldconfig -p | grep gmm
LIB=`ldconfig -p | grep gmm | awk '{print $4}'`; nm -DC $LIB | grep -E "Init|Destroy"
```

pszi1ard, 2020-01-14:
> @pszi1ard What Linux distribution are you using? What are the versions of Neo and gmmlib? Where did you get binaries for Neo and gmmlib from?

Ubuntu 18.04.3 with the latest packages from http://ppa.launchpad.net/ocl-dev/intel-opencl/ubuntu
(igc is 1.0.3041-1~ppa1~bionic1, intel-gmmlib is 19.4.1-1~ppa1~bionic1).

> Can you provide logs from below commands:
> 
> ```
> LD_DEBUG=libs clinfo

```
 $ LD_DEBUG=libs clinfo
     21294:	find library=libgtk3-nocsd.so.0 [0]; searching
     21294:	 search cache=/etc/ld.so.cache
     21294:	  trying file=/usr/lib/x86_64-linux-gnu/libgtk3-nocsd.so.0
     21294:	
     21294:	find library=libOpenCL.so.1 [0]; searching
     21294:	 search cache=/etc/ld.so.cache
     21294:	  trying file=/usr/lib/x86_64-linux-gnu/libOpenCL.so.1
     21294:	
     21294:	find library=libdl.so.2 [0]; searching
     21294:	 search cache=/etc/ld.so.cache
     21294:	  trying file=/lib/x86_64-linux-gnu/libdl.so.2
     21294:	
     21294:	find library=libc.so.6 [0]; searching
     21294:	 search cache=/etc/ld.so.cache
     21294:	  trying file=/lib/x86_64-linux-gnu/libc.so.6
     21294:	
     21294:	find library=libpthread.so.0 [0]; searching
     21294:	 search cache=/etc/ld.so.cache
     21294:	  trying file=/lib/x86_64-linux-gnu/libpthread.so.0
     21294:	
     21294:	
     21294:	calling init: /lib/x86_64-linux-gnu/libpthread.so.0
     21294:	
     21294:	
     21294:	calling init: /lib/x86_64-linux-gnu/libc.so.6
     21294:	
     21294:	
     21294:	calling init: /lib/x86_64-linux-gnu/libdl.so.2
     21294:	
     21294:	
     21294:	calling init: /usr/lib/x86_64-linux-gnu/libOpenCL.so.1
     21294:	
     21294:	
     21294:	calling init: /usr/lib/x86_64-linux-gnu/libgtk3-nocsd.so.0
     21294:	
     21294:	
     21294:	initialize program: clinfo
     21294:	
     21294:	
     21294:	transferring control: clinfo
     21294:	
     21294:	find library=libstdc++.so.6 [0]; searching
     21294:	 search cache=/etc/ld.so.cache
     21294:	  trying file=/usr/lib/x86_64-linux-gnu/libstdc++.so.6
     21294:	
     21294:	find library=libm.so.6 [0]; searching
     21294:	 search cache=/etc/ld.so.cache
     21294:	  trying file=/lib/x86_64-linux-gnu/libm.so.6
     21294:	
     21294:	find library=libgcc_s.so.1 [0]; searching
     21294:	 search cache=/etc/ld.so.cache
     21294:	  trying file=/lib/x86_64-linux-gnu/libgcc_s.so.1
     21294:	
     21294:	
     21294:	calling init: /lib/x86_64-linux-gnu/libgcc_s.so.1
     21294:	
     21294:	
     21294:	calling init: /lib/x86_64-linux-gnu/libm.so.6
[... 66 lines cut ...]
```
$ LIB=`ldconfig -p | grep gmm | awk '{print $4}'`; nm -DC $LIB | grep -E "Init|Destroy"
0000000000062d60 T GmmAdapterDestroy
0000000000062d40 T GmmDestroy
0000000000062c80 T GmmInit
0000000000062cc0 T InitializeGmm
000000000005ee10 T GmmLib::GmmPageTableMgr::InitContextAuxTableRegister(void*, _GMM_ENGINE_TYPE)
00000000000621d0 T GmmLib::GmmClientContext::DestroyResInfoObject(GmmLib::GmmResourceInfoLin*)
0000000000061d80 T GmmLib::GmmClientContext::DestroyResInfoObject(GmmLib::GmmResourceInfoLin*, _GmmClientAllocationCallbacks_*)
0000000000061d50 T GmmLib::GmmClientContext::DestroyPageTblMgrObject(GmmLib::GmmPageTableMgr*)
0000000000061d00 T GmmLib::GmmClientContext::DestroyPageTblMgrObject(GmmLib::GmmPageTableMgr*, _GmmClientAllocationCallbacks_*)
                 U std::ios_base::Init::Init()
                 U std::ios_base::Init::~Init()
0000000000062d60 T GmmAdapterDestroy
0000000000062d40 T GmmDestroy
0000000000062c80 T GmmInit
0000000000062cc0 T InitializeGmm
000000000005ee10 T GmmLib::GmmPageTableMgr::InitContextAuxTableRegister(void*, _GMM_ENGINE_TYPE)
00000000000621d0 T GmmLib::GmmClientContext::DestroyResInfoObject(GmmLib::GmmResourceInfoLin*)
0000000000061d80 T GmmLib::GmmClientContext::DestroyResInfoObject(GmmLib::GmmResourceInfoLin*, _GmmClientAllocationCallbacks_*)
0000000000061d50 T GmmLib::GmmClientContext::DestroyPageTblMgrObject(GmmLib::GmmPageTableMgr*)
0000000000061d00 T GmmLib::GmmClientContext::DestroyPageTblMgrObject(GmmLib::GmmPageTableMgr*, _GmmClientAllocationCallbacks_*)
                 U std::ios_base::Init::Init()
                 U std::ios_base::Init::~Init()
```

JacekDanecki, 2020-01-14:
@pszi1ard You are using Neo 19.50.15067, and this release used intel-gmmlib-19.3.4 with libigdgmm.so.12, but in your cache there is libigdgmm.so.11 from gmmlib 19.4.1.
Please update Neo to 20.01.15264, this release switched to gmmlib 19.4.1 with libigdgmm.so.11.

pszi1ard, 2020-01-14:
> @pszi1ard You are using Neo 19.50.15067, and this release used intel-gmmlib-19.3.4 with libigdgmm.so.12, but in your cache there is libigdgmm.so.11 from gmmlib 19.4.1.
> Please update Neo to 20.01.15264, this release switched to gmmlib 19.4.1 with libigdgmm.so.11.

Thanks for the suggestion? Any idea why is this happening? If I purge all packages ans simply install with apt, I get gmmlib 19.4.1 and intel-opencl 19.50.15067:
```
$ apt list  intel-gmmlib intel-opencl
Listing... Done
intel-gmmlib/bionic 19.4.1-1~ppa1~bionic1 amd64
intel-opencl/bionic 19.50.15067-1~ppa1~bionic1 amd64
```

JacekDanecki, 2020-01-14:
Have you executed apt-get update?
```
$ apt-cache policy intel-opencl-icd
intel-opencl-icd:
  Installed: (none)
  Candidate: 19.52.15209-1~ppa1~bionic1
  Version table:
     19.52.15209-1~ppa1~bionic1 500
        500 http://ppa.launchpad.net/jdanecki/intel-opencl/ubuntu bionic/main amd64 Packages
```
apt-get update
```
$ apt-cache policy intel-opencl-icd
intel-opencl-icd:
  Installed: (none)
  Candidate: 20.01.15264-1~ppa1~bionic1
  Version table:
     20.01.15264-1~ppa1~bionic1 500
        500 http://ppa.launchpad.net/jdanecki/intel-opencl/ubuntu bionic/main amd64 Packages
```

JacekDanecki, 2020-01-14:
You are using different ppa: http://ppa.launchpad.net/ocl-dev/intel-opencl/ubuntu
New Neo is on ~intel-opencl and ~jdanecki ppa:
* https://launchpad.net/~intel-opencl/+archive/ubuntu/intel-opencl
* https://launchpad.net/~jdanecki/+archive/ubuntu/intel-opencl

ocl-dev ppa is used only by external CI systems and provide packages for gmmlib and IGC only.

pszi1ard, 2020-01-14:
@JacekDanecki indeed, I just realized that I'm using a different PPA. Thanks for the help!

JacekDanecki, 2020-01-15:
@jeffhammond Starting with commit e3cc20fdcc0387c5f6705dd86eddd86e2157de84, Neo is being linked with gmmlib, so we don't use dlopen to load gmmlib anymore.
```
ldd libigdrcl.so 
        linux-vdso.so.1 (0x00007ffecd976000)
        libdl.so.2 => /lib64/libdl.so.2 (0x00007fd39e9a7000)
        libpthread.so.0 => /lib64/libpthread.so.0 (0x00007fd39e986000)
        libigdgmm.so.11 => /usr/local/lib64/libigdgmm.so.11 (0x00007fd39e712000)
        libstdc++.so.6 => /lib64/libstdc++.so.6 (0x00007fd39e518000)
        libm.so.6 => /lib64/libm.so.6 (0x00007fd39e3d2000)
        libgcc_s.so.1 => /lib64/libgcc_s.so.1 (0x00007fd39e3b8000)
        libc.so.6 => /lib64/libc.so.6 (0x00007fd39e1f0000)
        /lib64/ld-linux-x86-64.so.2 (0x00007fd39f133000)
```
If you don't have gmmlib installed
```
ldd libigdrcl.so 
        linux-vdso.so.1 (0x00007fff09bc4000)
        libdl.so.2 => /lib64/libdl.so.2 (0x00007ff7eadf4000)
        libpthread.so.0 => /lib64/libpthread.so.0 (0x00007ff7eadd3000)
        libigdgmm.so.11 => not found
        libstdc++.so.6 => /lib64/libstdc++.so.6 (0x00007ff7eabd9000)
        libm.so.6 => /lib64/libm.so.6 (0x00007ff7eaa93000)
        libgcc_s.so.1 => /lib64/libgcc_s.so.1 (0x00007ff7eaa79000)
        libc.so.6 => /lib64/libc.so.6 (0x00007ff7ea8b1000)
        /lib64/ld-linux-x86-64.so.2 (0x00007ff7eb580000)
```
clinfo will not present platform: Intel(R) OpenCL HD Graphics
```
clinfo
Number of platforms                               0
```
If you have installed incompatible gmmlib installed
```
ldd -r libigdrcl.so 
        linux-vdso.so.1 (0x00007ffdaf542000)
        libdl.so.2 => /lib64/libdl.so.2 (0x00007f975d2fd000)
        libpthread.so.0 => /lib64/libpthread.so.0 (0x00007f975d2dc000)
        libigdgmm.so.11 => /usr/local/lib64/libigdgmm.so.11 (0x00007f975d1fd000)
        libstdc++.so.6 => /lib64/libstdc++.so.6 (0x00007f975d003000)
        libm.so.6 => /lib64/libm.so.6 (0x00007f975cebd000)
        libgcc_s.so.1 => /lib64/libgcc_s.so.1 (0x00007f975cea3000)
        libc.so.6 => /lib64/libc.so.6 (0x00007f975ccdb000)
        /lib64/ld-linux-x86-64.so.2 (0x00007f975da89000)
undefined symbol: GmmAdapterDestroy     (./libigdrcl.so)
undefined symbol: InitializeGmm (./libigdrcl.so)
```
clinfo will return error
```
clinfo 
clinfo: symbol lookup error: /usr/local/lib64/intel-opencl/libigdrcl.so: undefined symbol: InitializeGmm
```
Mentioned commit fixes this issue. If you find new problem please fill new issue.

closed 2020-01-15 by JacekDanecki

JacekDanecki, 2020-01-16:
@pszi1ard Neo was updated to 20.01.15264 on ~ocl-dev ppa.