intel/compute-runtime #235: [SYCL] Weird performance behaviour

cagnulein, 2019-11-15:
I'm playing around with SYCL and i'm facing a strange behaviour. I'm posting here because from the analysis on the SYCL repo, everything point to some OCL GPU issue: more detail here https://github.com/intel/llvm/issues/813 .

Just to explain here the issue too: I've a really small piece of code that has only one huge for loop with nothing inside.
For what i saw, some number of iterations create huge delay for no reason at all.

Example:

```
        sycl::queue myQueue(sycl::gpu_selector{});
        sycl::buffer<unsigned char, 1> inputBuf(old_image, sycl::range<1>(IMAGE_WIDTH*IMAGE_HEIGHT));
        myQueue.submit([&](sycl::handler& cgh) {
            auto readImage = inputBuf.get_access<sycl::access::mode::read>(cgh);
            cgh.parallel_for<class simple_test>(sycl::range<1>(1821303172), [=](sycl::id<1> idx) {
            });
        });
```

has a execution time of:

**real 0m0.978s**
user 0m0.172s
sys 0m0.084s

the same algorithm, decreasing the iterations to 1761303172 instead of 1821303172, have these deterministic times:

**real 0m4.541s**
user 0m0.190s
sys 0m0.065s

You can find my test repo here: https://github.com/cagnulein/sycl-benchmarks/tree/master/issues

Futhermore, i've test several indexes for my empty loop and i discovered this:
<pre>
./test-around.sh
Index           Time 1st try    Time 2nd try    Time 3rd try

1761303161      1.548           1.561           1.563
<b>1761303162      3.177           3.173           3.194
1761303163      16.447          16.431          16.459</b>
1761303164      1.249           1.235           1.247
1761303165      1.953           1.967           1.955
1761303166      2.052           2.039           2.038
1761303167      2.165           2.172           2.156
1761303168      1.261           1.271           1.290
1761303169      1.392           1.411           1.407
1761303170      1.299           1.305           1.307
<b>1761303171      5.829           5.842           5.838
1761303172      4.509           4.500           4.518</b>
1761303173      1.288           1.291           1.308
1761303174      1.232           1.225           1.242
1761303175      1.813           1.838           1.796
<b>1761303176      2.507           2.528           2.511</b>
1761303177      1.706           1.716           1.699
<b>1761303178      8.490           8.477           8.506
1761303179      16.436          16.447          16.432</b>
1761303180      1.407           1.411           1.429
<b>1761303181      2.465           2.465           2.451
1761303182      8.492           8.502           8.479</b>
1761303183      1.263           1.246           1.239
1761303184      1.847           1.887           1.845
</pre>

With the help of @AlexeySachkov i investigated more with the help of intel/opencl-intercept-layer and i've discovered that the codes has always a slowdown here:
<pre>
>>>> clCreateKernel: program = 0xf81df0, kernel_name = _ZTSZZ4mainENK3$_0clERN2cl4sycl7handlerEE11simple_test
<<<< clCreateKernel: returned 0x16e59c0 -> CL_SUCCESS
>>>> clSetKernelExecInfo( _ZTSZZ4mainENK3$_0clERN2cl4sycl7handlerEE11simple_test ): param_name = CL_KERNEL_EXEC_INFO_INDIRECT_HOST_ACCESS_INTEL (00004200)
<<<< clSetKernelExecInfo -> CL_SUCCESS
>>>> clSetKernelExecInfo( _ZTSZZ4mainENK3$_0clERN2cl4sycl7handlerEE11simple_test ): param_name = CL_KERNEL_EXEC_INFO_INDIRECT_DEVICE_ACCESS_INTEL (00004201)
<<<< clSetKernelExecInfo -> CL_SUCCESS
>>>> clSetKernelExecInfo( _ZTSZZ4mainENK3$_0clERN2cl4sycl7handlerEE11simple_test ): param_name = CL_KERNEL_EXEC_INFO_INDIRECT_SHARED_ACCESS_INTEL (00004202)
<<<< clSetKernelExecInfo -> CL_SUCCESS
>>>> clEnqueueNDRangeKernel( _ZTSZZ4mainENK3$_0clERN2cl4sycl7handlerEE11simple_test ): queue = 0xf3ce60, kernel = 0x16e59c0, global_work_offset = < 0 >, global_work_size = < 1761303171 >, local_work_size = < NULL >
<<<< clEnqueueNDRangeKernel created event = 0x16e0b60 -> CL_SUCCESS
<b>>>>> clWaitForEvents: event_list = ( size = 1 )[ 0x16e0b60 ]</b>
<<<< clWaitForEvents -> CL_SUCCESS
>>>> clReleaseMemObject: [ ref count = 1 ] mem = 0x16e0d10
<<<< clReleaseMemObject: [ ref count = 0 ] -> CL_SUCCESS
>>>> clReleaseCommandQueue: [ ref count = 1 ] command_queue = 0xf3ce60
<<<< clReleaseCommandQueue: [ ref count = 0 ] -> CL_SUCCESS
>>>> clReleaseEvent: [ ref count = 1 ] event = 0x16e0b60
<<<< clReleaseEvent: [ ref count = 0 ] -> CL_SUCCESS
</pre>

@AlexeySachkov suggest to provide a pure OpenCL implementation and i will share it as soon i will finish it; in the meanwhile, if you have some suggestions please feel free to share with me :)

cagnulein, 2019-11-15:
Same issue happens with pure OPENCL

time ./rotate-opencl.gpu **1761303172** && time ./rotate-opencl.gpu **1821303172**
Using platform: Intel(R) OpenCL HD Graphics
Using device: Intel(R) Gen9 HD Graphics NEO

**real    0m4.412s**
user    0m0.394s
sys     0m0.058s
Using platform: Intel(R) OpenCL HD Graphics
Using device: Intel(R) Gen9 HD Graphics NEO

**real    0m0.851s**
user    0m0.417s
sys     0m0.024s

Here you can find my example code: https://github.com/cagnulein/sycl-benchmarks/tree/master/issues 
In particular the source OPENCL code: https://github.com/cagnulein/sycl-benchmarks/blob/master/issues/rotate-opencl.cpp
and the binary: https://github.com/cagnulein/sycl-benchmarks/blob/master/issues/rotate-opencl.gpu

@AlexeySachkov you were right! :)

cagnulein, 2019-11-15:
Other useful information with always in plain OPENCL

<pre>
./test-around-opencl.sh
Index           Time 1st try    Time 2nd try    Time 3rd try

1761303161      0.950           0.944           0.939
<b>1761303162      3.109           3.071           3.080
1761303163      16.302          16.329          16.351</b>
1761303164      0.649           0.660           0.663
1761303165      1.479           1.512           1.480
1761303166      1.593           1.578           1.591
1761303167      1.662           1.642           1.668
1761303168      0.698           0.687           0.696
1761303169      0.836           0.835           0.831
1761303170      0.714           0.679           0.676
<b>1761303171      5.755           5.717           5.745
1761303172      4.424           4.413           4.409</b>
1761303173      0.710           0.710           0.722
1761303174      0.670           0.669           0.653
1761303175      1.088           1.077           1.078
<b>1761303176      2.418           2.422           2.430</b>
1761303177      1.024           1.022           1.029
<b>1761303178      8.391           8.397           8.354
1761303179      16.353          16.373          16.331</b>
1761303180      0.805           0.786           0.776
1761303181      1.901           1.878           1.895
<b>1761303182      8.394           8.385           8.386</b>
1761303183      0.654           0.663           0.664
1761303184      1.435           1.414           1.443
</pre>

MichalMrozek, 2019-11-18:
Hello,
Looking at Kernel code , I think it is incorrect, here is what happens:
ratio = (n / Nthreads);
n=100
NThreads is equal to what you pass, without parameters it is 1761303171

since ratio is integer we divide small number by big number so we have ratio == 0
ratio = (100 / 1761303171) == 0

so Kernel does nothing as:
        "       start = ratio * ID;"
        "       stop  = ratio * (ID + 1);"

start = 0
stop = 0

This loop doesn't do anything
        "       for (int i=start; i<stop; i++)"
        "           C[i] = A[i] + B[i];"

So your case just executes NDR with different global sizes.
Now since this is OpenCL 1.2 kernel, the local workgroup size needs to be uniform, it needs to divide global work size evenly.

If we look at this global size:
1761303163      

Its divisors are:
1 31063 56701 1761303163

Since max local work group size is 256, driver can choose only 1.
So your sample for this global work size spawns 1761303163 Hardware Threads each servicing one work item essentially doing almost nothing. We rather benchmark here how fast we can spawn Hardware Threads

Now if we look at 1761303164     , divisors are as follows:
1 2 4 59 118 236 7463149 14926298 29852596 440325791 880651582 1761303164

Driver can choose 236 here, which results in 15 HW Threads ( SIMD 16 dispatch. 14 * 16 +  12 for last thread ).
So we execute here 7463149 workgorups ( 1761303164  / 236 ) each containing 15 HW threads, so we exectute total of
7463149 * 15 = 111947235 threads

Now if we look at ratio
1761303164      / 111947235  = 15.7 it follows more or less your time observation ratio:
16.302          / 0.649           = 25.1 

Ratio is not directly the same as Kernel is actually doing very little:

L0:
(W)      mov (8|M0)               r2.0<1>:ud    r0.0<1;1,0>:ud                  
(W)      or (1|M0)                cr0.0<1>:ud   cr0.0<0;1,0>:ud   0x4C0:uw         {Switch}
(W)      mov (1|M0)               r10.0<1>:uq   r5.3<0;1,0>:uq                  
(W)      mul (1|M0)               r3.0<1>:d     r7.0<0;1,0>:d     r2.1<0;1,0>:d    {Compacted}
(W)      send (1|M0)              r5       r10:uq  0xC         0x41401FF  //  wr:2+?, rd:1, A64 Scattered Read msc:1, to global memory
         add (16|M0)              r8.0<1>:d     r3.0<0;1,0>:d     r1.0<16;16,1>:uw
         add (16|M0)              r8.0<1>:d     r8.0<8;8,1>:d     r4.0<0;1,0>:d    {Compacted}
(W)      math.iqot (1|M0)         r3.1<1>:d     r5.0<0;1,0>:d     r6.3<0;1,0>:d   
         mul (16|M0)              r11.0<1>:d    r3.1<0;1,0>:d     r8.0<8;8,1>:d    {Compacted}
         add (16|M0)              r8.0<1>:d     r8.0<8;8,1>:d     1:w             
         mul (16|M0)              r13.0<1>:d    r3.1<0;1,0>:d     r8.0<8;8,1>:d    {Compacted}
         cmp (16|M0)   (lt)f0.0   null<1>:d     r11.0<8;8,1>:d    r13.0<8;8,1>:d   {Compacted}
(f0.0)   if (16|M0)                           L304              L304            
L168:
         shl (16|M0)              r15.0<1>:d    r11.0<8;8,1>:d    2:w             
         add (16|M0)              r11.0<1>:d    r11.0<8;8,1>:d    1:w             
         add (16|M0)              r17.0<1>:d    r15.0<8;8,1>:d    r6.0<0;1,0>:d    {Compacted}
         add (16|M0)              r21.0<1>:d    r15.0<8;8,1>:d    r6.1<0;1,0>:d    {Compacted}
         add (16|M0)              r25.0<1>:d    r15.0<8;8,1>:d    r6.2<0;1,0>:d    {Compacted}
         cmp (16|M0)   (lt)f0.1   null<1>:d     r11.0<8;8,1>:d    r13.0<8;8,1>:d   {Compacted}
         send (16|M0)             r19:w    r17     0xC         0x4205E00  //  wr:2+?, rd:2, Untyped Surface Read msc:30, to bti 0
         send (16|M0)             r23:w    r21     0xC         0x4205E01  //  wr:2+?, rd:2, Untyped Surface Read msc:30, to bti 1
         add (16|M0)              r19.0<1>:d    r19.0<8;8,1>:d    r23.0<8;8,1>:d   {Compacted}
         sends (16|M0)            null:w   r25     r19     0x8C        0x4025E02  //  wr:2+2, rd:0, Untyped Surface Write msc:30, to bti 2
(f0.1)   while (16|M0)                        L168                            
L304:
         endif (16|M0)                        L320                            
L320:
(W)      mov (8|M0)               r127.0<1>:ud  r2.0<8;8,1>:ud                   {Compacted}
(W)      mov (1|M0)               null<1>:ud    f0.0<0;1,0>:ud                  
(W)      send (8|M0)              null     r127    0x27        0x2000010  {EOT} //  wr:1+?, rd:0,  end of thread
L360:

So to summarize, your observations are perfectly correct :)
Sometime choosing bigger gws results in faster execution as there are better divisors, especially if Kernel is doing almost nothing.

cagnulein, 2019-11-18:
@MichalMrozek thanks for your answer, i'm still reading it but i think you had miswatch my code.
Infact in the [rotate-opencl.cpp](https://github.com/cagnulein/sycl-benchmarks/blob/master/issues/rotate-opencl.cpp) i didn't call the _full_kernel_code_ function, but the _kernel_code_ that does absolutely nothing.

Do you think that this point could change your analysis?

MichalMrozek, 2019-11-18:
Oh yeah then it is even simpler and above analysis are also correct.

cagnulein, 2019-11-18:
@MichalMrozek understood and your explanation fits at all. My only question is: do you think it could be managed in someway inside the OpenCL? For example, _let me say simple_, adding the for +1 in order to have a better divisors? It should help the developer a lot I guess. (i think i will not be the only one that will facing this "issue")

In the meanwhile, i will try to add extra loops on my [benchmarks](https://github.com/cagnulein/sycl-benchmarks/tree/master/rotate) in order to check if your explanation fits the realty :D

MichalMrozek, 2019-11-18:
Driver cannot modify global work size.
What you may try is to use flexible work group sizes, which is an OpenCL 2.0 feature.
Just compiler your program using 2.0 standard and driver will be able to choose non uniform local work group size.

https://www.khronos.org/registry/OpenCL/sdk/2.0/docs/man/xhtml/clBuildProgram.html
```
Options Controlling the OpenCL C Version
The following option controls the version of OpenCL C that the compiler accepts.
-cl-std=
Determine the OpenCL C language version to use. A value for this option must be provided. Valid values are:

...

CL2.0 - Support all OpenCL C programs that use the OpenCL C language features defined in section 6 of the OpenCL 2.0 specification.

```

MichalMrozek, 2019-11-28:
Closing as question is answered, feel free to re-open or create new issues for subsequent findings.

closed 2019-11-28 by MichalMrozek