intel/compute-runtime #236: SPIRVWriter.cpp:667: SPIRV::SPIRVInstruction* SPIRV::LLVMToSPIRV::transUnaryInst(llvm::UnaryInstruction*, SPIRV::SPIRVBasicBlock*): Assertion `Cast->getSrcTy()->getPointerAddressSpace() == SPIRAS_Generic' failed.

jeffersonchua, 2019-11-15:
Need help to fix this ..

[244/383] Generating ../../bin/scheduler/x64/gen10/scheduler_Gen10core.bin
FAILED: bin/scheduler/x64/gen10/scheduler_Gen10core.bin 
cd /v6/src2/icr/compute-runtime/runtime/scheduler && LD_LIBRARY_PATH=/v6/src2/icr/compute-runtime/build/bin /v6/src2/icr/compute-runtime/build/bin/cloc -q -file scheduler.cl -device cnl -cl-intel-greater-than-4GB-buffer-required -64 -out_dir /v6/src2/icr/compute-runtime/build/bin/scheduler/x64/gen10 -cpp_file -options -cl-kernel-arg-info\ -I/usr/local/include/igc\ -I/usr/local/include/igc/cif\ -I/usr/local/include/igc/ocl_igc_shared/executable_format\ -I/usr/local/include/igc/ocl_igc_shared/device_enqueue\ -I\ ../gen10\ \ -cl-std=CL2.0
cloc: ../lib/SPIRV/SPIRVWriter.cpp:667: SPIRV::SPIRVInstruction* SPIRV::LLVMToSPIRV::transUnaryInst(llvm::UnaryInstruction*, SPIRV::SPIRVBasicBlock*): Assertion `Cast->getSrcTy()->getPointerAddressSpace() == SPIRAS_Generic' failed.
[249/383] Building CXX object igdrcl_l...se.dir/gen10/command_queue_gen10.cpp.o
ninja: build stopped: subcommand failed.

Here's what I pulled
git clone -b release/8.x https://github.com/llvm/llvm-project
git clone -b ocl-open-80 https://github.com/intel/opencl-clang
git clone https://github.com/KhronosGroup/OpenCL-Headers.git
git clone https://github.com/KhronosGroup/OpenCL-ICD-Loader.git
git clone -b releases/igc-1.0.2714 https://github.com/intel/intel-graphics-compiler.git
git clone -b llvm_release_80 https://github.com/KhronosGroup/SPIRV-LLVM-Translator.git
git clone -b intel-gmmlib-18.3 https://github.com/intel/gmmlib.git
git clone -b 18.41 https://github.com/intel/compute-runtime

everything built fine and installed except can't get pass this step while building the compute-runtime.

Thank you.

Jeff.

JacekDanecki, 2019-11-15:
Why are you mixing very old Neo with new IGC? Neo on branch 18.41 was verified with older IGC, see [release 18.41.11649](https://github.com/intel/compute-runtime/releases/tag/18.41.11649)
IGC 1.0.2714 was verified with llvm/clang 9.

jeffersonchua, 2019-12-03:
> Why are you mixing very old Neo with new IGC? Neo on branch 18.41 was verified with older IGC, see release 18.41.11649

Just plain ignorant. I'll try the new release.

Thank you!

Jeff

JacekDanecki, 2019-12-03:
Closing issue. Feel free to create one if you find any problems with new Neo release.

closed 2019-12-03 by JacekDanecki

jeffersonchua, 2019-12-03:
Trying to compile igc using releases/igc-1.0.2714 with LLVM 9.0.1 and encountered the following error ... 
(I just wanted to a working compute-runtime ... but can't yet)

ninja: Entering directory `build'
[1/2] Linking CXX shared library Release/libigc.so.1.0.1
FAILED: Release/libigc.so.1.0.1 
: && /usr/bin/c++ -fPIC -fno-exceptions -fdata-sections -ffunction-sections -O2 -pipe -fmessage-length=0 -march=corei7 -mstackrealign -fms-extensions -Werror -Wno-unused-parameter -Wno-missing-field-initializers -Wwrite-strings -Wno-long-long -Wswitch -Wno-sign-compare -Wno-unused-result -Wno-enum-compare -Wno-type-limits -Wno-ignored-qualifiers -Wformat -Wformat-security -Wno-extra -Wno-write-strings -finline -fno-strict-aliasing -msse -msse2 -msse3 -mssse3 -msse4 -msse4.1 -msse4.2 -Wno-unknown-pragmas -fPIC -D_FORTIFY_SOURCE=2 -fstack-protector -finline-functions -funswitch-loops -Wno-maybe-uninitialized -lrt -fno-rtti -fvisibility=hidden -fvisibility-inlines-hidden -std=c++14 -DNDEBUG -g  -Wl,--no-undefined -Wl,-z,noexecstack -Wl,-z,relro -Wl,-z,now -Wl,--gc-sections -shared -Wl,-soname,libigc.so.1 -o Release/libigc.so.1.0.1 CMakeFiles/igc_dll.dir/common/CompilerStatsUtils.cpp.o CMakeFiles/igc_dll.dir/common/igc_regkeys.cpp.o CMakeFiles/igc_dll.dir/common/LLVMUtils.cpp.o CMakeFiles/igc_dll.dir/common/ShaderOverride.cpp.o CMakeFiles/igc_dll.dir/common/Stats.cpp.o CMakeFiles/igc_dll.dir/common/SysUtils.cpp.o CMakeFiles/igc_dll.dir/common/debug/Debug.cpp.o CMakeFiles/igc_dll.dir/common/debug/Dump.cpp.o CMakeFiles/igc_dll.dir/common/debug/TeeOutputStream.cpp.o CMakeFiles/igc_dll.dir/common/SystemThread.cpp.o CMakeFiles/igc_dll.dir/common/MDFrameWork.cpp.o CMakeFiles/igc_dll.dir/common/FunctionUpgrader.cpp.o CMakeFiles/igc_dll.dir/common/allocator.cpp.o CMakeFiles/igc_dll.dir/AdaptorOCL/CLElfLib/ElfReader.cpp.o CMakeFiles/igc_dll.dir/AdaptorOCL/CLElfLib/ElfWriter.cpp.o CMakeFiles/igc_dll.dir/AdaptorCommon/AddImplicitArgs.cpp.o CMakeFiles/igc_dll.dir/AdaptorCommon/customApi.cpp.o CMakeFiles/igc_dll.dir/AdaptorOCL/dllInterfaceCompute.cpp.o CMakeFiles/igc_dll.dir/AdaptorCommon/ImplicitArgs.cpp.o CMakeFiles/igc_dll.dir/AdaptorCommon/ProcessFuncAttributes.cpp.o CMakeFiles/igc_dll.dir/AdaptorCommon/TypesLegalizationPass.cpp.o CMakeFiles/igc_dll.dir/AdaptorCommon/IRUpgrader/UpgraderResourceAccess.cpp.o CMakeFiles/igc_dll.dir/AdaptorOCL/OCL/LoadBuffer.cpp.o CMakeFiles/igc_dll.dir/AdaptorOCL/OCL/Patch/patch_parser.cpp.o CMakeFiles/igc_dll.dir/AdaptorOCL/OCL/Platform/cmd_media_caps_g8.cpp.o CMakeFiles/igc_dll.dir/AdaptorOCL/OCL/Platform/cmd_parser_g8.cpp.o CMakeFiles/igc_dll.dir/AdaptorOCL/OCL/sp/spp_g8.cpp.o CMakeFiles/igc_dll.dir/AdaptorOCL/OCL/sp/sp_g8.cpp.o CMakeFiles/igc_dll.dir/AdaptorOCL/OCL/sp/gtpin_igc_ocl.cpp.o CMakeFiles/igc_dll.dir/AdaptorOCL/OCL/sp/sp_debug.cpp.o CMakeFiles/igc_dll.dir/AdaptorOCL/OCL/util/BinaryStream.cpp.o CMakeFiles/igc_dll.dir/AdaptorOCL/Upgrader/llvm9/Upgrader.cpp.o CMakeFiles/igc_dll.dir/AdaptorOCL/Upgrader/llvm9/BitcodeReader.cpp.o CMakeFiles/igc_dll.dir/AdaptorOCL/UnifyIROCL.cpp.o CMakeFiles/igc_dll.dir/AdaptorOCL/cmc.cpp.o CMakeFiles/igc_dll.dir/AdaptorOCL/MoveStaticAllocas.cpp.o CMakeFiles/igc_dll.dir/AdaptorOCL/SPIRV/libSPIRV/SPIRVBasicBlock.cpp.o CMakeFiles/igc_dll.dir/AdaptorOCL/SPIRV/libSPIRV/SPIRVDebug.cpp.o CMakeFiles/igc_dll.dir/AdaptorOCL/SPIRV/libSPIRV/SPIRVDecorate.cpp.o CMakeFiles/igc_dll.dir/AdaptorOCL/SPIRV/libSPIRV/SPIRVEntry.cpp.o CMakeFiles/igc_dll.dir/AdaptorOCL/SPIRV/libSPIRV/SPIRVFunction.cpp.o CMakeFiles/igc_dll.dir/AdaptorOCL/SPIRV/libSPIRV/SPIRVInstruction.cpp.o CMakeFiles/igc_dll.dir/AdaptorOCL/SPIRV/libSPIRV/SPIRVModule.cpp.o CMakeFiles/igc_dll.dir/AdaptorOCL/SPIRV/libSPIRV/SPIRVStream.cpp.o CMakeFiles/igc_dll.dir/AdaptorOCL/SPIRV/libSPIRV/SPIRVType.cpp.o CMakeFiles/igc_dll.dir/AdaptorOCL/SPIRV/libSPIRV/SPIRVValue.cpp.o CMakeFiles/igc_dll.dir/AdaptorOCL/SPIRV/SPIRVReader.cpp.o CMakeFiles/igc_dll.dir/AdaptorOCL/SPIRV/SPIRVUtil.cpp.o CMakeFiles/igc_dll.dir/AdaptorOCL/SPIRV/SPIRVException.cpp.o CMakeFiles/igc_dll.dir/AdaptorOCL/ocl_igc_interface/impl/igc_features_and_workarounds_impl.cpp.o CMakeFiles/igc_dll.dir/AdaptorOCL/ocl_igc_interface/impl/igc_ocl_device_ctx_impl.cpp.o CMakeFiles/igc_dll.dir/AdaptorOCL/ocl_igc_interface/impl/igc_ocl_translation_ctx_impl.cpp.o CMakeFiles/igc_dll.dir/AdaptorOCL/ocl_igc_interface/impl/ocl_gen_binary_impl.cpp.o CMakeFiles/igc_dll.dir/AdaptorOCL/ocl_igc_interface/impl/ocl_translation_output_impl.cpp.o CMakeFiles/igc_dll.dir/AdaptorOCL/ocl_igc_interface/impl/gt_system_info_impl.cpp.o CMakeFiles/igc_dll.dir/AdaptorOCL/ocl_igc_interface/impl/platform_impl.cpp.o CMakeFiles/igc_dll.dir/AdaptorOCL/cif/cif/builtins/memory/buffer/impl/buffer_impl.cpp.o CMakeFiles/igc_dll.dir/AdaptorOCL/cif/cif/builtins/builtins_registry.cpp.o CMakeFiles/igc_dll.dir/AdaptorOCL/cif/cif/export/cif_main.cpp.o CMakeFiles/igc_dll.dir/AdaptorOCL/cif/cif/export/registry.cpp.o CMakeFiles/igc_dll.dir/AdaptorOCL/cif/cif/helpers/error.cpp.o  -Wl,-rpath,/opt/llvm/lib: Release/libCompiler.a Release/libGenISAIntrinsics.a visa/igfxcmjit64.a -Wl,--whole-archive Release/libBiFLibOcl.a -Wl,--no-whole-archive -Wl,--start-group /opt/llvm/lib/libLLVMipo.so.9 /opt/llvm/lib/libLLVMIRReader.so.9 /opt/llvm/lib/libLLVMBitWriter.so.9 /opt/llvm/lib/libLLVMBinaryFormat.so.9 /opt/llvm/lib/libLLVMAsmParser.so.9 /opt/llvm/lib/libLLVMBitReader.so.9 /opt/llvm/lib/libLLVMLinker.so.9 /opt/llvm/lib/libLLVMCodeGen.so.9 /opt/llvm/lib/libLLVMScalarOpts.so.9 /opt/llvm/lib/libLLVMTransformUtil [... cut ...]
/usr/lib/gcc/x86_64-pc-linux-gnu/8.3.1/../../../../x86_64-pc-linux-gnu/bin/ld: CMakeFiles/igc_dll.dir/AdaptorOCL/Upgrader/llvm9/BitcodeReader.cpp.o: in function `(anonymous namespace)::BitcodeReader::parseValueSymbolTable(unsigned long)':
/v6/src2/igc/intel-graphics-compiler/IGC/AdaptorOCL/Upgrader/llvm9/BitcodeReader.cpp:2197: undefined reference to `llvm::BitstreamCursor::EnterSubBlock(unsigned int, unsigned int*)'

Perhaps it's the wrong LLVM version I'm using? I'm using the git release/9.x branch.

Thanks,
Jeff

lwesiers, 2019-12-04:
Hello there,

Please switch back to the LLVM 9.0.0 version (even better to the this tag : [llvmorg-9.0.0](https://github.com/llvm/llvm-project/releases/tag/llvmorg-9.0.0) ). Soon we will fix this build break for version 9.0.1

jeffersonchua, 2019-12-04:
With 9.0.0, it's having issue during "linking" ...

[ 75%] Linking CXX shared library Release/libigc.so
/usr/lib/gcc/x86_64-pc-linux-gnu/8.3.1/../../../../x86_64-pc-linux-gnu/bin/ld:
CMakeFiles/igc_dll.dir/AdaptorOCL/Upgrader/llvm9/BitcodeReader.cpp.o: in
function `llvm::BitstreamCursor::advance(unsigned int)':

Adding "`/opt/llvm/bin/llvm-config --libs` -L/opt/llvm/lib" solved the
linking issue ... . that worked on the command line, but how to pass that
to "make" or (Makefile) or cmake?

Thanks,
Jeff

My best.,
Jeff