intel/compute-runtime #237: SYCL: Segmentation Fault

cagnulein, 2019-12-02:
I have this little project https://github.com/cagnulein/sycl-benchmarks/commit/df43b1246c8dce52f6bec3c3fe22dbd18f11e8fd that creates a segmentation fault.

<pre>
Thread 1 "rotate_usm.gpu" received signal SIGSEGV, Segmentation fault.
0x00007ffff5e649cb in ?? () from /usr/local/lib/intel-opencl/libigdrcl.so
(gdb) bt
#0  0x00007ffff5e649cb in ?? () from /usr/local/lib/intel-opencl/libigdrcl.so
#1  0x00007ffff5e65162 in ?? () from /usr/local/lib/intel-opencl/libigdrcl.so
#2  0x00007ffff5e047a6 in ?? () from /usr/local/lib/intel-opencl/libigdrcl.so
#3  0x00007ffff702ccc9 in cl::sycl::detail::usm::USMDispatcher::setKernelArgMemPointer(_pi_kernel*, unsigned int, void const*) ()
   from /home/sysele/sycl/build/lib/libsycl.so
#4  0x00007ffff7019dde in cl::sycl::detail::ExecCGCommand::enqueueImp() () from /home/sysele/sycl/build/lib/libsycl.so
#5  0x00007ffff70117c1 in cl::sycl::detail::Command::enqueue(cl::sycl::detail::EnqueueResultT&, cl::sycl::detail::BlockingT) ()
   from /home/sysele/sycl/build/lib/libsycl.so
#6  0x00007ffff701bc43 in cl::sycl::detail::Scheduler::addCG(std::unique_ptr >, std::shared_ptr) () from /home/sysele/sycl/build/lib/libsycl.so
#7  0x0000000000406f12 in cl::sycl::handler::finalize (this=0x7fffffffd2e8)
    at /home/sysele/sycl/build/lib/clang/10.0.0/include/CL/sycl/handler.hpp:422
#8  0x0000000000403aa7 in cl::sycl::detail::queue_impl::submit_impl (this=0x7b5830, cgf=..., self=...)
    at /home/sysele/sycl/build/lib/clang/10.0.0/include/CL/sycl/detail/queue_impl.hpp:240
#9  0x0000000000403853 in cl::sycl::detail::queue_impl::submit (this=0x7b5830, cgf=..., Python Exception  There is no member named _M_use_count.: 
self=)
    at /home/sysele/sycl/build/lib/clang/10.0.0/include/CL/sycl/detail/queue_impl.hpp:114
#10 0x000000000040357c in cl::sycl::queue::submit (this=0x7fffffffd9c8, cgf=...)
    at /home/sysele/sycl/build/lib/clang/10.0.0/include/CL/sycl/queue.hpp:87
#11 0x000000000040345e in main (argc=1, argv=0x7fffffffdad8) at rotate_usm.cpp:144
</pre>

The strange thing is if i reduce the buffer (20x40 bytes instead of 20000x40000 bytes) i don't have the segfault anymore.

What do you think?
@romanovvlad thinks that could be a issue in compute-runtime

Related topic here: https://github.com/intel/llvm/issues/887#issuecomment-560151562

JacekDanecki, 2019-12-19:
I've reproduced this issue. As I can see app segfaults for IMAGE_HEIGHT>24759.

JacekDanecki, 2019-12-19:
As I can see the problem is in your application (typo). With below patch application works without segfaults
```
@@ -129,7 +73,7 @@ int main(int argc, char *argv[]) {
        return 0;
     }
 
-    unsigned char** new_image = (unsigned char**)malloc_shared(DestBitmapWidth * sizeof(unsigned char*), myQueue.get_device(), myQueue.get_context());
+    unsigned char** new_image = (unsigned char**)malloc_shared(DestBitmapHeight * sizeof(unsigned char*), myQueue.get_device(), myQueue.get_context());
     for(int i=0; i<DestBitmapHeight; i++)
         new_image[i] = (unsigned char*)malloc_shared(DestBitmapWidth * sizeof(unsigned char), myQueue.get_device(), myQueue.get_context());

```

cagnulein, 2019-12-19:
Thank you @JacekDanecki , getting a segfault gave me the idea that it wasn't a fault of my code, but you're right, it could be a buffer overrun. I will give it a try. Thanks again!

closed 2019-12-19 by cagnulein