intel/compute-runtime #240: FFTGeneratedStockhamAction::compileKernels failed

shuwang1, 2019-12-09:
Hello,

On an [Intel Compute Stick STK2m3W64CC] (https://www.intel.com/content/www/us/en/products/boards-kits/compute-stick/stk2m3w64cc.html), I am developing a multithread (12 threads) application, each thread following the given 1D clFFT example, [https://github.com/clMathLibraries/clFFT](https://github.com/clMathLibraries/clFFT),  but changed the FFT size to fftlen = 16384

fftlen_bit=14

batches = 81,

inplace c2c

1d clFFT.

The clFFT was continuously and repeatedly run with different inputs. After a few runs, it stopped after showing the following messages:

```
Platform found: Intel(R) OpenCL HD Graphics, status: 0 
Device found on the above platform: Intel(R) Gen9 HD Graphics NEO, status: 0
...
FFTGeneratedStockhamAction::compileKernels failed
Segmentation fault (core dumped)
```

I searched the internet and couldn't find much related information. Any clue? Suggestions will be appreciated.

TIA

JacekDanecki, 2019-12-13:
Can you provide reproducer?

MichalMrozek, 2020-01-17:
We are closing due to inaction, feel free to reopen.

closed 2020-01-17 by MichalMrozek