intel/compute-runtime #249: BPlaidBench failed

SuchtyTV, 2020-01-18:
Hi, 

should this test have failed?

```
(plaidml-venv) niclas@MacBook-Pro ~ % plaidbench keras mobilenet

Running 1024 examples with mobilenet, batch size 1, on backend plaid
INFO:plaidml:Opening device "opencl_amd_radeon_pro_5300m_compute_engine.0"
Compiling network... Warming up... Running...
Example finished, elapsed: 0.438s (compile), 10.278s (execution)

-----------------------------------------------------------------------------------------
Network Name         Inference Latency         Time / FPS          
-----------------------------------------------------------------------------------------
mobilenet            10.04 ms                  0.00 ms / 553745.79 fps
Correctness: FAIL, max_error: 80.5865478515625, max_abs_error: 0.48130345344543457, fail_ratio: 0.9
```

Thnak you!

JacekDanecki, 2020-01-20:
Are you using AMD GPU device, not Neo. See your log: 
```
INFO:plaidml:Opening device "opencl_amd_radeon_pro_5300m_compute_engine.0"
```
Log with Neo you can find [here](https://github.com/intel/compute-runtime/issues/95#issuecomment-454776847)

SuchtyTV, 2020-01-23:
Should not I use the AMD - device?

SuchtyTV, 2020-01-23:
Works with Metal:

(plaidml-venv) niclas@MacBook-Pro ~ % plaidbench keras mobilenet

Running 1024 examples with mobilenet, batch size 1, on backend plaid
INFO:plaidml:Opening device "metal_amd_radeon_pro_5300m.0"
Compiling network... Warming up... Running...
Example finished, elapsed: 1.823s (compile), 5.980s (execution)

-----------------------------------------------------------------------------------------
Network Name         Inference Latency         Time / FPS          
-----------------------------------------------------------------------------------------
mobilenet            5.84 ms                   0.00 ms / 1000000000.00 fps
Correctness: PASS, max_error: 1.675534622336272e-05, max_abs_error: 7.674098014831543e-07, fail_ratio: 0.0

alalek, 2020-01-24:
> I use the AMD - device

You should report issues in a proper place.
This tracker is about issues of Intel OpenCL runtime implementation for Intel devices only.

JacekDanecki, 2020-01-24:
Closing this issue as it's not related to Neo. Feel free to open new issue, when you find issues in Neo working with plaidbench.

closed 2020-01-24 by JacekDanecki