intel/compute-runtime #251: copr upgraded to intel-icg-core and makes disapear platform

metal3d, 2020-01-26:
Hi,

I'm using Fedora 31 and uses copr package which provided intel-opencl package. Latest upgrade removed the package to intel-igc-core et intel-icg-opencl. Now, I've got no platform with clinfo:

```
$ clinfo --list
Number of platforms                               0
```

And LuxRender doesn't see my card...

Any solution ?

metal3d, 2020-01-26:
OK... actually I was using "allowerasing" and "best" option - it has installed the newest package of intel-igc-* that are not compatible with intel-opencl package.

I only had to install intel-opencl which downgrades igc package and everything is OK.

But that means that intel-opencl package is not sync with IGC packages.

JacekDanecki, 2020-01-27:
Yes, IGC was updated on Friday to 1.0.3151, and Neo will be updated today to 20.03.15346.

JacekDanecki, 2020-01-27:
Neo was updated
```
dnf list intel-opencl
Last metadata expiration check: 0:00:34 ago on Mon Jan 27 08:35:45 2020.
Available Packages
intel-opencl.src                                                     20.03.15346-1.fc31                                                   copr:copr.fedorainfracloud.org:jdanecki:intel-opencl
intel-opencl.x86_64                                                  20.03.15346-1.fc31                                                   copr:copr.fedorainfracloud.org:jdanecki:intel-opencl
```
Closing issue, please reopen or create new one, if you find any problems.

closed 2020-01-27 by JacekDanecki

metal3d, 2020-02-04:
It works, no problem :)

Thanks a lot BTW - I can finally use LuxRender with OpenCL activated on my laptop and make renders 10x faster.