intel/compute-runtime #252: Error while downloading on Ubuntu 18.4

manishagajbe, 2020-01-29:
$sudo add-apt-repository ppa:intel-opencl/intel-opencl
Cannot add PPA: 'ppa:~intel-opencl/ubuntu/intel-opencl'.
ERROR: '~intel-opencl' user or team does not exist.

JacekDanecki, 2020-01-30:
It's working fine
```
# add-apt-repository ppa:intel-opencl/intel-opencl
 Replaces Beignet for Gen8 (Broadwell) and beyond.
Project on github: https://github.com/intel/compute-runtime
Website: https://01.org/compute-runtime
 More info: https://launchpad.net/~intel-opencl/+archive/ubuntu/intel-opencl
Press [ENTER] to continue or Ctrl-c to cancel adding it.

Hit:1 http://ppa.launchpad.net/intel-opencl/intel-opencl/ubuntu bionic InRelease
Hit:2 http://archive.ubuntu.com/ubuntu bionic InRelease                  
Hit:3 http://archive.ubuntu.com/ubuntu bionic-updates InRelease          
Hit:4 http://security.ubuntu.com/ubuntu bionic-security InRelease
Hit:5 http://archive.ubuntu.com/ubuntu bionic-backports InRelease
```
Are you using any proxy on your setup?
If I unset proxy variables I can see output like you
```
# unset http_proxy
# unset https_proxy
# add-apt-repository ppa:intel-opencl/intel-opencl
Cannot add PPA: 'ppa:~intel-opencl/ubuntu/intel-opencl'.
ERROR: '~intel-opencl' user or team does not exist.
```

JacekDanecki, 2020-02-12:
Closing as it looks like configuration issue. Feel free to reopen or create new one if you find new problems.

closed 2020-02-12 by JacekDanecki