intel/compute-runtime #254: Abort at core/os_interface/linux/drm_neo.cpp:199 on Broadwell

mamarley, 2020-02-01:
If I try to use compute-runtime 20.04.15428 on Ubuntu (compiled with gmmlib 19.4.1 and graphics-compiler 1.0.3151) on a laptop with an i7-5600U, any OpenCL application ("clinfo" even shows this behavior) aborts immediately with the following message:

> Abort was called at 199 line in file:
> /build/intel-compute-runtime-20.04.15428/core/os_interface/linux/drm_neo.cpp
> Aborted

I also tried with version 20.02.15268 as shipped by Ubuntu and got the same behavior, though the file indicated in the message is in a different directory.  The problem only seems to happen on Broadwell; I tried Skylake and Coffee Lake and it seems to work normally.

JacekDanecki, 2020-02-03:
What kernel version are you using on Broadwell?
Can you provide output from commands:
```
uname -a
strace -e ioctl clinfo
```

mamarley, 2020-02-03:
I'm currently on 5.5.1, but the issue happened with 5.5.0 as well.

> michael@mamarley-laptop:~$ uname -a
> Linux mamarley-laptop 5.5.1-050501-lowlatency #202002010506 SMP PREEMPT Sat Feb 1 05:07:24 EST 2020 x86_64 x86_64 x86_64 GNU/Linux
> michael@mamarley-laptop:~$ strace -e ioctl clinfo
> ioctl(4, DRM_IOCTL_VERSION, 0x7ffe574b6070) = 0
> ioctl(4, DRM_IOCTL_I915_GETPARAM, 0x7ffe574b7150) = 0
> ioctl(4, DRM_IOCTL_I915_GETPARAM, 0x7ffe574b7150) = 0
> ioctl(4, DRM_IOCTL_I915_GETPARAM, 0x7ffe574b7110) = 0
> ioctl(4, DRM_IOCTL_I915_GETPARAM, 0x7ffe574b7110) = 0
> ioctl(4, DRM_IOCTL_I915_GETPARAM, 0x7ffe574b7150) = 0
> ioctl(4, DRM_IOCTL_I915_GEM_CONTEXT_SETPARAM, 0x7ffe574b7150) = -1 EINVAL (Invalid argument)
> ioctl(4, DRM_IOCTL_I915_GETPARAM, 0x7ffe574b6c70) = 0
> ioctl(4, DRM_IOCTL_I915_GETPARAM, 0x7ffe574b6c70) = 0
> ioctl(4, DRM_IOCTL_I915_GETPARAM, 0x7ffe574b6c70) = 0
> ioctl(4, DRM_IOCTL_I915_GETPARAM, 0x7ffe574b6c70) = 0
> ioctl(4, DRM_IOCTL_I915_GEM_CONTEXT_GETPARAM, 0x7ffe574b6c70) = 0
> ioctl(4, DRM_IOCTL_I915_GETPARAM, 0x7ffe574b69e0) = 0
> ioctl(4, DRM_IOCTL_I915_GEM_CONTEXT_GETPARAM, 0x7ffe574b6c60) = 0
> ioctl(4, DRM_IOCTL_I915_GEM_CONTEXT_GETPARAM, 0x7ffe574b6c80) = 0
> ioctl(4, DRM_IOCTL_I915_GETPARAM, 0x7ffe574b6c50) = 0
> ioctl(4, DRM_IOCTL_I915_GEM_USERPTR, 0x7ffe574b7130) = 0
> ioctl(4, DRM_IOCTL_I915_GEM_CONTEXT_CREATE, 0x7ffe574b6f90) = 0
> ioctl(4, DRM_IOCTL_I915_GEM_CONTEXT_SETPARAM, 0x7ffe574b6f80) = -1 ENODEV (No such device)
> Abort was called at 199 line in file:
> /build/intel-compute-runtime-20.04.15428/core/os_interface/linux/drm_neo.cpp
> --- SIGABRT {si_signo=SIGABRT, si_code=SI_TKILL, si_pid=1690, si_uid=1000} ---
> +++ killed by SIGABRT +++
> Aborted

JacekDanecki, 2020-02-03:
I assume you get -ENODEV because preemption is not supported on Broadwell, and kernel returns such error. See: https://elixir.bootlin.com/linux/latest/source/drivers/gpu/drm/i915/gem/i915_gem_context.c#L485
It looks like Neo code at https://github.com/intel/compute-runtime/blob/master/core/os_interface/linux/drm_neo.cpp#L220 is wrong if we compare it with kernel code: https://elixir.bootlin.com/linux/latest/source/drivers/gpu/drm/i915/gem/i915_gem_context.c#L2343
As a workaround you can set 
```
nonPersistentContextsSupported = false;
```
in function Drm::checkNonPersistentContextsSupport(), see: https://github.com/intel/compute-runtime/blob/master/core/os_interface/linux/drm_neo.cpp#L215, until we provide fix.

JacekDanecki, 2020-02-04:
Commit 4a6e3359f0a4e8f14e2e8fc92d0b52f436700452 removed abort, so this fixes this issue.

closed 2020-02-04 by JacekDanecki