intel/compute-runtime #255: include dependency information in deb

yxliang01, 2020-02-05:
The dependency information can be included in the deb package, so that doing `apt-get build-dep intel-opencl-icd` can work and download the right dependencies.

JacekDanecki, 2020-02-17:
Build dependencies are already in control file, see for example [control file for Ubuntu 18.04](https://github.com/JacekDanecki/neo-specs/blob/launchpad/launchpad/intel-opencl/bionic/debian/control)
I don't see issue with this command on my setup. Do you observe any missing dependency in control file?
```
$ apt-get build-dep intel-opencl-icd
Reading package lists... Done
Picking 'intel-opencl' as source package instead of 'intel-opencl-icd'
Building dependency tree       
Reading state information... Done
The following NEW packages will be installed:
  libigc-dev libigdgmm-dev
0 upgraded, 2 newly installed, 0 to remove and 310 not upgraded.
Need to get 223 kB of archives.
After this operation, 2306 kB of additional disk space will be used.
Do you want to continue? [Y/n] 
```

yxliang01, 2020-02-17:
hmm.. weird. Back then when I tried to do the command, it says no dependency information for the package. I can't access that environment at the moment. Could you @JacekDanecki provide the `apt show intel-opencl-icd` output? I remember the environment was ubuntu 16.04 (if I'm not mistaken), it might be that only the new versions have the dependency information included in the package deb.

JacekDanecki, 2020-02-18:
Output from tests on Ubuntu 16.04
```
$ apt-get build-dep intel-opencl-icd
Reading package lists... Done
Picking 'intel-opencl' as source package instead of 'intel-opencl-icd'
Building dependency tree       
Reading state information... Done
The following NEW packages will be installed:
  libigc-dev libigdgmm-dev
0 upgraded, 2 newly installed, 0 to remove and 124 not upgraded.
Need to get 223 kB of archives.
After this operation, 2305 kB of additional disk space will be used.

$ lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 16.04.6 LTS
Release:        16.04
Codename:       xenial

$ apt show intel-opencl-icd
Package: intel-opencl-icd
Version: 20.06.15619-1~ppa1~xenial2
Priority: optional
Section: libs
Source: intel-opencl
Maintainer: Jacek Danecki <jacek.danecki@intel.com>
Installed-Size: 5039 kB
Provides: opencl-icd
Depends: libc6 (>= 2.17), libgcc1 (>= 1:3.4), libigdgmm11 (>= 19.4.1-1~), libstdc++6 (>= 5.2), libigc (>= 1.0.3289-1~)
Breaks: intel-opencl
Replaces: intel-opencl
Download-Size: 936 kB
APT-Sources: http://ppa.launchpad.net/jdanecki/intel-opencl/ubuntu xenial/main amd64 Packages
Description: Intel graphics compute runtime for OpenCL
```
Dependency information was provided from the beginning of these packages.

JacekDanecki, 2020-03-20:
Closing as this request looks like invalid, please reopen it or create new issue if you think otherwise.

closed 2020-03-20 by JacekDanecki