intel/compute-runtime #257: Driver built with default cmake options will not work for some tools

kangshan0910, 2020-02-18:
On Clear Linux, the compute-runtime is built with default cmake options. But the built driver does not work for oneAPI. After investigation, the default built driver's version is '1.0.0' which is not allowed in oneAPI. I find if adding -DNEO_DRIVER_VERSION="%{version}" can be a workaround. My concern is that why this option is not set by default?

JacekDanecki, 2020-02-18:
It should be fixed in Clear Linux [spec](https://github.com/clearlinux-pkgs/intel-compute-runtime/blob/master/intel-compute-runtime.spec). See Fedora 31 [spec](https://github.com/JacekDanecki/neo-specs/blob/master/specs/fedora/31/neo.spec#L27) as reference.

kangshan0910, 2020-02-19:
Thank you for your reply. Got it. I have added it in Clear Linux [spec](https://github.com/clearlinux-pkgs/intel-compute-runtime/blob/master/intel-compute-runtime.spec).

closed 2020-02-19 by kangshan0910