intel/compute-runtime #270: Error: Failed to load libigdfcl.so.1

jianbinfang, 2020-03-10:
Hey guys, 

I am installing compute-runtime from the source (20.09.15980). Before that, I have successfully installed gmmlib-intel-gmmlib-19.4.1 and intel-graphics-compiler-igc-1.0.3445 in my local directory and export these libraries. But now it reports the following error message when compiling the compute-runtime. I am not sure what the matter is? Do I have to load this very library explicitly? 
--------------------------------------
[ 16%] Generating ../../../../bin/built_ins/x64/gen8/aux_translation_Gen8core.spv, ../../../../bin/built_ins/x64/gen8/aux_translation_Gen8core.bin, ../../../../bin/built_ins/x64/gen8/aux_translation_Gen8core.cpp, ../../../../bin/built_ins/x64/gen8/aux_translation_Gen8core.gen
Error: Failed to load libigdfcl.so.1
shared/source/built_ins/kernels/CMakeFiles/builtins_Gen8core.dir/build.make:77: recipe for target 'bin/built_ins/x64/gen8/aux_translation_Gen8core.spv' failed

jianbinfang, 2020-03-10:
Basically, I am sure that the libraries have been successfully installed in ~/local/ 
├── bin
│   ├── GenX_IR
│   └── iga64
├── include
│   ├── iga
│   ├── igc
│   ├── igdgmm
│   ├── opencl-c-base.h
│   ├── opencl-c.h
│   └── visa
└── lib
    ├── libiga64.so -> libiga64.so.1
    ├── libiga64.so.1 -> libiga64.so.1.0.1
    ├── libiga64.so.1.0.1
    ├── libigc.so -> libigc.so.1
    ├── libigc.so.1 -> libigc.so.1.0.1
    ├── libigc.so.1.0.1
    ├── libigdfcl.so -> libigdfcl.so.1
    ├── libigdfcl.so.1 -> libigdfcl.so.1.0.1
    ├── libigdfcl.so.1.0.1
    ├── libigdgmm.so -> libigdgmm.so.11
    ├── libigdgmm.so.11 -> libigdgmm.so.11.1.0
    ├── libigdgmm.so.11.1.0
    ├── libopencl-clang.so.9
    └── pkgconfig

JacekDanecki, 2020-03-10:
If you run "make VERBOSE=1", you will see this step is setting LD_LIBRARY_PATH to directory bin in your build directory, for example:
```
[ 26%] Generating ../../../../bin/built_ins/x64/gen8/aux_translation_Gen8core.spv, ../../../../bin/built_ins/x64/gen8/aux_translation_Gen8core.bin, ../../../../bin/built_ins/x64/gen8/aux_translation_Gen8core.cpp, ../../../../bin/built_ins/x64/gen8/aux_translation_Gen8core.gen
cd /media/data/sources/intel/github/shared/source/built_ins/kernels && LD_LIBRARY_PATH=/media/data/sources/intel/github/build/gcc/bin /media/data/sources/intel/github/build/gcc/bin/ocloc -q -file aux_translation.builtin_kernel -device bdw -64 -out_dir /media/data/sources/intel/github/build/gcc/bin/built_ins/x64/gen8 -cpp_file -options -cl-kernel-arg-info
Error: Failed to load libigdfcl.so.1
```
So, as you have installed gmmlib and igc in not standard directories, so linker can't find them, you can link libraries to bin directory in your build directory.
You can also add your local directory ~/local to linker cache by adding file local.conf (containing location of your local directory in user home directory) to directory /etc/ld.so.conf.d

JacekDanecki, 2020-03-10:
You can also modify igc-opencl.pc file so prefix variable will point at directory containing igc headers and libraries.
For example
```
prefix=/tmp/igc
exec_prefix=${prefix}
includedir=${prefix}/include/igc
libdir=${exec_prefix}/lib64
```
With below patch
```
diff --git a/CMakeLists.txt b/CMakeLists.txt
index ff8a5408d8..a78d6de316 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -304,6 +304,7 @@ if(NOT IGC__IGC_TARGETS)
             if(DEFINED IGC_DIR AND IS_DIRECTORY "${IGC_DIR}/${__tmp_LIBDIR}/pkgconfig/")
                 string(REPLACE "${NEO__IGC_INCLUDEDIR}" "${IGC_DIR}/include/igc" NEO__IGC_INCLUDE_DIRS "${NEO__IGC_INCLUDE_DIRS}")
             endif()
+            set(NEO__IGC_LIBRARY_PATH ${NEO__IGC_LIBDIR})
 
             set(NEO__IGC_INCLUDE_DIR ${NEO__IGC_INCLUDE_DIRS})
             message(STATUS "IGC include dirs: ${NEO__IGC_INCLUDE_DIR}")
```
and running cmake with PKG_CONFIG_PATH set to directory containing igc-opencl.pc file, for example:
```
export PKG_CONFIG_PATH=/tmp/igc ; cmake ..
```
LD_LIBRARY_PATH variable will be set correctly:
```
[ 26%] Generating ../../../../bin/built_ins/x64/gen8/aux_translation_Gen8core.spv, ../../../../bin/built_ins/x64/gen8/aux_translation_Gen8core.bin, ../../../../bin/built_ins/x64/gen8/aux_translation_Gen8core.cpp, ../../../../bin/built_ins/x64/gen8/aux_translation_Gen8core.gen
cd /media/data/sources/intel/github/shared/source/built_ins/kernels && LD_LIBRARY_PATH=/tmp/igc/lib64:/media/data/sources/intel/github/build/gcc/bin /media/data/sources/intel/github/build/gcc/bin/ocloc -q -file aux_translation.builtin_kernel -device bdw -64 -out_dir /media/data/sources/intel/github/build/gcc/bin/built_ins/x64/gen8 -cpp_file -options -cl-kernel-arg-info
```
I'll merge above patch soon.

JacekDanecki, 2020-03-10:
New PR #271 with fix was created. Please confirm it works on your setup.

jianbinfang, 2020-03-11:
> New PR #271 with fix was created. Please confirm it works on your setup.

Thank Jacek, your patch solved the issue.

jianbinfang, 2020-03-11:
I met with a similar issue when doing the unit test (very close to finalizing the making process). Do I have to modify the opencl/test/unit_test/aub_tests/CMakeLists.txt file? 

[ 93%] Built target run_unit_tests
[ 93%] Linking CXX executable ../../../../bin/igdrcl_aub_tests
[ 96%] Built target igdrcl_aub_tests
Creating output directory for bdw AUBs generation...
[ 96%] Built target bdw_aub_tests
re-creating working directory for BDW 1x3x8 AUBs generation...
Running AUB generation for BDW 1x3x8 in /home/jianbin/work/code/intel/compute-runtime/build/bin/bdw_aub
/home/jianbin/work/code/intel/compute-runtime/build/bin/igdrcl_aub_tests: error while loading shared libraries: libigdgmm.so.11: cannot open shared object file: No such file or directory
opencl/test/unit_test/aub_tests/CMakeFiles/run_bdw_aub_tests.dir/build.make:61: recipe for target 'run_bdw_aub_tests' failed
make[2]: *** [run_bdw_aub_tests] Error 127
CMakeFiles/Makefile2:6255: recipe for target 'opencl/test/unit_test/aub_tests/CMakeFiles/run_bdw_aub_tests.dir/all' failed
make[1]: *** [opencl/test/unit_test/aub_tests/CMakeFiles/run_bdw_aub_tests.dir/all] Error 2
Makefile:151: recipe for target 'all' failed
make: *** [all] Error 2

jianbinfang, 2020-03-11:
Basically, when checking the shared library dependency,  it listed the following dependencies. We note it can actually find the related gmm library. Then what is the matter? 

$ ldd ~/work/code/intel/compute-runtime/build/bin/igdrcl_aub_tests 
	linux-vdso.so.1 (0x00007fffb8d67000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007f9697877000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f9697658000)
	libigdgmm.so.11 => ~/local/lib/libigdgmm.so.11 (0x00007f96973e8000)
	libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f969705f000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f9696cc1000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f9696aa9000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f96966b8000)
	/lib64/ld-linux-x86-64.so.2 (0x00007f96ae794000)

JacekDanecki, 2020-03-11:
Can you try with PR #273 ?

jianbinfang, 2020-03-12:
> Can you try with PR #273 ?

It does work now, thanks :-)

closed 2020-03-13 by jianbinfang

onlytheworld, 2020-08-07:
@JacekDanecki 

Hello, I came across with a similar issue when building igc from the source (This issue does not occur when I build and install igc first and then build neo):

`[ 85%] Generating ../../../../bin/built_ins/x64/gen8/aux_translation_Gen8core.spv, ../../../../bin/built_ins/x64/gen8/aux_translation_Gen8core.bin, ../../../../bin/built_ins/x64/gen8/aux_translation_Gen8core.gen
cd /home/only/test/workspace1/neo/shared/source/built_ins/kernels && LD_LIBRARY_PATH=/home/only/test/workspace1/build/bin /home/only/test/workspace1/build/bin/ocloc -q -file aux_translation.builtin_kernel -device bdw -64 -out_dir /home/only/test/workspace1/build/bin/built_ins/x64/gen8 -options -cl-kernel-arg-info
shared/source/built_ins/kernels/CMakeFiles/builtins_Gen8core.dir/build.make:181: recipe for target 'bin/built_ins/x64/gen8/aux_translation_Gen8core.spv' failed
make[2]: *** [bin/built_ins/x64/gen8/aux_translation_Gen8core.spv] Error 250
make[2]: Leaving directory '/home/only/test/workspace1/build'
CMakeFiles/Makefile2:37655: recipe for target 'shared/source/built_ins/kernels/CMakeFiles/builtins_Gen8core.dir/all' failed
make[1]: *** [shared/source/built_ins/kernels/CMakeFiles/builtins_Gen8core.dir/all] Error 2
make[1]: Leaving directory '/home/only/test/workspace1/build'
Makefile:170: recipe for target 'all' failed
make: *** [all] Error 2`

Differently, it gives few info about why it failed. I have tried PR #271 and PR #273, but they do not work well. Can you give me some suggestions about it?