intel/compute-runtime #275: OpenCL kernel build error

boxerab, 2020-03-12:
I just updated to the latest neo release on Ubuntu Eoan with latest Linux kernel release.
Now, when I try to build my cl program, I get this error :+1: 

`Error: clBuildProgram failed.. Error code : CL_OUT_OF_HOST_MEMORY`

Project can be found here:

https://github.com/GrokImageCompression/latke

Thanks.

JacekDanecki, 2020-03-19:
Can you provide steps to reproduce this issue, as I can't reproduce this. I've tested both Neo releases 20.09 and 20.10

Neo 20.09.15980
```
dpkg -l intel-opencl-icd
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name             Version                  Architecture Description
+++-================-========================-============-=========================================
ii  intel-opencl-icd 20.09.15980-1~ppa1~eoan1 amd64        Intel graphics compute runtime for OpenCL

./debayer_image edinburgh_bayer.png 
Initializing GPU.
Platform found : Intel(R) Corporation
Selected Platform Vendor : Intel(R) Corporation
Device 0 : Intel(R) Gen9 HD Graphics NEO
Device ID : 0x55811e64a900

Creating program debayerImage.cl from source
Building program debayerImage.cl
Build Options are : -cl-std=CL2.0 -D OPENCL_2_X -I ./  -D TILE_ROWS=5 -D TILE_COLS=32 -D OUTPUT_CHANNELS=4
Created kernel malvar_he_cutler_demosaic
opencl processing time per image = 5.177815 ms
ubuntu19.10:~/github/latke/build$ ./debayer_buffer edinburgh_bayer.png 
Initializing GPU.
Platform found : Intel(R) Corporation
Selected Platform Vendor : Intel(R) Corporation
Device 0 : Intel(R) Gen9 HD Graphics NEO
Device ID : 0x55fd8a8ee920

Creating program debayerBuffer.cl from source
Building program debayerBuffer.cl
Build Options are : -cl-std=CL2.0 -D OPENCL_2_X -I ./  -D TILE_ROWS=5 -D TILE_COLS=32 -D OUTPUT_CHANNELS=4
Created kernel malvar_he_cutler_demosaic
opencl processing time per image = 2.499732 ms
```
Neo 20.10.16087
```
dpkg -l intel-opencl-icd
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name             Version                  Architecture Description
+++-================-========================-============-=========================================
ii  intel-opencl-icd 20.10.16087-1~ppa1~eoan1 amd64        Intel graphics compute runtime for OpenCL

./debayer_image edinburgh_bayer.png 
Initializing GPU.
Platform found : Intel(R) Corporation
Selected Platform Vendor : Intel(R) Corporation
Device 0 : Intel(R) Gen9 HD Graphics NEO
Device ID : 0x55d5751566c0

Creating program debayerImage.cl from source
Building program debayerImage.cl
Build Options are : -cl-std=CL2.0 -D OPENCL_2_X -I ./  -D TILE_ROWS=5 -D TILE_COLS=32 -D OUTPUT_CHANNELS=4
Created kernel malvar_he_cutler_demosaic
opencl processing time per image = 4.313718 ms

./debayer_buffer edinburgh_bayer.png 
Initializing GPU.
Platform found : Intel(R) Corporation
Selected Platform Vendor : Intel(R) Corporation
Device 0 : Intel(R) Gen9 HD Graphics NEO
Device ID : 0x55eae6d3c6e0

Creating program debayerBuffer.cl from source
Building program debayerBuffer.cl
Build Options are : -cl-std=CL2.0 -D OPENCL_2_X -I ./  -D TILE_ROWS=5 -D TILE_COLS=32 -D OUTPUT_CHANNELS=4
Created kernel malvar_he_cutler_demosaic
opencl processing time per image = 5.312624 ms
```

boxerab, 2020-03-19:
Thanks, Jacek. I had an older version of the driver, 19.x. After upgrading to latest, it works fine. Thanks!

boxerab, 2020-03-19:
Actually, with latest icd, I don't even see my GPU in clinfo.

boxerab, 2020-03-19:
This looks like a local issue on my system. Closing

closed 2020-03-19 by boxerab