intel/compute-runtime #276: Wrong number of Slice for Iris Pro Graphics P580

TApplencourt, 2020-03-13:
Hi,

We use a  Xeon E3-1585 v5 equipped with Iris Pro Graphics P580. Our Linux version is 5.4.10-1.E3_v5_patched.el7.x86_64 x86_64

`$cat /sys/kernel/debug/dri/0/i915_sseu_status
SSEU Device Info
  Available Slice Mask: 0007
  Available Slice Total: 3
  Available Subslice Total: 9
  Available Slice0 subslices: 3
  Available Slice1 subslices: 3
  Available Slice2 subslices: 3
  Available EU Total: 72
  Available EU Per Subslice: 8
  Has Pooled EU: no
  Has Slice Power Gating: yes
  Has Subslice Power Gating: no
  Has EU Power Gating: yes
SSEU Device Status
  Enabled Slice Mask: 0007
  Enabled Slice Total: 3
  Enabled Subslice Total: 9
  Enabled Slice0 subslices: 3
  Enabled Slice1 subslices: 3
  Enabled Slice2 subslices: 3
  Enabled EU Total: 72
  Enabled EU Per Subslice: 8
```

But when using L0 or OpenCL (last release using `CL_DEVICE_SLICE_COUNT_INTEL` extension) to query the number of Slice / Sublice, the result is incorrect. 

For example using (https://github.com/alcf-perfengr/alcl/blob/master/L0/tiny_zeinfo.c) give us:
```
$./tiny_zeinfo
>>> Initializing L0 Platform and Device... (0x193a)
Platforn #0: driver_version 4
Device #0 Intel(R) Gen9
   numSlices 2
   numSubslicesPerSlice 4
   numEUsPerSubslice 8
   physicalEUSimdWidth 8
   numThreadsPerEU 7
```
Regards, 

PS: Please note than `clinfo` give us the correct number of ` Max compute units`.

zzdanowicz, 2020-03-27:
What is Device and Revision ID of your device ?

TApplencourt, 2020-03-27:
```
 $clinfo  | grep 'Device Vendor ID'
  Device Vendor ID                                0x8086

$lspci  | head -n2
00:00.0 Host bridge: Intel Corporation Xeon E3-1200 v5/E3-1500 v5/6th Gen Core Processor Host Bridge/DRAM Registers (rev 0a)
00:02.0 Display controller: Intel Corporation Iris Pro Graphics P580 (rev 09)
```

zzdanowicz, 2020-03-27:
Does clinfo displays Device ID? Vendor ID is just Intel value?

TApplencourt, 2020-03-27:
Just Intel Value. But I found the Device ID
```
$cat /sys/class/drm/card0/device/device
0x193a
```
Confirmed by : https://ark.intel.com/content/www/us/en/ark/products/93742/intel-xeon-processor-e3-1585-v5-8m-cache-3-50-ghz.html)

zzdanowicz, 2020-03-27:
This is expected Device ID. We are looking into the problem.

kamdiedrich, 2020-04-09:
Please check with https://github.com/intel/compute-runtime/commit/27f688ea83462f0bd7a8d4a07fd5ba8cbbdf8d70

TApplencourt, 2020-04-09:
```
>>> Initializing L0  Platform and Device...
Platforn #0: driver_version 8
Device #0 Intel(R) Gen9 (0x193a)
   numSlices 3
   numSubslicesPerSlice 3
   numEUsPerSubslice 8
   physicalEUSimdWidth 8
   numThreadsPerEU 7
```

Fixed Thanks a lot!

closed 2020-04-09 by TApplencourt