intel/compute-runtime #277: HUGETLB support required on Linux

stef, 2020-03-14:
I'm trying to build this fine piece of sw on alpine linux, with a bunch of patches (see https://github.com/aports-ugly/aports/tree/master/ugly/compute-runtime) i made it to compile, but tests segfault, specifically:
```
compute-runtime-20.08.15750/build/bin/igdrcl_tests --product tgllp \
        --slices 1 --subslices 6 --eu_per_ss 16 --gtest_catch_exceptions=1 \
        --gtest_repeat=1 --gtest_shuffle --gtest_random_seed=0 \
        --disable_default_listener
```

I tracked down the issue to the following, in https://github.com/intel/compute-runtime/blob/ee5ebde417716c8cacfbc42c0c7cd840339188d9/shared/source/memory_manager/gfx_partition.cpp#L119 down the callstack https://github.com/intel/compute-runtime/blob/master/shared/source/os_interface/linux/os_memory_linux.cpp#L17 is being called, this looks like this:
```
void *OSMemoryLinux::reserveCpuAddressRange(size_t sizeToReserve) {
    return mmapWrapper(0, sizeToReserve, PROT_NONE, MAP_PRIVATE | MAP_ANONYMOUS | MAP_NORESERVE | MAP_HUGETLB, -1, 0);
}
```

Turns out my kernel does not support HUGETLB, which is not noted as a requirement for this fine piece of sw. I recompiled everything after removing the `MAP_HUGETLB` from the above `mmap`. However tests still fail, since in https://github.com/intel/compute-runtime/blob/ee5ebde417716c8cacfbc42c0c7cd840339188d9/shared/source/memory_manager/gfx_partition.cpp#L121 it is checked if the mmaped pointer is aligned to 64KB boundary, which it is not most of the time.

I also patched https://github.com/intel/compute-runtime/blob/ee5ebde417716c8cacfbc42c0c7cd840339188d9/shared/source/memory_manager/gfx_partition.h#L66 to use `MemoryConstants::pageSize` instead of `MemoryConstants::pageSize64k` and recompiled everything, now the tests actually seem to complete except for two (which seem to be testing 64K allocations:

```
Running igdrcl_tests 1x4x8 in aports/ugly/compute-runtime/src/compute-runtime-20.08.15750/build/bin/ehl
cd aports/ugly/compute-runtime/src/compute-runtime-20.08.15750/build/bin && /usr/bin/cmake -E env GTEST_OUTPUT=xml:test_logs/test_details_ehl_1_4_8.xml aports/ugly/compute-runtime/src/compute-runtime-20.08.15750/build/bin/igdrcl_tests --product ehl --slices 1 --subslices 4 --eu_per_ss 8 --gtest_catch_exceptions=1 --gtest_repeat=1 --gtest_shuffle --gtest_random_seed=0 --disable_default_listener
product family: ehl (28)
set timeout to: 45
Iteration: 1. random_seed: 41949

aports/ugly/compute-runtime/src/compute-runtime-20.08.15750/unit_tests/memory_manager/memory_manager_tests.cpp:1134: Failure
Expected equality of these values:
  0u
    Which is: 0
  (uintptr_t)galloc->getGpuAddress() % MemoryConstants::pageSize64k
    Which is: 61440
[  FAILED  ][ EHL ][ 41949 ] OsAgnosticMemoryManager.GivenEnabled64kbPagesWhenHostMemoryAllocationIsCreatedThenAlignedto64KbAllocationIsReturned

aports/ugly/compute-runtime/src/compute-runtime-20.08.15750/unit_tests/memory_manager/memory_manager_allocate_in_preferred_pool_tests.inl:291: Failure
Expected equality of these values:
  0u
    Which is: 0
  allocation->getGpuAddress() & MemoryConstants::page64kMask
    Which is: 61440
[  FAILED  ][ EHL ][ 41949 ] MemoryManagerTest.givenEnabled64kbPagesWhenGraphicsMemoryMustBeHostMemoryAndIsAllocatedWithNullptrForBufferThen64kbAllocationIsReturned

=====================
==   ULTs FAILED   ==
=====================
Tests run:      14060
Tests passed:   13871
Tests skipped:  187
Tests failed:   2
Tests disabled: 15
 Time elapsed:  1921 ms
=====================
[  FAILED  ][ EHL ][ 41949 ] OsAgnosticMemoryManager.GivenEnabled64kbPagesWhenHostMemoryAllocationIsCreatedThenAlignedto64KbAllocationIsReturned
[  FAILED  ][ EHL ][ 41949 ] MemoryManagerTest.givenEnabled64kbPagesWhenGraphicsMemoryMustBeHostMemoryAndIsAllocatedWithNullptrForBufferThen64kbAllocationIsReturned

make[2]: *** [unit_tests/CMakeFiles/run_ehl_unit_tests.dir/build.make:62: run_ehl_unit_tests] Error 1
make[2]: Leaving directory 'aports/ugly/compute-runtime/src/compute-runtime-20.08.15750/build'
make[1]: *** [CMakeFiles/Makefile2:7079: unit_tests/CMakeFiles/run_ehl_unit_tests.dir/all] Error 2
make[1]: Leaving directory 'aports/ugly/compute-runtime/src/compute-runtime-20.08.15750/build'
make: *** [Makefile:152: all] Error 2

```

So the questions are: 
  - is HUGETLB support really necessary or can we fall back to an map if there is no support? 
 - is the alignment to 64K really necessary? and if no HUGETBL support is available can we simply use 4K alignment, or do we need to allocate a bit more and align "manually" by dropping the preceding memory that is not aligned?

stef, 2020-03-14:
after removing the MAP_HUGETLB flag and reducing the heapGranularity to 4K and recompiling the tests get much further:
```
cd ./aports/ugly/compute-runtime/src/compute-runtime-20.08.15750/build/bin && /usr/bin/cmake -E env GTEST_OUTPUT=xml:test_logs/test_details_tgllp_1_6_16.xml ./aports/ugly/compute-runtime/src/compute-runtime-20.08.15750/build/bin/igdrcl_tests --product tgllp --slices 1 --subslices 6 --eu_per_ss 16 --gtest_catch_exceptions=1 --gtest_repeat=1 --gtest_shuffle --gtest_random_seed=0 --disable_default_listener
product family: tgllp (29)
set timeout to: 45
Iteration: 1. random_seed: 66416

=====================
==   ULTs PASSED   ==
=====================
Tests run:      14058
Tests passed:   14006
Tests skipped:  52
Tests failed:   0
Tests disabled: 15
 Time elapsed:  1856 ms
=====================
make[2]: Leaving directory './aports/ugly/compute-runtime/src/compute-runtime-20.08.15750/build'
[ 93%] Built target run_tgllp_unit_tests
make -f unit_tests/CMakeFiles/run_ehl_unit_tests.dir/build.make unit_tests/CMakeFiles/run_ehl_unit_tests.dir/depend
make[2]: Entering directory './aports/ugly/compute-runtime/src/compute-runtime-20.08.15750/build'
cd ./aports/ugly/compute-runtime/src/compute-runtime-20.08.15750/build && /usr/bin/cmake -E cmake_depends "Unix Makefiles" ./aports/ugly/compute-runtime/src/compute-runtime-20.08.15750 ./aports/ugly/compute-runtime/src/compute-runtime-20.08.15750/unit_tests ./aports/ugly/compute-runtime/src/compute-runtime-20.08.15750/build ./aports/ugly/compute-runtime/src/compute-runtime-20.08.15750/build/unit_tests ./aports/ugly/compute-runtime/src/compute-runtime-20.08.15750/build/unit_tests/CMakeFiles/run_ehl_unit_tests.dir/DependInfo.cmake --color=
make[2]: Leaving directory './aports/ugly/compute-runtime/src/compute-runtime-20.08.15750/build'
make -f unit_tests/CMakeFiles/run_ehl_unit_tests.dir/build.make unit_tests/CMakeFiles/run_ehl_unit_tests.dir/build
make[2]: Entering directory './aports/ugly/compute-runtime/src/compute-runtime-20.08.15750/build'
cd ./aports/ugly/compute-runtime/src/compute-runtime-20.08.15750/build/bin && echo Running igdrcl_tests 1x4x8 in ./aports/ugly/compute-runtime/src/compute-runtime-20.08.15750/build/bin/ehl
Running igdrcl_tests 1x4x8 in ./aports/ugly/compute-runtime/src/compute-runtime-20.08.15750/build/bin/ehl
cd ./aports/ugly/compute-runtime/src/compute-runtime-20.08.15750/build/bin && /usr/bin/cmake -E env GTEST_OUTPUT=xml:test_logs/test_details_ehl_1_4_8.xml ./aports/ugly/compute-runtime/src/compute-runtime-20.08.15750/build/bin/igdrcl_tests --product ehl --slices 1 --subslices 4 --eu_per_ss 8 --gtest_catch_exceptions=1 --gtest_repeat=1 --gtest_shuffle --gtest_random_seed=0 --disable_default_listener
product family: ehl (28)
set timeout to: 45
Iteration: 1. random_seed: 68802

=====================
==   ULTs PASSED   ==
=====================
Tests run:      14058
Tests passed:   13871
Tests skipped:  187
Tests failed:   0
Tests disabled: 15
 Time elapsed:  1877 ms
=====================
make[2]: Leaving directory './aports/ugly/compute-runtime/src/compute-runtime-20.08.15750/build'
[ 93%] Built target run_ehl_unit_tests
make -f unit_tests/CMakeFiles/run_skl_unit_tests.dir/build.make unit_tests/CMakeFiles/run_skl_unit_tests.dir/depend
make[2]: Entering directory './aports/ugly/compute-runtime/src/compute-runtime-20.08.15750/build'
cd ./aports/ugly/compute-runtime/src/compute-runtime-20.08.15750/build && /usr/bin/cmake -E cmake_depends "Unix Makefiles" ./aports/ugly/compute-runtime/src/compute-runtime-20.08.15750 ./aports/ugly/compute-runtime/src/compute-runtime-20.08.15750/unit_tests ./aports/ugly/compute-runtime/src/compute-runtime-20.08.15750/build ./aports/ugly/compute-runtime/src/compute-runtime-20.08.15750/build/unit_tests ./aports/ugly/compute-runtime/src/compute-runtime-20.08.15750/build/unit_tests/CMakeFiles/run_skl_unit_tests.dir/DependInfo.cmake --color=
make[2]: Leaving directory './aports/ugly/compute-runtime/src/compute-runtime-20.08.15750/build'
make -f unit_tests/CMakeFiles/run_skl_unit_tests.dir/build.make unit_tests/CMakeFiles/run_skl_unit_tests.dir/build
make[2]: Entering directory './aports/ugly/compute-runtime/src/compute-runtime-20.08.15750/build'
cd ./aports/ugly/compute-runtime/src/compute-runtime-20.08.15750/build/bin && echo Running igdrcl_tests 1x3x8 in ./aports/ugly/compute-runtime/src/compute-runtime-20.08.15750/build/bin/skl
Running igdrcl_tests 1x3x8 in ./aports/ugly/compute-runtime/src/compute-runtime-20.08.15750/build/bin/skl
cd ./aports/ugly/compute-runtime/src/compute-runtime-20.08.15750/build/bin && /usr/bin/cmake -E env GTEST_OUTPUT=xml:test_logs/test_details_skl_1_3_8.xml ./aports/ugly/compute-runtime/src/compute-runtime-20.08.15750/build/bin/igdrcl_tests --product skl --slices 1 --subslices 3 --eu_per_ss 8 --gtest_catch_exceptions=1 --gtest_repeat=1 --gtest_shuffle --gtest_random_seed=0 --disable_default_listener
product family: skl (18)
set timeout to: 45
Iteration: 1. random_seed: 71199

=====================
==   ULTs PASSED   ==
=====================
Tests run:      14058
Tests passed:   14046
Tests skipped:  12
Tests failed:   0
Tests disabled: 15
 Time elapsed:  2069 ms
=====================
make[2]: Leaving directory './aports/ugly/compute-runtime/src/compute-runtime-20.08.15750/build'
[ 93%] Built target run_skl_unit_tests
make -f unit_tests/CMakeFiles/run_lkf_unit_tests.dir/build.make unit_tests/CMakeFiles/run_lkf_unit_tests.dir/depend
[... 233 lines cut ...]

[... cut ...]
    Which is: 1
  hwDeviceIds.size()
    Which is: 0
make[2]: *** [unit_tests/CMakeFiles/run_unit_tests.dir/build.make:58: run_unit_tests] Segmentation fault (core dumped)
make[2]: Leaving directory './aports/ugly/compute-runtime/src/compute-runtime-20.08.15750/build'
make[1]: *** [CMakeFiles/Makefile2:7613: unit_tests/CMakeFiles/run_unit_tests.dir/all] Error 2
make[1]: Leaving directory './aports/ugly/compute-runtime/src/compute-runtime-20.08.15750/build'
make: *** [Makefile:152: all] Error 2
```

stef, 2020-03-14:
since tests are not deterministic, they instead of the above, they fail at this location sometimes:
```
[ 94%] Built target neo_page_fault_manager_tests
make -f unit_tests/page_fault_manager_tests/CMakeFiles/run_page_fault_manager_tests.dir/build.make unit_tests/page_fault_manager_tests/CMakeFiles/run_page_fault_manager_tests.dir/depend
make[2]: Entering directory './aports/ugly/compute-runtime/src/compute-runtime-20.08.15750/build'
cd ./aports/ugly/compute-runtime/src/compute-runtime-20.08.15750/build && /usr/bin/cmake -E cmake_depends "Unix Makefiles" ./aports/ugly/compute-runtime/src/compute-runtime-20.08.15750 ./aports/ugly/compute-runtime/src/compute-runtime-20.08.15750/unit_tests/page_fault_manager_tests ./aports/ugly/compute-runtime/src/compute-runtime-20.08.15750/build ./aports/ugly/compute-runtime/src/compute-runtime-20.08.15750/build/unit_tests/page_fault_manager_tests ./aports/ugly/compute-runtime/src/compute-runtime-20.08.15750/build/unit_tests/page_fault_manager_tests/CMakeFiles/run_page_fault_manager_tests.dir/DependInfo.cmake --color=
make[2]: Leaving directory './aports/ugly/compute-runtime/src/compute-runtime-20.08.15750/build'
make -f unit_tests/page_fault_manager_tests/CMakeFiles/run_page_fault_manager_tests.dir/build.make unit_tests/page_fault_manager_tests/CMakeFiles/run_page_fault_manager_tests.dir/build
make[2]: Entering directory './aports/ugly/compute-runtime/src/compute-runtime-20.08.15750/build'
cd ./aports/ugly/compute-runtime/src/compute-runtime-20.08.15750/build/bin && echo Target Directory is: ./aports/ugly/compute-runtime/src/compute-runtime-20.08.15750/build/bin
Target Directory is: ./aports/ugly/compute-runtime/src/compute-runtime-20.08.15750/build/bin
cd ./aports/ugly/compute-runtime/src/compute-runtime-20.08.15750/build/bin && echo Running page fault manager tests
Running page fault manager tests
cd ./aports/ugly/compute-runtime/src/compute-runtime-20.08.15750/build/bin && ./aports/ugly/compute-runtime/src/compute-runtime-20.08.15750/build/bin/neo_page_fault_manager_tests
set timeout to: 45
Iteration: 1
Iteration: 1. random_seed: 0

./aports/ugly/compute-runtime/src/compute-runtime-20.08.15750/core/unit_tests/page_fault_manager/linux/cpu_page_fault_manager_linux_tests.cpp:23: Failure
Value of: pageFaultManager->handlerInvoked
  Actual: false
Expected: true
[  FAILED  ][ SKL ][ 0 ] PageFaultManagerLinuxTest.whenPageFaultIsRaisedThenHandlerIsInvoked
make[2]: *** [unit_tests/page_fault_manager_tests/CMakeFiles/run_page_fault_manager_tests.dir/build.make:63: run_page_fault_manager_tests] Segmentation fault (core dumped)
make[2]: Leaving directory './aports/ugly/compute-runtime/src/compute-runtime-20.08.15750/build'
make[1]: *** [CMakeFiles/Makefile2:8732: unit_tests/page_fault_manager_tests/CMakeFiles/run_page_fault_manager_tests.dir/all] Error 2
make[1]: Leaving directory './aports/ugly/compute-runtime/src/compute-runtime-20.08.15750/build'
make: *** [Makefile:152: all] Error 2
```

MichalMrozek, 2020-03-16:
Hello,

HUGETLB is not needed, we just need to handle proper reservation in CPU Virtual Address range.
This is the goal of this function, we need to carve some CPU Virtual Address range so there willl be no CPU addresses causing collisions.

64KB Alignment is needed, so if allocator doesn’t return that by default, we need to manually align.

stef, 2020-03-16:
thanks i'll do a manual alignment to 64K, let's hope that the failing tests above will pass then (i assuem at least he pagefaultmgr tests will succeed). thanks!

stef, 2020-03-20:
i changed the code to this, and with this the related tests do not fail anymore: 
```c++
void *OSMemoryLinux::reserveCpuAddressRange(size_t sizeToReserve) {
    void *ptr = mmapWrapper(0, sizeToReserve, PROT_NONE, MAP_PRIVATE | MAP_ANONYMOUS | MAP_NORESERVE | MAP_HUGETLB, -1, 0);
    if((long)ptr!=-1) return ptr;
    // probably does not support MAP_HUGETLB, lets try without
    ptr = mmapWrapper(0, sizeToReserve+MemoryConstants::pageSize64k, PROT_NONE, MAP_PRIVATE | MAP_ANONYMOUS | MAP_NORESERVE, -1, 0);
       return NULL;
    }
    // force 64K alignment
    size_t offset = MemoryConstants::pageSize64k - ((uintptr_t) ptr) & (MemoryConstants::pageSize64k - 1);
    // unmap prefixed padding
    if(offset) {
       if(munmap(ptr, offset)!=0) {
          return NULL;
       }
    }
    // adjust the ptr to the aligned address
    ptr = (void*) ((uintptr_t)ptr + offset);
    // if 0 != sizeToReserve mod pageSized, there is some slack, thus
    // we cannot unmap that page, but only the one after it
    size_t slack = MemoryConstants::pageSize - (sizeToReserve % MemoryConstants::pageSize);
    // munmap excess trailing padding
    if ((MemoryConstants::pageSize64k - offset - slack) &&
          munmap((void*) ((uintptr_t)ptr+sizeToReserve+slack), MemoryConstants::pageSize64k - offset - slack)!=0) {
       return NULL;
    }
    return ptr;
}
```

stef, 2020-03-20:
i still get the unit test failures mentioned in https://github.com/intel/compute-runtime/issues/277#issuecomment-599143695 and in https://github.com/intel/compute-runtime/issues/277#issuecomment-599143523 the latter one i get randomly, the former i get quite reliably. shall i open new issues for those, or shall we handle them here in this issue?

MichalMrozek, 2020-04-01:
Starting from commit https://github.com/intel/compute-runtime/commit/2ac968e6c21b830957bd7a7abdfb898b6042c8d2 HUGE_TLB is no longer required

stef, 2020-04-01:
great! looking forward to testing that when there is a release. thanks a lot!

AdamCetnerowski, 2020-04-16:
@stef - did you have a chance to check this? If it looks good, please close the issue

stef, 2020-04-16:
Hey Adam, is the only patch necessary 2ac968e? if so, i can try to patch it against compute-runtime-20.08.15750 - otherwise i think i have to wait until there is a release, we normally do not package stuff from git with only a hash but no release.

AdamCetnerowski, 2020-04-16:
It's included in 20.14.16441

stef, 2020-04-16:
aaah, great, i'll have a look.

JacekDanecki, 2020-05-15:
Closing as fix was provided. Feel free to create new issue when you find any problems.

closed 2020-05-15 by JacekDanecki