intel/compute-runtime #279: installing the LevelZero driver with a version-less symlink

fwyzard, 2020-03-17:
Currently, on Linux, the Level Zero driver is installed to `CMAKE_INSTALL_PREFIX/lib64/libze_intel_gpu.so.0.8.0` and sym-linked to `CMAKE_INSTALL_PREFIX/lib64/libze_intel_gpu.so.0.8`.

Would it be possible to add a version-less symlink like `CMAKE_INSTALL_PREFIX/lib64/libze_intel_gpu.so` as well ?

This could help the Level Zero loader find the driver with a hard-coded library name.

fwyzard, 2020-03-17:
See https://github.com/oneapi-src/level-zero/issues/1.

fwyzard, 2020-03-17:
The change would actually be very simple:
```patch
diff --git a/level_zero/CMakeLists.txt b/level_zero/CMakeLists.txt
index e15ae9707..d547973ea 100644
--- a/level_zero/CMakeLists.txt
+++ b/level_zero/CMakeLists.txt
@@ -353,7 +353,6 @@ if(UNIX)
         LIBRARY
             DESTINATION ${CMAKE_INSTALL_LIBDIR}
             COMPONENT ${PROJECT_NAME}
-            NAMELINK_SKIP
     )
 else()
     install(TARGETS ${TARGET_NAME_L0} RUNTIME
```

Let me know if this makes sense, and if you would like a pull request.

ravindra-ganapathi, 2020-03-17:
The loader's patches are behind driver's, but you should see soon the loader updated to work with level zero driver v0.8. As I responded on loader repo, we will update the readme to make it easier to build from source.

Thanks
Ravi

rwmcguir, 2020-03-17:
> Would it be possible to add a version-less symlink like `CMAKE_INSTALL_PREFIX/lib64/libze_intel_gpu.so` as well ?
> 
> This could help the Level Zero loader find the driver with a hard-coded library name.

.so symlinks are standardized to be installed by -dev packages, and currently the intel-level-zero-gpu does not ship a -dev package. Creating one here, would be asking for package conflicts down the road should this package ever need a -dev package. Both normally linked libraries and libraries that dlopen() should only use SONAME to ensure SEMVER style API/ABI compatibilty. The fix for this today is easy by simply checking out a slightly older version that uses the v0.4, <cough> which should be tagged appropriately, but is missing currently. The previous NEO tag should be fine.

Some background, today since the Level Zero specification is not at v1.0 and is pre-release, the SONAME is including the MINOR so avoid runtime issues that are fluxing to quickly. After v1.0 is released the SONAME will be switched to only using MAJOR and normal SEMVER will resume, hopefully very soon, think of this as alpha. So these breakages will become far less likely. Also community adoption of perhaps beyond simple SEMVER compatibility perhaps will be a thing.

Today this is not ideal, but the kinks in the release process are being worked out, this is just first public exposure. compute-runtime hasn't previously needed level-zero as a dependency so the association is being mapped.

fwyzard, 2020-03-17:
Understood, thanks.

> Some background, today since the Level Zero specification is not at v1.0 and is pre-release, the SONAME is including the MINOR so avoid runtime issues that are fluxing to quickly. After v1.0 is released the SONAME will be switched to only using MAJOR and normal SEMVER will resume

I see - yes, that will likely be good enough to avoid a similar issue in the long term.

closed 2020-03-17 by fwyzard