intel/compute-runtime #281: Uninstall/remove NEO opencl graphics runtime

Mushtaq-BGA, 2020-03-18:
Hi,

I followed below steps to install NEO runtime

add-apt-repository ppa:intel-opencl/intel-opencl
apt-get update
apt-get install intel-opencl-icd

It installs successfully, now if i have to uninstall or remove neo runtime i am doing the below way 
**apt-get remove intel-opencl-icd**

is this the right way ?, 
My goal is to disable GPU runtime and enable whenever it is required.

i did try once and got messed up.

Need help on this
Thanks

JacekDanecki, 2020-03-18:
You can use apt-get or apt commands to remove Neo. "remove" apt parameter will remove only Neo binaries, and leave configuration file in /etc/OpenCL/vendors/intel.icd. If you want to remove also configuration file use "purge" parameter.
```
# apt purge intel-opencl-icd
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following packages will be REMOVED:
  intel-opencl-icd*
0 upgraded, 0 newly installed, 1 to remove and 72 not upgraded.
After this operation, 5637 kB disk space will be freed.
Do you want to continue? [Y/n] y
(Reading database ... 25161 files and directories currently installed.)
Removing intel-opencl-icd (20.10.16087-1~ppa1~bionic1) ...
(Reading database ... 25154 files and directories currently installed.)
Purging configuration files for intel-opencl-icd (20.10.16087-1~ppa1~bionic1) ...
```
If you want to disable temporary Neo, you can rename /etc/OpenCL/vendors/intel.icd to file without .icd extension, for example:
```
mv /etc/OpenCL/vendors/intel.icd /etc/OpenCL/vendors/intel.icd.disabled
```
ocl-icd by default is looking only for *.icd files in /etc/OpenCL/vendors directory, see https://www.systutorials.com/docs/linux/man/7-libOpenCL for more details.

And when you need Neo again, you can restore it by command:
```
mv /etc/OpenCL/vendors/intel.icd.disabled /etc/OpenCL/vendors/intel.icd
```

Mushtaq-BGA, 2020-04-01:
Hi Jacek, 

Thanks for the answer! you can close this issue.

closed 2020-04-01 by JacekDanecki