intel/compute-runtime #287: Level zero for Broadwell GPU

jinz2014, 2020-03-31:
Is it possible that level zero can support Broadwell GPUs ?

I got segfault when running a program as shown below.

Thanks

Thread 1 "a.out" received signal SIGSEGV, Segmentation fault.
0x0000000000000000 in ?? ()
(gdb) bt
#0 0x0000000000000000 in ?? ()
#1 0x00007ffff6573b94 in L0::Kernel::create(unsigned int, L0::Module*, _ze_kernel_desc_t const*, _ze_result_t*) ()
from /usr/local/lib/libze_intel_gpu.so.0.8
#2 0x00007ffff6575cf6 in L0::ModuleImp::createKernel(_ze_kernel_desc_t const*, _ze_kernel_handle_t**) ()
from /usr/local/lib/libze_intel_gpu.so.0.8
#3 0x00007ffff6568207 in L0::BuiltinFunctionsLibImpl::loadBuiltIn(unsigned int, char const*) ()
from /usr/local/lib/libze_intel_gpu.so.0.8
#4 0x00007ffff656832b in L0::BuiltinFunctionsLibImpl::initFunctions() () from /usr/local/lib/libze_intel_gpu.so.0.8
#5 0x00007ffff656bb9d in L0::Device::create(L0::DriverHandle*, NEO::Device*) () from /usr/local/lib/libze_intel_gpu.so.0.8
#6 0x00007ffff656cc0b in L0::DriverHandleImp::initialize(std::vector<std::unique_ptr<NEO::Device, std::default_deleteNEO::Device >, std::allocator<std::unique_ptr<NEO::Device, std::default_deleteNEO::Device > > >) () from /usr/local/lib/libze_intel_gpu.so.0.8
#7 0x00007ffff656cd9b in L0::DriverHandle::create(std::vector<std::unique_ptr<NEO::Device, std::default_deleteNEO::Device >, std::allocator<std::unique_ptr<NEO::Device, std::default_deleteNEO::Device > > >) () from /usr/local/lib/libze_intel_gpu.so.0.8
#8 0x00007ffff656da0a in L0::DriverImp::initialize(bool*) () from /usr/local/lib/libze_intel_gpu.so.0.8
#9 0x00007ffff656d60e in ?? () from /usr/local/lib/libze_intel_gpu.so.0.8
#10 0x00007ffff621c827 in __pthread_once_slow () from /lib/x86_64-linux-gnu/libpthread.so.0
#11 0x00007ffff656d8f0 in L0::init(_ze_init_flag_t) () from /usr/local/lib/libze_intel_gpu.so.0.8
#12 0x00007ffff7b2df6a in zeInit () from /usr/local/lib/libze_loader.so.0.91
#13 0x0000555555557285 in main ()

ravindra-ganapathi, 2020-03-31:
Level Zero supports Gen9 and later generation only as documented in our release page, we don't expect to add support for older generations

jinz2014, 2020-03-31:
Is it possible that you could explain how a user can add level zero support for older generations ?

bfliflet, 2020-03-31:
The patterns for BDW/Gen8 support exist in the opencl portions of the compute runtime.  The coding patterns to enable new platforms/SKUs in Level Zero are the roughly the same as SKL/Gen9.  I would start by analyzing the CMake files for **GEN9** and **SKL** while creating equivalent patches for **GEN8** / **BDW**. Then, add the equivalent .h/.inl/.cpp files that contain the bdw/gen8 file suffixes.  The effort to enable is much less than validation.  We're exploring options internally to integrate such unofficial support via community applied patches.  If you have specific questions, I'll try and answer them to enable you.

The specific bug you're hitting is related to not having precompiled built-ins available for your device.  These are optimized kernels that are used for memory copies, etc.  Once you add product support above, the built-ins should get compiled along with the rest of the runtime driver.

jinz2014, 2020-03-31:
Thanks again for your answer and explanation. I will try to learn about Level Zero for GEN9/SKL.

closed 2020-03-31 by jinz2014