intel/compute-runtime #288: oneAPI: ze_device_kernel_properties_t.dp4aSupported contains garbage

maleadt, 2020-04-02:
```
ze_device_kernel_properties_t kernel_properties;
kernel_properties.version = ZE_DEVICE_KERNEL_PROPERTIES_VERSION_CURRENT;
kernel_properties.dp4aSupported = 100;
zeDeviceGetKernelProperties(allDevices[d], &kernel_properties);

printf("dp4aSupported: %i\n", kernel_properties.dp4aSupported);
```

This retains the 100 value. Setting it to zero first (this field, or the entire struct) results in a 0 value afterwards, so I assume the driver isn't touching this value. As per the docs, it's valid isn't optional though.

Got sent here by https://github.com/oneapi-src/level-zero/issues/8. Full repro:

```c
#include <level_zero/ze_api.h>
#include <stdio.h>
#include <stdlib.h>

int main() {
    // Initialize the driver
    zeInit(ZE_INIT_FLAG_NONE);

    // Discover all the driver instances
    uint32_t driverCount = 0;
    zeDriverGet(&driverCount, NULL);

    ze_driver_handle_t* allDrivers = malloc(driverCount * sizeof(ze_driver_handle_t));
    zeDriverGet(&driverCount, allDrivers);

    // Find a driver instance with a GPU device
    for(int i = 0; i < driverCount; ++i) {
        uint32_t deviceCount = 0;
        zeDeviceGet(allDrivers[i], &deviceCount, NULL);

        ze_device_handle_t* allDevices = malloc(deviceCount * sizeof(ze_device_handle_t));
        zeDeviceGet(allDrivers[i], &deviceCount, allDevices);

        for(int d = 0; d < deviceCount; ++d) {
            ze_device_properties_t device_properties;
            zeDeviceGetProperties(allDevices[d], &device_properties);

            ze_device_kernel_properties_t kernel_properties;
            kernel_properties.version = ZE_DEVICE_KERNEL_PROPERTIES_VERSION_CURRENT;
            kernel_properties.dp4aSupported = 100;
            zeDeviceGetKernelProperties(allDevices[d], &kernel_properties);

            printf("dp4aSupported: %i\n", kernel_properties.dp4aSupported);
        }

        free(allDevices);
    }

    free(allDrivers);

    return 0;
}

```

jandres742, 2020-04-02:
That's correct. Looks like we are not setting it in the driver. Fix WIP.

MaciejPlewka, 2020-04-03:
Fixed with this commit 
https://github.com/intel/compute-runtime/commit/08546b65c73dd7c27189984eb95aa3de5d6009b0

JacekDanecki, 2020-04-17:
Closing as it was fixed, please create new issue if you find any new problems.

closed 2020-04-17 by JacekDanecki