intel/compute-runtime #290: How to load additional data while using intel_sub_group_block_read?

OliverScherf, 2020-04-02:
Hello,

I have a kernel that reads data from an array. I implemented a solution using local memory to reduce global memory reads. Now I want to tune the kernel further and use the `intel_sub_group_block_read` functions.
My goal is to read 9 * 8 floats (using float8) into an array.

Please correct me if I'm wrong somewhere:
The kernel is compiled in SIMD-8 mode using `__attribute__((intel_reqd_sub_group_size(8)))`, hence a subgroup contains 8 work items. 
```
#define SG_SIZE 8
#define VEC_LEN 8

int idx = get_global_id(0) * 8; // 1 work item should work with 8 floats.
int sid = get_sub_group_local_id();

float8 sg_pixels =  as_float8(intel_sub_group_block_read8((const __global uint *)(&input[idx])); 
float pixels_of_sg[VEC_LEN][SG_SIZE + 1]; // because block_reads return data in a column-major way, we need to declare the array like this

for (int i = 0; i < SG_SIZE; ++i) { // get values of the other work items in the sub group
            float8 tmp = intel_sub_group_shuffle(sg_pixels, i);
            // fill pixels_of_sg array
            pixels_of_sg[0][i] tmp.s0;
            pixels_of_sg[1][i] tmp.s1;
            ...
            pixels_of_sg[7][i] tmp.s7;
            // Is this an efficient operation or an expensive procedure?
}
// pixels_of_sg[0][0] - pixels_of_sg[7][7] is filled with data

// at this point, I want to load 1 * 8 pixels into my array, every work-item has to do this
float8 tmp = vload8(0, &input[idx - sid + SG_SIZE]); // read the data right after the last sub group work items
pixels_of_sg[0][SG_SIZE] = tmp.s0;
pixels_of_sg[1][SG_SIZE] = tmp.s1;
...
pixels_of_sg[7][SG_SIZE] = tmp.s1;

// work with 9 * 8 the read values (pixels_of_sg)
```

This procedure feels inefficient, but I can't think of a good another solution. The main problem is the differing layout major (block_read -> column major, vload -> row major).
It would be much easier, if I could stack the values from block_read and the vload8 together, but that doesnt work, am I right?

What other possibilities are there? I was thinking about increasing the sub group size to 16. But as I only need 9 values, 7 threads would be wasted and doing nothing.

Any help is much appreciated.

bashbaug, 2020-04-20:
Hello @OliverScherf,

I find that the easiest way to think about this is in terms of multiple block reads.

First, remember that a scalar block read is equivalent to the following, assuming that the pointer `ptr` is the same for all work-items in the subgroup:

```c
uint value;

// This is equivalent to:
//    value = intel_sub_group_block_read_ui(ptr);
uint index = get_sub_group_local_id();
value = ptr[ index ];
```

For a two-component vector block read, the index value must be incremented by the maximum subgroup size to move from the first component to the second component.  So, a two-component vector block read is equivalent to:

```c
uint2 value;

// This is equivalent to:
//    value = intel_sub_group_block_read_ui2(ptr);
uint index = get_sub_group_local_id();
value.x = ptr[ index ];
index += get_max_sub_group_size();
value.y = ptr[ index ];
```

If you want to read a 9 x 8 block using block reads, I think you can use a similar technique to perform one 8-component block read, then a scalar block read to read the 9th row.  In the snip below, I'm storing the 9 x 8 block in a vec16 variable, but only the first 9 components are used.  You could use an alternative representation if you'd like:

```c
uint16 value;

// Load the first eight rows:
value.lo = intel_sub_group_block_read_ui8(ptr);

// Load the last (9th) row:
value.s8 = intel_sub_group_block_read_ui(ptr + 8 * get_max_sub_group_size());
```

Will this work for you?

OliverScherf, 2020-05-28:
Thanks for your reply.
It looks good to me!

closed 2020-05-28 by OliverScherf