intel/compute-runtime #293: using intel-compute platform within python crashes

hiaselhans, 2020-04-13:
When using intel-compute-runtime in a project with python bindings (using [pybind11](https://github.com/pybind/pybind11) ) instantiating a queue fails at the following line:
https://github.com/intel/compute-runtime/blob/ed6a9e8a5e26583aad5ee6b8adee6f21fca2764c/shared/source/built_ins/built_ins.cpp#L42

here is a minimal reproducer:

```
#include <pybind11/pybind11.h>
#include <pybind11/stl.h>

#ifdef __APPLE__
    #include <OpenCL/cl.hpp>
#else
    #include <CL/cl.hpp>
#endif

namespace py = pybind11;

int check() {
    std::vector<cl::Platform> all_platforms;
    std::cout << "find Platforms" << std::endl;
    cl::Platform::get(&all_platforms);
    std::cout << "found " << all_platforms.size() << "platforms" << std::endl;

    return 0;
}

PYBIND11_MODULE(pyopencl, m) {
    m.doc() = "opencl python module";

    m.def("check", &check, "A function which doesnt do much");
}
```

It does work fine with other vendors platforms and without using pybind11 tough...

hiaselhans, 2020-04-13:
Some Platform Info:

`intel-compute-runtime 20.13.16352-1`

`Linux 5.6.3-arch1-1`

```
clinfo -l
Platform #0: Intel(R) OpenCL HD Graphics
 `-- Device #0: Intel(R) Gen9 HD Graphics NEO
Platform #1: AMD Accelerated Parallel Processing
 `-- Device #0: gfx804
```

hiaselhans, 2020-04-13:
cmake file:

```
CMAKE_MINIMUM_REQUIRED(VERSION 3.3)
project(compute_demo)

find_library(OpenCL REQUIRED)
find_package(pybind11 REQUIRED)
set(PYBIND11_CPP_STANDARD -std=c++14)

pybind11_add_module(pyopencl py.cpp)
target_link_libraries(pyopencl PRIVATE ${OpenCL_LIBRARIES})
```

JacekDanecki, 2020-04-14:
It looks like problem with IGC libraries. I've modified Your CMakeLists.txt, and use:
```
find_package(OpenCL REQUIRED)
```
instead of
```
find_library(OpenCL REQUIRED
```
built your reproducer, and it worked correctly:
```
$ python3
Python 3.7.6 (default, Dec 19 2019, 22:50:01) 
[GCC 9.2.1 20190827 (Red Hat 9.2.1-1)] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import pyopencl
>>> pyopencl.check()
find Platforms
found 1platforms
0
```
I've tested on Fedora 30.

Under Arch as I can see llvm/clang was updated to version 10. Have you recompiled  IGC libraries with newer llvm/clang?
If not current IGC is using llvm/clang 9, for example:
```
arch:~# pacman -Qo /usr/lib/libigc.so.1
/usr/lib/libigc.so.1 is owned by intel-graphics-compiler 1:1.0.3627-1
arch:~# ldd /usr/lib/libigc.so.1
        linux-vdso.so.1 (0x00007ffc0d349000)
        librt.so.1 => /usr/lib/librt.so.1 (0x00007f6b59798000)
        libLLVM-9.so => /usr/lib/libLLVM-9.so (0x00007f6b5494b000)
        libdl.so.2 => /usr/lib/libdl.so.2 (0x00007f6b54946000)
        libstdc++.so.6 => /usr/lib/libstdc++.so.6 (0x00007f6b5475d000)
        libm.so.6 => /usr/lib/libm.so.6 (0x00007f6b54617000)
        libgcc_s.so.1 => /usr/lib/libgcc_s.so.1 (0x00007f6b545fd000)
        libc.so.6 => /usr/lib/libc.so.6 (0x00007f6b54435000)
        /usr/lib64/ld-linux-x86-64.so.2 (0x00007f6b5b3fc000)
        libpthread.so.0 => /usr/lib/libpthread.so.0 (0x00007f6b54413000)
        libffi.so.7 => /usr/lib/../lib/libffi.so.7 (0x00007f6b54407000)
        libedit.so.0 => /usr/lib/../lib/libedit.so.0 (0x00007f6b543cd000)
        libz.so.1 => /usr/lib/../lib/libz.so.1 (0x00007f6b543b3000)
        libncursesw.so.6 => /usr/lib/../lib/libncursesw.so.6 (0x00007f6b54342000)
        libxml2.so.2 => /usr/lib/../lib/libxml2.so.2 (0x00007f6b541da000)
        libicuuc.so.65 => /usr/lib/../lib/libicuuc.so.65 (0x00007f6b53ff8000)
        liblzma.so.5 => /usr/lib/../lib/liblzma.so.5 (0x00007f6b53fd0000)
        libicudata.so.65 => /usr/lib/../lib/libicudata.so.65 (0x00007f6b5251f000)
arch:~# pacman -Qo /usr/lib/libigdfcl.so.1
/usr/lib/libigdfcl.so.1 is owned by intel-graphics-compiler 1:1.0.3627-1
arch:~# ldd /usr/lib/libigdfcl.so.1 | grep -Ei "clang|llvm"
        libLLVM-9.so => /usr/lib/libLLVM-9.so (0x00007f4e7d49b000)
        libopencl-clang.so.9 => /usr/lib/libopencl-clang.so.9 (0x00007f4e7d3b6000)
        libclangBasic.so.9 => /usr/lib/../lib/libclangBasic.so.9 (0x00007f4e7c7b0000)
        libclangFrontend.so.9 => /usr/lib/../lib/libclangFrontend.so.9 (0x00007f4e7c601000)
        libclangFrontendTool.so.9 => /usr/lib/../lib/libclangFrontendTool.so.9 (0x00007f4e7c5f8000)
        libclangSerialization.so.9 => /usr/lib/../lib/libclangSerialization.so.9 (0x00007f4e7c433000)
        libLLVMSPIRVLib.so.9 => /usr/lib/../lib/libLLVMSPIRVLib.so.9 (0x00007f4e7bc33000)
        libclangDriver.so.9 => /usr/lib/../lib/../lib/libclangDriver.so.9 (0x00007f4e7b823000)
        libclangParse.so.9 => /usr/lib/../lib/../lib/libclangParse.so.9 (0x00007f4e7b727000)
        libclangSema.so.9 => /usr/lib/../lib/../lib/libclangSema.so.9 (0x00007f4e7ae83000)
        libclangEdit.so.9 => /usr/lib/../lib/../lib/libclangEdit.so.9 (0x00007f4e7ae73000)
        libclangAST.so.9 => /usr/lib/../lib/../lib/libclangAST.so.9 (0x00007f4e7a9e4000)
        libclangLex.so.9 => /usr/lib/../lib/../lib/libclangLex.so.9 (0x00007f4e7a923000)
        libclangCodeGen.so.9 => /usr/lib/../lib/../lib/libclangCodeGen.so.9 (0x00007f4e7a41b000)
        libclangRewriteFrontend.so.9 => /usr/lib/../lib/../lib/libclangRewriteFrontend.so.9 (0x00007f4e7a39d000)
        libclangARCMigrate.so.9 => /usr/lib/../lib/../lib/libclangARCMigrate.so.9 (0x00007f4e7a091000)
        libclangStaticAnalyzerFrontend.so.9 => /usr/lib/../lib/../lib/libclangStaticAnalyzerFrontend.so.9 (0x00007f4e7a020000)
        libclangAnalysis.so.9 => /usr/lib/../lib/../lib/../lib/libclangAnalysis.so.9 (0x00007f4e78462000)
        libclangASTMatchers.so.9 => /usr/lib/../lib/../lib/../lib/libclangASTMatchers.so.9 (0x00007f4e783d2000)
        libclangRewrite.so.9 => /usr/lib/../lib/../lib/../lib/libclangRewrite.so.9 (0x00007f4e783bf000)
        libclangStaticAnalyzerCheckers.so.9 => /usr/lib/../lib/../lib/../lib/libclangStaticAnalyzerCheckers.so.9 (0x00007f4e77f7a000)
        libclangStaticAnalyzerCore.so.9 => /usr/lib/../lib/../lib/../lib/libclangStaticAnalyzerCore.so.9 (0x00007f4e77d3e000)
        libclangCrossTU.so.9 => /usr/lib/../lib/../lib/../lib/libclangCrossTU.so.9 (0x00007f4e77d2e000)
        libclangIndex.so.9 => /usr/lib/../lib/../lib/../lib/../lib/libclangIndex.so.9 (0x00007f4e77cdb000)
        libclangFormat.so.9 => /usr/lib/../lib/../lib/../lib/../lib/../lib/libclangFormat.so.9 (0x00007f4e77c40000)
        libclangToolingCore.so.9 => /usr/lib/../lib/../lib/../lib/../lib/../lib/libclangToolingCore.so.9 (0x00007f4e77c2a000)
        libclangToolingInclusions.so.9 => /usr/lib/../lib/../lib/../lib/../lib/../lib/../lib/libclangToolingInclusions.so.9 (0x00007f4e77c1d000)

```
I've downgraded llvm/clang with packages:
```
extra/clang 
llvm9-libs
```
so I can use clinfo.
On Arch I've installed such llvm/clang packages:
```
arch:~# pacman -Q | grep -E "llvm|clang"
clang 9.0.1-1
intel-opencl-clang 9.0.1-1
llvm 10.0.0-1
llvm-libs 10.0.0-1
llvm9-libs 9.0.1-1
spirv-llvm-translator 9.0.1.1-1
```
and your reproducer works correctly:
```
arch:~/github/build# python3
Python 3.8.2 (default, Apr  8 2020, 14:31:25) 
[GCC 9.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import pyopencl
>>> pyopencl.check()
find Platforms
found 2platforms
0
```
clinfo output on Arch
```
clinfo -l         
Platform #0: Intel(R) OpenCL HD Graphics
 `-- Device #0: Intel(R) Gen9 HD Graphics NEO
Platform #1: Clover
```

hiaselhans, 2020-04-16:
thanks @JacekDanecki for the help!

changing find_package to find_library didn't change my situation.
and i think there's still clang 9 on my system. 

```
> pacman -Q | grep -E "llvm|clang"
clang 9.0.1-1
intel-opencl-clang 9.0.1-1
llvm-libs 9.0.1-3
spirv-llvm-translator 9.0.1.1-1

```

the thing was that i was using a conda distributed python interpreter.
It did work with the system one.

hiaselhans, 2020-04-16:
so thanks a lot for helping me out! now i'm trying to get it to run with that other python interpreter.
conda python is compiled using gcc 7.3.
i would need to have intel-graphics-compiler installed with anaconda?
I don't know much about the internals of opencl-icd invokation but it seems to me there is no compiler found?

JacekDanecki, 2020-04-16:
Can you see output from:
```
LD_DEBUG=libs python
```
how IGC libraries are loaded using anaconda python?
Maybe you need to set LD_LIBRARY_PATH somewhere in conda configuration?

hiaselhans, 2020-04-18:
it failed with 

`/usr/lib/libopencl-clang.so.9: error: symbol lookup error: undefined symbol: LLVMInitializeAVRTarget, version LLVM_9 (fatal)`

installing intel-compute-runtime within conda did solve the problem for me. It seems conda llvm doesn't play well with system libraries.

JacekDanecki, 2020-04-20:
Closing as issue was resolved. Feel free to open new issue if you find any new problems with Neo.

closed 2020-04-20 by JacekDanecki