intel/compute-runtime #294: zeEventPoolCreate does not accept NULL devices

maleadt, 2020-04-17:
The spec says:

```
zeEventPoolCreate(
    ...
    uint32_t numDevices,                            ///< [in] number of device handles
    ze_device_handle_t* phDevices,                  ///< [in][optional][range(0, numDevices)] array of device handles which
                                                    ///< have visibility to the event pool.
                                                    ///< if nullptr, then event pool is visible to all devices supported by the
                                                    ///< driver instance.
    ...
```

The implementation does not seem to support this, and segfaults on dereferencing this NULL:

https://github.com/intel/compute-runtime/blob/a604eee07babc2d76b1c8d359954227aca6fa65d/level_zero/core/source/driver/driver_handle_imp.cpp#L222-L228

It's also not clear what `numDevices` needs to be in this case.

aravindksg, 2020-04-17:
@maleadt - Thanks for the catch. We are currently testing a fix for this. Shall update once the patch is merged upstream.

aravindksg, 2020-04-24:
@maleadt Fix for this has been posted in https://github.com/intel/compute-runtime/commit/2dbffe8b640caddd3001a293ffa9958efea116e8 

Please retest

AdamCetnerowski, 2020-05-22:
Thank you for reporting the issue. As it was fixed, I am closing this issue. If you still see the same problem, please re-open.

closed 2020-05-22 by AdamCetnerowski

maleadt, 2020-05-26:
Sorry for the delay, yes this works, thanks!