intel/compute-runtime #295: Ubuntu repo renaming

jeffhammond, 2020-04-17:
I could not figure out how to allow Apt to use the Ubuntu repo.  I could not find the word "accept" on the reference man page in the error message, removing the repo failed, and adding the repo again did nothing.

There is a solution, but it takes a while to find, hence it should be documented on https://github.com/intel/compute-runtime/blob/master/opencl/doc/DISTRIBUTIONS.md and https://github.com/intel/compute-runtime/blob/master/level_zero/doc/DISTRIBUTIONS.md so as to minimize the UX disruption.

# Solution
```sh
 apt-get --allow-releaseinfo-change update
```

# Error
```sh
jrhammon@klondike:~$ sudo apt-get update -y
Ign:1 http://archive.canonical.com/ubuntu trusty InRelease
Hit:2 http://us.archive.ubuntu.com/ubuntu bionic InRelease                                
Hit:3 http://archive.canonical.com/ubuntu trusty Release                                                                                  
Hit:4 http://us.archive.ubuntu.com/ubuntu bionic-updates InRelease                                                                        
Hit:5 http://ppa.launchpad.net/graphics-drivers/ppa/ubuntu bionic InRelease                                         
Hit:6 http://security.ubuntu.com/ubuntu bionic-security InRelease                             
Hit:7 http://us.archive.ubuntu.com/ubuntu bionic-backports InRelease                          
Hit:8 http://repo.urz.uni-heidelberg.de/sycl/deb ./bionic InRelease                           
Get:10 http://ppa.launchpad.net/intel-opencl/intel-opencl/ubuntu bionic InRelease [15.9 kB]
Reading package lists... Done                                   
E: Repository 'http://ppa.launchpad.net/intel-opencl/intel-opencl/ubuntu bionic InRelease' changed its 'Label' value from 'Intel(R) Graphics Compute Runtime for OpenCL(TM)' to 'Intel(R) Graphics Compute Runtime'
N: This must be accepted explicitly before updates for this repository can be applied. See apt-secure(8) manpage for details.
jrhammon@klondike:~$ sudo add-apt-repository --remove ppa:intel-opencl/intel-opencl
 The Intel(R) Graphics Compute Runtime for oneAPI Level Zero and OpenCL(TM) Driver is an open source project providing compute API support
(Level Zero, OpenCL) for Intel graphics hardware architectures
(HD Graphics, Xe).

Project on github: https://github.com/intel/compute-runtime
Website: https://01.org/compute-runtime
 More info: https://launchpad.net/~intel-opencl/+archive/ubuntu/intel-opencl
Press [ENTER] to continue or Ctrl-c to cancel removing it.

jrhammon@klondike:~$ sudo add-apt-repository --remove ppa:intel-opencl/intel-opencl
 The Intel(R) Graphics Compute Runtime for oneAPI Level Zero and OpenCL(TM) Driver is an open source project providing compute API support
(Level Zero, OpenCL) for Intel graphics hardware architectures
(HD Graphics, Xe).

Project on github: https://github.com/intel/compute-runtime
Website: https://01.org/compute-runtime
 More info: https://launchpad.net/~intel-opencl/+archive/ubuntu/intel-opencl
Press [ENTER] to continue or Ctrl-c to cancel removing it.

jrhammon@klondike:~$ sudo add-apt-repository --remove ppa:intel-opencl/intel-opencl
 The Intel(R) Graphics Compute Runtime for oneAPI Level Zero and OpenCL(TM) Driver is an open source project providing compute API support
(Level Zero, OpenCL) for Intel graphics hardware architectures
(HD Graphics, Xe).

Project on github: https://github.com/intel/compute-runtime
Website: https://01.org/compute-runtime
 More info: https://launchpad.net/~intel-opencl/+archive/ubuntu/intel-opencl
Press [ENTER] to continue or Ctrl-c to cancel removing it.

```

JacekDanecki, 2020-04-20:
You can use apt instead of apt-get --allow-releaseinfo-change. See: https://askubuntu.com/questions/989906/explicitly-accept-change-for-ppa-label
When you add ppa repository first time, you can use apt-get as is described in DISTRIBUTIONS.md files. As I've changed descriptions of ppa repositories som time ago, you need to run apt or apt-get --allow-releaseinfo-change to update your local database.
I'll update DISTRIBUTIONS.md and change apt-get with apt to avoid similar issues in future.

JacekDanecki, 2020-04-21:
Commit e7c23a7384b0966f606496abb98969f4fe7bf9b8 updated DISTRIBUTIONS.md files, closing issue.

closed 2020-04-21 by JacekDanecki