intel/compute-runtime #296: Does it support Int8?

wdkwyf, 2020-04-20:
Hi,

I check `source/platform/extensions.cpp`, about  `"cl_khr_int8 "`, it didn't include int8 support. I wonder whether I can use NEO'S int8 support in GPU?

JacekDanecki, 2020-04-20:
I don't see such extension in https://www.khronos.org/registry/OpenCL/specs/2.2/html/OpenCL_Ext.html. Do you want to use vector type int8, as described in https://www.khronos.org/registry/OpenCL/sdk/2.1/docs/man/xhtml/vectorDataTypes.html or basic type as 8 bit integer described here:  https://www.khronos.org/registry/OpenCL/sdk/2.1/docs/man/xhtml/scalarDataTypes.html. You don't need to enable any extension to use such types.
More info about OpenCL data types you can find here: https://www.khronos.org/registry/OpenCL/sdk/2.1/docs/man/xhtml/dataTypes.html

wdkwyf, 2020-04-20:
I use Simplified Mode to convert my own F32 IR model to int8 and want to run it on GPU. But it's very slow. 
I heard the new Xe GPU support Int8, so I ask whether NEO support the new GPU?

closed 2020-04-21 by JacekDanecki