intel/compute-runtime #297: zeDriverGetMemAllocProperties does not set phDevice

maleadt, 2020-04-20:
Spec:

```
__ze_api_export ze_result_t __zecall
zeDriverGetMemAllocProperties(
    ze_driver_handle_t hDriver,                     ///< [in] handle of the driver instance
    const void* ptr,                                ///< [in] memory pointer to query
    ze_memory_allocation_properties_t* pMemAllocProperties, ///< [in,out] query result for memory allocation properties
    ze_device_handle_t* phDevice                    ///< [out][optional] device associated with this allocation
    );
```

Lacking more details, but similar to other oneAPI functions I interpret the optional `phDevice` argument to return the device if the user does not pass nullptr (i.e. it's optional from the request point of view, not the implementation's one). However, the implementation ignores this argument:

https://github.com/intel/compute-runtime/blob/62ef24c52aa8bdfd7b49a8ffa1d2008928e0a7ea/level_zero/core/source/driver/driver_handle_imp.cpp#L80-L90

jandres742, 2020-04-20:
For getMemAllocProperties(), we already added the fix, https://github.com/intel/compute-runtime/commit/b58371df4e98ee60b135183b2fbfde2c0a8094a8. Please confirm.

AdamCetnerowski, 2020-05-22:
Thank you for reporting the issue. As it was fixed, I am closing this issue. If you still see the same problem, please re-open.

closed 2020-05-22 by AdamCetnerowski

maleadt, 2020-05-26:
Confirmed, thanks!