intel/compute-runtime #298: zero-size allocations

maleadt, 2020-04-20:
With C's malloc, zero-size allocations are generally expected to work (either returning a NULL pointer, or an actual address, it does not matter as long as the behavior is OK). With NEO's implementation of oneAPI, it seems like 0-byte allocations return the previously-allocated pointer:

```julia
julia> device_alloc(dev, 1)
1 byte DeviceBuffer at 0xfffffffffff10000

julia> device_alloc(dev, 1)
1 byte DeviceBuffer at 0xfffffffffff00000

julia> device_alloc(dev, 0)
0 bytes DeviceBuffer at 0xfffffffffff00000

julia> device_alloc(dev, 0)
0 bytes DeviceBuffer at 0xfffffffffff00000
```

This is Julia code, but just calling zeDriverAllocDeviceMem directly so that doesn't matter.

The above behavior is of course problematic when freeing pointers:

```julia
julia> a = device_alloc(dev, 1)
1 byte DeviceBuffer at 0xffffffffffef0000

julia> b = device_alloc(dev, 0)
0 bytes DeviceBuffer at 0xffffffffffef0000

julia> free(b)

julia> free(a)
ERROR: ZeError: generic error code for invalid arguments (code 2013265924, ZE_RESULT_ERROR_INVALID_ARGUMENT)
Stacktrace:
 [1] throw_api_error(::oneL0._ze_result_t) at /home/tim/Julia/pkg/oneL0/src/error.jl:102
 [2] macro expansion at /home/tim/Julia/pkg/oneL0/src/error.jl:109 [inlined]
 [3] zeDriverFreeMem at /home/tim/Julia/pkg/CUDAapi/src/call.jl:93 [inlined]
 [4] free(::oneL0.DeviceBuffer) at /home/tim/Julia/pkg/oneL0/src/memory.jl:29
 [5] top-level scope at REPL[36]:1
```

jandres742, 2020-04-20:
Confirmed behavior. Will start working on a patch.

jandres742, 2020-05-09:
Confirmed with Level-Zero spec: Zero allocations are not allowed.

See here:
https://spec.oneapi.com/versions/0.6.0/oneL0/core/api.html?highlight=zedriverallocdevicemem#_CPPv422zeDriverAllocDeviceMem18ze_driver_handle_tPK26ze_device_mem_alloc_desc_t6size_t6size_t18ze_device_handle_tPPv

and:
https://spec.oneapi.com/versions/0.6.0/oneL0/core/api.html?highlight=zedriverallocdevicemem#_CPPv428ZE_RESULT_ERROR_INVALID_SIZE

maleadt, 2020-05-26:
What's the reason for this? It's pretty convenient if it would work, and mimic libc in that regard.

jandres742, 2020-05-28:
Correct. However, in this case, driver implementation is favoring performance over convenience, since any allocation, including zero-sized ones, have associated metadata that needs to be allocated and tracked by the driver. So if application or upper middleware library requires zero-sized allocations, this could be better handled above level-zero, to avoid adding complexity in the driver

AdamCetnerowski, 2020-06-24:
Thank you for the question. Since it's answered, I'll close the issue now.

closed 2020-06-24 by AdamCetnerowski