intel/compute-runtime #299: advise and prefetch error when unsupported

maleadt, 2020-04-20:
Spec:

```
///     - Prefetching may not be supported for all allocation types for all devices.
///       If memory prefetching is not supported for the specified memory range
///       the prefetch hint may be ignored.
```

```
///     - Not all memory advice hints may be supported for all allocation types
///       for all devices.
///       If a memory advice hint is not supported by the device it will be ignored.
```

I would have expected a silent return when using these APIs with unsupported allocation types, instead I'm getting a ZE_RESULT_ERROR_UNKNOWN:

https://github.com/intel/compute-runtime/blob/691a4ea8237f9f38a3eeac472e045c2de126302f/level_zero/core/source/cmdlist/cmdlist_hw.inl#L524-L529

jandres742, 2020-04-20:
interpretation of spec looks correct. Will post patch.

jandres742, 2020-05-08:
Fix merged in 031ea894654a0b65b77dcf184ec8e6ca11f3d0af, please confirm.

maleadt, 2020-05-26:
I'm not seeing the fix:

```
julia> using oneAPI

julia> using oneAPI.oneL0

julia> drv = first(drivers())
ZeDriver(Ptr{Nothing} @0x0000000001dab210)

julia> dev = first(devices(drv))
ZeDevice(GPU, vendor 0x8086, device 0x1912): Intel(R) Gen9

julia> buf = device_alloc(dev, 1024)
1024 bytes DeviceBuffer at 0xfffffffffffe0000

julia> queue = ZeCommandQueue(dev)
ZeCommandQueue(Ptr{Nothing} @0x00000000028c54b0, ZeDevice(Ptr{Nothing} @0x0000000001863e90, ZeDriver(Ptr{Nothing} @0x0000000001dab210)), 0)

julia> list = ZeCommandList(dev)
ZeCommandList(Ptr{Nothing} @0x0000000002896e10)

julia> append_prefetch!(list, pointer(buf), sizeof(buf))
ERROR: ZeError: unknown or internal error (code 2147483647, ZE_RESULT_ERROR_UNKNOWN)
Stacktrace:
 [1] throw_api_error(::oneAPI.oneL0._ze_result_t) at /home/tim/Julia/pkg/oneAPI/src/level-zero/error.jl:102
 [2] macro expansion at /home/tim/Julia/pkg/oneAPI/src/level-zero/error.jl:109 [inlined]
 [3] zeCommandListAppendMemoryPrefetch at /home/tim/Julia/depot/packages/CUDAapi/XuSHC/src/call.jl:93 [inlined]
 [4] append_prefetch!(::ZeCommandList, ::ZePtr{Nothing}, ::Int64) at /home/tim/Julia/pkg/oneAPI/src/level-zero/copy.jl:13
 [5] top-level scope at REPL[10]:1
 ```

Tested with NEO v20.20.16837. I'll try copying the test that's added here.

jandres742, 2020-05-27:
Ummm. Let me take a closer look.

jandres742, 2020-05-28:
@maleadt I see that you are passing to append_prefetch a buffer allocated with device_alloc. Is that a wrapper for zeDriverAllocDeviceMem()?

maleadt, 2020-05-28:
Yes, but turns out it does work, I had a different version of NEO on my LD_LIBRARY_PATH. So this is fixed after all.

Would have been easier to debug if there were an API call to get the actual version of the runtime, and not only the API version. Or does such an API exist?

closed 2020-05-28 by maleadt

HoppeMateusz, 2020-05-28:
@maleadt You can query driver version with: _zeDriverGetProperties_:
https://spec.oneapi.com/versions/0.7/oneL0/core/api.html#_CPPv422ze_driver_properties_t

driverVersion should return unique identifier.

jandres742, 2020-05-29:
You can parse as this:

```cpp
TEST_F(DriverVersionTest, returnsExpectedDriverVersion) {
    ze_driver_properties_t properties;
    ze_result_t res = driverHandle->getProperties(&properties);
    EXPECT_EQ(ZE_RESULT_SUCCESS, res);

    uint32_t versionMajor = static_cast<uint32_t>((properties.driverVersion & 0xFF000000) >> 24);
    uint32_t versionMinor = static_cast<uint32_t>((properties.driverVersion & 0x00FF0000) >> 16);
    uint32_t versionBuild = static_cast<uint32_t>(properties.driverVersion & 0x0000FFFF);

    EXPECT_EQ(static_cast<uint32_t>(strtoul(L0_PROJECT_VERSION_MAJOR, NULL, 10)), versionMajor);
    EXPECT_EQ(static_cast<uint32_t>(strtoul(L0_PROJECT_VERSION_MINOR, NULL, 10)), versionMinor);
    EXPECT_EQ(static_cast<uint32_t>(strtoul(NEO_VERSION_BUILD, NULL, 10)), versionBuild);
}
```