intel/compute-runtime #300: Clinfo not finding on Intel UHD Graphics 620

ling1729, 2020-04-21:
I installed with 
```
add-apt-repository ppa:intel-opencl/intel-opencl
apt update
apt install intel-opencl-icd
```
On WSL 2 Ubuntu 18.04, however clinfo reports 0 platforms. 

However, there is a `/etc/OpenCL/vendors/intel.icd file`. This is the output of  `LD_DEBUG=files clinfo`
```
68:
        68:     WARNING: Unsupported flag value(s) of 0x8000000 in DT_FLAGS_1.
        68:
        68:     file=libOpenCL.so.1 [0];  needed by clinfo [0]
        68:     file=libOpenCL.so.1 [0];  generating link map
        68:       dynamic: 0x00007ff7137edcd0  base: 0x00007ff7135e4000   size: 0x000000000020a050
        68:         entry: 0x00007ff7135e8180  phdr: 0x00007ff7135e4040  phnum:                  8
        68:
        68:
        68:     file=libdl.so.2 [0];  needed by clinfo [0]
        68:     file=libdl.so.2 [0];  generating link map
        68:       dynamic: 0x00007ff7135e2d90  base: 0x00007ff7133e0000   size: 0x0000000000203110
        68:         entry: 0x00007ff7133e0e50  phdr: 0x00007ff7133e0040  phnum:                  7
        68:
        68:
        68:     file=libc.so.6 [0];  needed by clinfo [0]
        68:     file=libc.so.6 [0];  generating link map
        68:       dynamic: 0x00007ff7133d9b80  base: 0x00007ff712fef000   size: 0x00000000003f0ae0
        68:         entry: 0x00007ff713010cb0  phdr: 0x00007ff712fef040  phnum:                 10
        68:
        68:
        68:     calling init: /lib/x86_64-linux-gnu/libc.so.6
        68:
        68:
        68:     calling init: /lib/x86_64-linux-gnu/libdl.so.2
        68:
        68:
        68:     calling init: /usr/lib/x86_64-linux-gnu/libOpenCL.so.1
        68:
68:
        68:     initialize program: clinfo
        68:
        68:
        68:     transferring control: clinfo
        68:
        68:
        68:     file=/usr/lib/x86_64-linux-gnu/intel-opencl/libigdrcl.so [0];  dynamically loaded by /usr/lib/x86_64-linux-gnu/libOpenCL.so.1 [0]
        68:     file=/usr/lib/x86_64-linux-gnu/intel-opencl/libigdrcl.so [0];  generating link map
        68:       dynamic: 0x00007ff712eac1a0  base: 0x00007ff712921000   size: 0x00000000006cdee8
        68:         entry: 0x00007ff712941930  phdr: 0x00007ff712921040  phnum:                  7
        68:
        68:
        68:     file=libpthread.so.0 [0];  needed by /usr/lib/x86_64-linux-gnu/intel-opencl/libigdrcl.so [0]
        68:     file=libpthread.so.0 [0];  generating link map
        68:       dynamic: 0x00007ff71291bd68  base: 0x00007ff712702000   size: 0x000000000021e480
        68:         entry: 0x00007ff712708230  phdr: 0x00007ff712702040  phnum:                  9
        68:
        68:
        68:     file=libigdgmm.so.11 [0];  needed by /usr/lib/x86_64-linux-gnu/intel-opencl/libigdrcl.so [0]
        68:     file=libigdgmm.so.11 [0];  generating link map
        68:       dynamic: 0x00007ff7126fedc0  base: 0x00007ff712492000   size: 0x000000000026f260
        68:         entry: 0x00007ff712498c90  phdr: 0x00007ff712492040  phnum:                  7
        68:
        68:
        68:     file=libstdc++.so.6 [0];  needed by /usr/lib/x86_64-linux-gnu/intel-opencl/libigdrcl.so [0]
68:     file=libstdc++.so.6 [0];  generating link map
        68:       dynamic: 0x00007ff71248ad28  base: 0x00007ff712109000   size: 0x0000000000388040
        68:         entry: 0x00007ff712195490  phdr: 0x00007ff712109040  phnum:                  8
        68:
        68:
        68:     file=libm.so.6 [0];  needed by /usr/lib/x86_64-linux-gnu/intel-opencl/libigdrcl.so [0]
        68:     file=libm.so.6 [0];  generating link map
        68:       dynamic: 0x00007ff712107d90  base: 0x00007ff711d6b000   size: 0x000000000039d158
        68:         entry: 0x00007ff711d76a80  phdr: 0x00007ff711d6b040  phnum:                  7
        68:
        68:
        68:     file=libgcc_s.so.1 [0];  needed by /usr/lib/x86_64-linux-gnu/intel-opencl/libigdrcl.so [0]
        68:     file=libgcc_s.so.1 [0];  generating link map
        68:       dynamic: 0x00007ff711d69de0  base: 0x00007ff711b53000   size: 0x0000000000217430
        68:         entry: 0x00007ff711b55ac0  phdr: 0x00007ff711b53040  phnum:                  7
        68:
        68:
        68:     calling init: /lib/x86_64-linux-gnu/libpthread.so.0
        68:
        68:
        68:     calling init: /lib/x86_64-linux-gnu/libgcc_s.so.1
        68:
        68:
        68:     calling init: /lib/x86_64-linux-gnu/libm.so.6
        68:
        68:
        68:     calling init: /usr/lib/x86_64-linux-gnu/libstdc++.so.6
 68:
        68:
        68:     calling init: /usr/lib/x86_64-linux-gnu/libigdgmm.so.11
        68:
        68:
        68:     calling init: /usr/lib/x86_64-linux-gnu/intel-opencl/libigdrcl.so
        68:
        68:     opening file=/usr/lib/x86_64-linux-gnu/intel-opencl/libigdrcl.so [0]; direct_opencount=1
        68:
Number of platforms                               0
        68:
        68:     calling fini: clinfo [0]
        68:
        68:
        68:     calling fini: /usr/lib/x86_64-linux-gnu/libOpenCL.so.1 [0]
        68:
        68:
        68:     calling fini: /usr/lib/x86_64-linux-gnu/intel-opencl/libigdrcl.so [0]
        68:
        68:
        68:     calling fini: /lib/x86_64-linux-gnu/libdl.so.2 [0]
        68:
        68:
        68:     calling fini: /usr/lib/x86_64-linux-gnu/libigdgmm.so.11 [0]
        68:
        68:
        68:     calling fini: /lib/x86_64-linux-gnu/libpthread.so.0 [0]
        68:
        68:
        68:     calling fini: /lib/x86_64-linux-gnu/libpthread.so.0 [0]
        68:
        68:
        68:     calling fini: /usr/lib/x86_64-linux-gnu/libstdc++.so.6 [0]
        68:
        68:
        68:     calling fini: /lib/x86_64-linux-gnu/libm.so.6 [0]
        68:
        68:
        68:     calling fini: /lib/x86_64-linux-gnu/libgcc_s.so.1 [0]
        68:
```

alalek, 2020-04-21:
> WSL 2

See https://github.com/microsoft/WSL/issues/3789

---

You may check this `lspci | grep VGA` anyway.

JacekDanecki, 2020-04-22:
I don't think it's supported. Besides lspci output can you also provide log from strace:
```
strace -s 255 -f clinfo
```

ling1729, 2020-04-28:
I have retried this dual booting into ubuntu, and it now is successfully detected.

closed 2020-04-28 by ling1729