intel/compute-runtime #302: clGetDeviceIDs(): error

dogomatt, 2020-04-24:
hashcat -I
hashcat (v5.1.0) starting...

clGetDeviceIDs(): CL_DEVICE_NOT_FOUND

OpenCL Info:

Platform ID #1
Vendor  : Intel(R) Corporation
Name    : Intel(R) OpenCL
Version : OpenCL 1.2 

Device ID #1
Type           : CPU
Vendor ID      : 8
Vendor         : Intel(R) Corporation
Name           : Intel(R) Xeon(R) CPU           X5472  @ 3.00GHz
Version        : OpenCL 1.2 (Build 475)
Processor(s)   : 8
Clock          : 3000
Memory         : 5005/20020 MB allocatable
OpenCL Version : OpenCL C 1.2 
Driver Version : 1.2.0.475

Platform ID #2
Vendor  : Mesa
Name    : Clover
Version : OpenCL 1.1 Mesa 20.0.5
########################################################################

DIST : ARCH
DE : KDE

########################################################################
Currently installed

extra/intel-tbb 2020.2-1 [installed]
extra/xf86-video-intel 1:2.99.917+906+g846b53da-1 (xorg-drivers) [installed]
community/intel-gmmlib 20.1.1-1 [installed]
community/intel-graphics-compiler 1:1.0.3771-1 [installed]
community/intel-opencl-clang 10.0.0.1-1 [installed]
community/libmfx 20.1.0-1 [installed]
aur/intel-compute-runtime-git 19.21.13045.r47.gc0383491-1 (0, 0.0) [installed: 20.16.16582.r45.g2dbffe8b6-1]
aur/intel-opencl 5.0.r63503-2 (27, 0.047604) [installed: 5.0.r63503-2]
aur/intel-opencl-runtime 1:18.1.0.013-2 (52, 0.104309) [installed: 1:18.1.0.013-2]
#########################################################################

JacekDanecki, 2020-04-24:
It looks like issue with hashcat application, as clinfo and clpeak work fine under Arch. 
Need more debug here.

dogomatt, 2020-04-25:
Ok after looking into opencl runtime and specs on intel , the latest runtime 18.1 doesn't support xeon processors, the previous version 16.2 supports them , that's what the problem is

I'll leave it open in case someone else has the same problem

closed 2020-04-26 by dogomatt