intel/compute-runtime #303: Make process terminating at 93%

ducbueno, 2020-04-24:
Hello.

I'm trying to install the compute-runtime in Manjaro Linux.

I have installed all the pre-requisites (without erros of any kind), however when run `make` for the compute-runtime the process terminates at 93% showing the following as the first error:

make[2]: *** [shared/source/built_ins/kernels/CMakeFiles/builtins_Gen8core.dir/build.make:98: bin/built_ins/x64/gen8/aux_translation_Gen8core.spv] Error 250

This is what appears before the error:

[ 93%] Generating ../../../../bin/built_ins/x64/gen8/aux_translation_Gen8core.spv, ../../../../bin/built_ins/x64/gen8/aux_translation_Gen8core.bin, ../../../../bin/built_ins/x64/gen8/aux_translation_Gen8core.cpp, ../../../../bin/built_ins/x64/gen8/aux_translation_Gen8core.gen
[ 93%] Generating ../../../../bin/built_ins/x64/gen9/aux_translation_Gen9core.spv, ../../../../bin/built_ins/x64/gen9/aux_translation_Gen9core.bin, ../../../../bin/built_ins/x64/gen9/aux_translation_Gen9core.cpp, ../../../../bin/built_ins/x64/gen9/aux_translation_Gen9core.gen
[ 93%] Generating ../../../../bin/built_ins/x64/gen9/aux_translation_Gen9lp.spv, ../../../../bin/built_ins/x64/gen9/aux_translation_Gen9lp.bin, ../../../../bin/built_ins/x64/gen9/aux_translation_Gen9lp.cpp, ../../../../bin/built_ins/x64/gen9/aux_translation_Gen9lp.gen
[ 93%] Generating ../../../../bin/built_ins/x64/gen12lp/aux_translation_Gen12LPlp.spv, ../../../../bin/built_ins/x64/gen12lp/aux_translation_Gen12LPlp.bin, ../../../../bin/built_ins/x64/gen12lp/aux_translation_Gen12LPlp.cpp, ../../../../bin/built_ins/x64/gen12lp/aux_translation_Gen12LPlp.gen
[ 93%] Generating ../../../../bin/built_ins/x64/gen11/aux_translation_Gen11lp.spv, ../../../../bin/built_ins/x64/gen11/aux_translation_Gen11lp.bin, ../../../../bin/built_ins/x64/gen11/aux_translation_Gen11lp.cpp, ../../../../bin/built_ins/x64/gen11/aux_translation_Gen11lp.gen

Does anyone know what is going on? 

If here is not the appropriate place to ask this kind of question, I would appreciate if anyone could direct me to the correct place.

Thanks in advance.

bfliflet, 2020-04-24:
This is the build phase we're the driver precompiles some internal kernels to SPIRV to embed inside the driver itself.  Are the other platforms passing for the same kernel while gen8 is failing?  If no, temporarily, you might be able to disable GEN8 support in the CMake system while we figure out why it's failing.

JacekDanecki, 2020-05-08:
To disable GEN8 support you can run make, like we do on Travis, see: https://github.com/intel/compute-runtime/blob/master/scripts/docker/Dockerfile-ubuntu-18.04-gcc-gen-9
Can you also check if you can reproduce this compilation error using [the latest Neo](https://github.com/intel/compute-runtime/releases/tag/20.18.16699) with updated IGC to [1.0.3899](https://github.com/intel/intel-graphics-compiler/releases/tag/igc-1.0.3899)

boegel, 2020-06-30:
I'm running into a very similar error (`Error 250`) when building version 20.14.16441 from source (on top of `gmmlib` 20.1.1 and `intel-graphics-compiler` 1.0.3698, on CentOS 7.8).

Configuring with "`-DSUPPORT_GEN8=0 -DSUPPORT_GEN9=1 -DSUPPORT_GEN11=0 -DSUPPORT_GEN12LP=0`" as suggested doesn't help me at all...

I can try again with the latest version of everything, but can somebody clarify what "`Error 250`" actually means exactly?
Where can I find the actual error message what tells me more about that is wrong here?

JacekDanecki, 2020-07-02:
Packages from Neo 20.14.16441 for Centos 7 you can find on [SF mirror](https://sourceforge.net/projects/intel-compute-runtime/files/20.14.16441/centos-7/) 

Regarding "Error 250" you can check [lines > 98 in ocloc_api.cpp](https://github.com/intel/compute-runtime/blob/master/shared/offline_compiler/source/ocloc_api.cpp#L98) and see what step returns this error.

KipHamiltons, 2021-01-15:
I fixed this issue by adding the cmake flag `-DIGC_DIR=<IGC install prefix>` to my build.

JacekDanecki, 2021-08-24:
Error 250 can be reported by ocloc when there is problem with libigdfcl library loading, as reported here: https://gist.github.com/KipHamiltons/8d1d518c5adf8745ab401c8812029ba9#file-output-txt-L365
IGC headers and libraries are searched by pkg-config tool executed by cmake, so when IGC is installed in different location than specified in igc-opencl.pc, it can be overwritten by IGC_DIR variable as mentioned by @KipHamiltons 

If we add -DIGC_DIR=/home/tests/igc to cmake parameters, path to libigdfcl will be added to LD_LIBRARY_PATH set before ocloc execution. It can be verified by "make VERBOSE=1"
```
cd /home/tests/compute-runtime/shared/source/built_ins/kernels && LD_LIBRARY_PATH=/home/tests/igc/lib/:/home/tests/compute-runtime/build/bin /home/tests/compute-runtime/build/bin/ocloc -q -spv_only -file copy_image3d_to_buffer_stateless.builtin_kernel -out_dir /home/tests/compute-runtime/build/bin/built_ins/x64/spirv/ -output_no_suffix -options -cl-kernel-arg-info
```

closed 2021-08-24 by JacekDanecki