intel/compute-runtime #304: free(): invalid pointer

unphased, 2020-05-02:
Ubuntu 18.04 4.15.0-1079-oem (this is a Dell laptop, with the OS installed via first party recovery image)

NEO runtime 20.17.16650 (pretty recent)

[clinfo.log](https://github.com/intel/compute-runtime/files/4568640/clinfo.log)

kernel code to reproduce: 

```
  kernel void voxelize(const global float* a, constant uint* b, uint c,
      global float* d, float3 e, float3 f, global uchar* g, constant uint4 h) {}
```

```
(gdb) whe
#0  0x00007fffc5e7ce97 in raise () at /lib/x86_64-linux-gnu/libc.so.6
#1  0x00007fffc5e7e801 in abort () at /lib/x86_64-linux-gnu/libc.so.6
#2  0x00007fffc5ec7897 in  () at /lib/x86_64-linux-gnu/libc.so.6
#3  0x00007fffc5ece90a in  () at /lib/x86_64-linux-gnu/libc.so.6
#4  0x00007fffc5ed5e1c in free () at /lib/x86_64-linux-gnu/libc.so.6
#5  0x00007fff6d5b49d9 in  () at /usr/local/lib/libigdfcl.so.1
#6  0x00007fff6d5c0ced in  () at /usr/local/lib/libigdfcl.so.1
#7  0x00007fff809781d8 in  () at /usr/local/lib/intel-opencl/libigdrcl.so
#8  0x00007fff807b3f50 in  () at /usr/local/lib/intel-opencl/libigdrcl.so
#9  0x00007fff80756bb8 in  () at /usr/local/lib/intel-opencl/libigdrcl.so
... <snip>
```

JacekDanecki, 2020-05-08:
Do you check value returned by clBuildProgram?
As I can see IGC returns error during compilation:
```
$ ocloc -file empty_kernel.cl -device skl
1:1:41: error: parameter may not be qualified with an address space
__kernel void TestKernel(constant uint4 a) {}
                                        ^

Build failed with error code: -11

$ cat empty_kernel.cl 
__kernel void TestKernel(constant uint4 a) {}
```

unphased, 2020-05-08:
The program crashes with `free(): invalid pointer` before `clBuildProgram()` returns. :)

JacekDanecki, 2020-05-11:
As I can't reproduce this crash on my side, can you recompile IGC as DEBUG, and provide backtrace?

JacekDanecki, 2020-07-01:
Closing due to inactivity.

closed 2020-07-01 by JacekDanecki