intel/compute-runtime #305: valgrind reports memory leaks with clGetPlatformIDs

knzivid, 2020-05-04:
I am investigating the source of memory leaks in my application that uses OpenCL. One among them occurs with `clGetPlatformIDs`

I am using compute-runtime 20.16.16582 on Arch Linux with an Intel HD 630 on a 9900K.

I build this minimal example in debug mode with clang

```
#define __CL_ENABLE_EXCEPTIONS
#include <CL/cl.hpp>

auto getPlatformIds()
{
    cl_uint n;
    clGetPlatformIDs(0, nullptr, &n);
    return n;
}

int main() {
    getPlatformIds();
}
```

Valgrind reports some leaks

```
...
==55682== 5,377 bytes in 1 blocks are definitely lost in loss record 1,833 of 1,837
==55682==    at 0x483BEB8: memalign (vg_replace_malloc.c:908)
==55682==    by 0x483BFCE: posix_memalign (vg_replace_malloc.c:1072)
==55682==    by 0x12A5CAB9: ??? (in /usr/lib/libigc.so.1.0.1)
==55682==    by 0x12BD7B54: ??? (in /usr/lib/libigc.so.1.0.1)
==55682==    by 0x12A8DDAF: ??? (in /usr/lib/libigc.so.1.0.1)
==55682==    by 0x12B678B9: ??? (in /usr/lib/libigc.so.1.0.1)
==55682==    by 0x12B69E5E: ??? (in /usr/lib/libigc.so.1.0.1)
==55682==    by 0x5463321: ??? (in /usr/lib/intel-opencl/libigdrcl.so)
==55682==    by 0x54F2402: ??? (in /usr/lib/intel-opencl/libigdrcl.so)
==55682==    by 0x570692E: __pthread_once_slow (in /usr/lib/libpthread-2.31.so)
==55682==    by 0x54F29B8: ??? (in /usr/lib/intel-opencl/libigdrcl.so)
==55682==    by 0x52530D1: ??? (in /usr/lib/intel-opencl/libigdrcl.so)
==55682== 
==55682== LEAK SUMMARY:
==55682==    definitely lost: 5,377 bytes in 1 blocks
==55682==    indirectly lost: 0 bytes in 0 blocks
==55682==      possibly lost: 304 bytes in 1 blocks
==55682==    still reachable: 166,582 bytes in 1,902 blocks
==55682==         suppressed: 0 bytes in 0 blocks
```

How can I go about debugging this? Is this a false positive?

JacekDanecki, 2020-05-11:
If you run valgrind with --leak-check=full parameter you can see more details about leaks.
One is related to  pthread_create function called when DrmGemCloseWorker thread is created, and second with IGC initialization.
In case of Neo leak it looks like DrmGemCloseWorker thread was not close correctly so pthread_join function was not called. It needs more debugging.
In case of IGC leak I've asked IGC team to look at trace.

knzivid, 2020-05-11:
[Complete log](https://gist.github.com/knzivid/f6bce906d1842937fbc2320e74e1e284) with full leak check.

If you have an issue created for IGC, can you please link it here?

JacekDanecki, 2020-05-13:
Fix for leak in Neo was found, and will be merged soon.

JacekDanecki, 2020-05-14:
commit f3cb94c491d3dcda7f0736920a31ca575106d62e contains fix for leak in Neo.
Fix for leak in IGC is tested and will be merged soon.

JacekDanecki, 2020-06-09:
[IGC commit](https://github.com/intel/intel-graphics-compiler/commit/241078902a30eca9c7ad687f1191e4f3cd1da305) contains fix for leak.

JacekDanecki, 2020-06-10:
Closing issue, as fixes were provided. Feel free to create to issue, when you find any problems.

closed 2020-06-10 by JacekDanecki