intel/compute-runtime #306: ocloc asm doesn't produce binary.

TApplencourt, 2020-05-04:
Hi, 

When using an `ocloc` version after  `20.09.15980`, I'm not at able to reassemble a kernel using `ocloc asm`. I got a `Warning while processing kernel!` and then the new binary is not created.

M(N)WE
```bash
$cat kernel.cl
__kernel void f0() {
}
$ocloc compile -file kernel.cl  -device skl
Build succeeded.
$ ocloc disasm -file kernel_Gen9core.bin -device skl
Warning : Path to dump folder not specificed - using ./dump as default.
Path to patch list not provided - using defaults, skipping patchokens as undefined.
Trying to disassemble f0.krn
$ocloc asm -out new.bin -device skl
Warning : Path to dump folder not specificed - using ./dump as default.
Trying to assemble f0.asm
Warning while processing kernel!
$test -e new.bin
$echo $?
1
```

Thanks!

KrystianChmielewski, 2020-05-18:
Hi,

Commit 10cb4d597f7c3bee6defee68f39e5d3cebefe0d0 fixed it. Should be working as expected now :)

Krystian

TApplencourt, 2020-05-29:
I confirm this is now working :) 
This was a nice "one-character" fix.

I close the issue then,
Thanks you a lot!

closed 2020-05-29 by TApplencourt