intel/compute-runtime #310: Modifying CL cache folder location

w3sip, 2020-05-08:
Hello,

We're trying to generate and maintain app-specific CL cache for our application. From what I can see, accomplishing that on Linux is as simple as setting environment variable cl_cache_dir.

On Windows, however, it is either setting a registry value named "c:\path\to\process\binary.exe" in the HKEY_LOCAL_MACHINE\SOFTWARE\Intel\IGFX\OCL\cl_cache_dir registry key, *or* setting environment variable named "c:\path\to\process\binary.exe". 

Is that approach intentional, especially with regards to environment variable name on Windows? Need to understand if it will be retained in the future versions of the driver.

PiotrRozenfeld, 2020-05-28:
Good question – the driver behavior around cl_cache is indeed a little complex. We’ll use this opportunity to rework our documentation in [opencl/doc/FAQ.md](https://github.com/intel/compute-runtime/blob/master/opencl/doc/FAQ.md).

Short answer:
* Using environmental variable named cl_cache_dir should work the same between Windows and Linux
* Using windows registry to globally configure multiple pairs of <app path / cl_cache dir> should have no corresponding environmental variable behavior. 

The fact that driver is reading from variable _named_ “c:\path\to\process\binary.exe" is not intentional and should be fixed shortly.

Note: Debug keys are not considered a part of the product and may change over time.

JacekDanecki, 2020-07-09:
Closing as an answer was provided. Feel free to create new issue if you have more questions.

closed 2020-07-09 by JacekDanecki