intel/compute-runtime #315: How do I make Debian package artifacts?

realtime-neil, 2020-05-22:
I can see the Debian artifacts available for download. How do I make my own from a git checkout?

JacekDanecki, 2020-05-25:
You can run:
```
make package
```
or you can use scripts from [neo-specs repo](https://github.com/JacekDanecki/neo-specs/tree/launchpad/launchpad/intel-opencl)

realtime-neil, 2020-05-25:
I can't get the `make` target to work...

```
$ cd $(mktemp -dt compute-runtime.XXXXXX) && git clone https://github.com/intel/compute-runtime.git . && make package
Cloning into '.'...
remote: Enumerating objects: 96351, done.
remote: Total 96351 (delta 0), reused 0 (delta 0), pack-reused 96351
Receiving objects: 100% (96351/96351), 39.15 MiB | 33.92 MiB/s, done.
Resolving deltas: 100% (82161/82161), done.
make: *** No rule to make target 'package'.  Stop.
$ 
```

...but those scripts look promising.

JacekDanecki, 2020-05-26:
You need to run cmake before make, see: https://github.com/intel/compute-runtime/blob/master/BUILD.md

realtime-neil, 2020-05-26:
Ah, okay. I was looking at your bionic [`debian/rules` file](https://github.com/JacekDanecki/neo-specs/blob/launchpad/launchpad/intel-opencl/bionic/debian/rules#L7) and I didn't realize that `debhelper` could detect a Cmake project without an explicit `--buildsystem=cmake`. Thanks for the help!

JacekDanecki, 2020-05-28:
Closing issue, as answer was provided.

closed 2020-05-28 by JacekDanecki