intel/compute-runtime #317: Request: Cherry Trail / Braswell (Gen8LP) support

ipaucek4680, 2020-05-24:
In https://github.com/intel/compute-runtime/issues/168#issuecomment-496568640, a developer said that Cherry Trail / Braswell support is technically possible in NEO.

Reasons for this request:
1. Braswell chips are not old (e.g. for N3160, it is in[ "Launched" status according to Intel Ark](https://ark.intel.com/content/www/us/en/ark/products/91831/intel-celeron-processor-n3160-2m-cache-up-to-2-24-ghz.html)), is Gen8LP (NEO targets Gen 8 and above), and are still useful.
2. The [beignet runtime](https://github.com/intel/beignet) is deprecated and unmaintained, not to mention that it does not support newer LLVM versions (especially LLVM 8 and above).

If support for Gen8LP is implemented, I would be happy to be a tester and report bugs that I find on my PCs with N3050 and N3160 (and probably @gilbahat as well).

AdamCetnerowski, 2020-07-10:
We are investigating the effort required to provide basic support for platform. No ETA, but we will update this issue when more information is available.

PiotrRozenfeld, 2020-09-07:
We concluded the engineering investigation into Cherrytrail / NEO. Unfortunately, the amount of work required for even most basic platform support is far more than anticipated given the current driver design.

closed 2020-09-07 by PiotrRozenfeld