intel/compute-runtime #318: Build system shouldn't overwrite LD_LIBRARY_PATH

maleadt, 2020-05-26:
... but extend it

```patch
-        set(cloc_cmd_prefix LD_LIBRARY_PATH=${NEO__IGC_LIBRARY_PATH}:$<TARGET_FILE_DIR:ocloc_lib> $<TARGET_FILE:ocloc>)
+        set(cloc_cmd_prefix LD_LIBRARY_PATH=$ENV{LD_LIBRARY_PATH}:${NEO__IGC_LIBRARY_PATH}:$<TARGET_FILE_DIR:ocloc_lib> $<TARGET_FILE:ocloc>)
```

JacekDanecki, 2020-05-28:
Do you have installed IGC in not standard location, so you need to use LD_LIBRARY_PATH?
Can you provide log from cmake?

maleadt, 2020-05-28:
> Do you have installed IGC in not standard location, so you need to use LD_LIBRARY_PATH?

I need LD_LIBRARY_PATH for other purposes (specifically, to point to glibc in a musl-native environment).

JacekDanecki, 2020-05-29:
Are you using Alpine or Void Linux for development?

maleadt, 2020-05-29:
> Are you using Alpine or Void Linux for development?

This isn't my development system, but the build environment where we generate packages. It's Alpine-based, but also provides a glibc build to be able to execute binaries (since the build system executes `ocloc`, which is problematic for cross-compilation by itself).

Anyway, all that is not really relevant, as developers could have LD_LIBRARY_PATH set for a variety of other reasons. Isn't it pretty uncontroversial to append to that variable instead of overwriting it?

JacekDanecki, 2020-05-29:
I've asked about Alpine because we have 3 others issues reported on this distro, and I also saw people from Void Linux are working on adding Neo stack.

LD_LIBRARY_PATH was set to minimize impact of libraries installed in the system during ocloc execution. I'd add $ENV{LD_LIBRARY_PATH} at the end instead of at the beginning, but let me discuss this with the team.

maleadt, 2020-05-29:
> I've asked about Alpine because we have 3 others issues reported on this distro, and I also saw people from Void Linux are working on adding Neo stack.

Alpine (or specifically, musl) support would be mostly interesting to us to be able to do proper cross compilation, with a host version of `ocloc` for use during build for generating target binaries. I actually tried to do exactly that, but cross-compiling IGC turned out to be [problematic](https://github.com/JuliaPackaging/Yggdrasil/pull/816/files). Even with a bunch of musl-related patches, I ran into an issue where (I think) OpenCL PCHs generated during IGC host compilation turned out incompatible with the target environment: https://github.com/intel/intel-graphics-compiler/issues/128#issuecomment-609065962

Anyway, for now oneAPI seems like a Linux/x86/glibc-only thing, so I decided to punt on all that.

closed 2022-11-18 by Compute-Runtime-Automation