intel/compute-runtime #319: Docker builds are outdated

maleadt, 2020-05-27:
I just found the [intel/oneapi image](https://hub.docker.com/r/intel/oneapi/tags), but the latest tag provides v20.12.16259 which is pretty old. Could this be updated? It would be useful for CI purposes.

A related question: is there a way to query the actual driver version, i.e., 16259 and not the API version 0.91? This to blacklist certain tests in my wrapper package.

JacekDanecki, 2020-05-28:
If Neo was configured with cmake parameters:
``` 
 -DNEO_OCL_VERSION_MAJOR=$(NEO_MAJOR) \
 -DNEO_OCL_VERSION_MINOR=$(NEO_MINOR) \
 -DNEO_VERSION_BUILD=$(NEO_BUILD) \
```
see: https://github.com/JacekDanecki/neo-specs/blob/launchpad/launchpad/intel-opencl/bionic/debian/rules
you can run
```
clinfo | grep "Driver Version"
  Driver Version                                  20.19.16754
```
0.91 is current API version for Level Zero loader used to build Neo, see https://github.com/intel/compute-runtime/blob/master/manifests/manifest.yml#L33

We're not responsible for updating intel/oneapi docker images, let me find correct contact. If you want to use the latest docker images with Neo, you can use: https://hub.docker.com/r/intelopencl/intel-opencl. I've updated them 2 days ago to Neo 20.19.16754.

maleadt, 2020-05-28:
> you can run
> 
> ```
> clinfo | grep "Driver Version"
>   Driver Version                                  20.19.16754
> ```

Is there an API for that? I'm not shipping `clinfo`.

> We're not responsible for updating intel/oneapi docker images, let me find correct contact. If you want to use the latest docker images with Neo, you can use: https://hub.docker.com/r/intelopencl/intel-opencl. I've updated them 2 days ago to Neo 20.19.16754.

Those don't ship oneAPI (libze, libze_intel_gpu), right?

jandres742, 2020-05-29:
> A related question: is there a way to query the actual driver version, i.e., 16259 and not the API version 0.91? This to blacklist certain tests in my wrapper package.

Replied also in #299

JacekDanecki, 2020-05-29:
You can use:  clGetDeviceInfo with CL_​DRIVER_​VERSION, see https://www.khronos.org/registry/OpenCL/sdk/2.2/docs/man/html/clGetDeviceInfo.html, and here is example: https://github.com/jdanecki/FractalCL/blob/master/ocl.c#L68
intel-opencl images don't include Level Zero libraries yet, but today I'll make update to 20.20.16837, and will add them. If you want me to add anything else, let me know by creating issue in https://github.com/JacekDanecki/neo-hub repository.

Regarding intel/oneapi images I've created new issue in https://github.com/intel/oneapi-containers/issues/2. As I know Neo will be updated soon in these images.

JacekDanecki, 2020-05-29:
[intel-opencl](https://hub.docker.com/r/intelopencl/intel-opencl) docker images were updated to 20.20.16837, and Level Zero packages were added.
Closing issue as all questions were answered. Feel free to create new.

closed 2020-05-29 by JacekDanecki