intel/compute-runtime #320: OpConvertUToPtr breaks simple memory store

maleadt, 2020-05-28:
I have a kernel that very simply sets a pointer to the value 42:

```c
#include <stdint.h>

kernel void dummy(__global int* ptr) {
    *ptr = 42;
    return;
}
```

This is generated from Julia source code though, where we pass pointers as integers and cast them to a pointer before using them. Example in LLVM IR:

```llvm
source_filename = "kernel.cl"
target datalayout = "e-i64:64-v16:16-v24:32-v32:32-v48:64-v96:128-v192:256-v256:256-v512:512-v1024:1024"
target triple = "spir64"

define spir_kernel void @dummy(i32) {
top:
  %1 = inttoptr i32 %0 to i32 addrspace(1)*
  store i32 2, i32 addrspace(1)* %1, align 1
  ret void
}

!llvm.module.flags = !{!0}
!opencl.ocl.version = !{!1}
!opencl.spir.version = !{!2}

!0 = !{i32 1, !"wchar_size", i32 4}
!1 = !{i32 1, i32 0}
!2 = !{i32 1, i32 2}
```

This results in the following SPIR-V IR (compiled using Khronos' LLVM to SPIR-V translator):

```
119734787 65536 393230 11 0 
2 Capability Addresses 
2 Capability Kernel 
5 ExtInstImport 1 "OpenCL.std"
3 MemoryModel 2 2 
5 EntryPoint 6 5 "dummy"
3 Source 3 100000 
3 Name 7 "top"
4 TypeInt 3 32 0 
4 Constant 3 10 2 
2 TypeVoid 2 
4 TypeFunction 4 2 3 
4 TypePointer 8 5 3 

5 Function 2 5 0 4 
3 FunctionParameter 3 6 

2 Label 7 
4 ConvertUToPtr 8 9 6 
5 Store 9 10 2 1 
1 Return 

1 FunctionEnd
```

As rendered by `spirv-as`:

```
               OpCapability Addresses
               OpCapability Kernel
          %1 = OpExtInstImport "OpenCL.std"
               OpMemoryModel Physical64 OpenCL
               OpEntryPoint Kernel %5 "dummy"
               OpSource OpenCL_C 100000
               OpName %top "top"
       %uint = OpTypeInt 32 0
     %uint_2 = OpConstant %uint 2
       %void = OpTypeVoid
          %4 = OpTypeFunction %void %uint
%_ptr_CrossWorkgroup_uint = OpTypePointer CrossWorkgroup %uint
          %5 = OpFunction %void None %4
          %6 = OpFunctionParameter %uint
        %top = OpLabel
          %9 = OpConvertUToPtr %_ptr_CrossWorkgroup_uint %6
               OpStore %9 %uint_2 Aligned 1
               OpReturn
               OpFunctionEnd
```

<details><summary>Here's C++ code I use to load and execute this kernel.</summary>
<p>

```c++
#include <cassert>
#include <climits>
#include <cstdio>
#include <cstdlib>
#include <cstring>
#include <level_zero/ze_api.h>

#define BUFFERSIZE 1024

#define check(ans)                                                             \
  { do_check((ans), __FILE__, __LINE__); }
void do_check(ze_result_t code, const char *file, int line) {
  if (code != ZE_RESULT_SUCCESS) {
    fprintf(stderr, "Failed: %d at %s %d\n", code, file, line);
    exit(1);
  }
}

int main() {
  check(zeInit(ZE_INIT_FLAG_GPU_ONLY));

  ze_driver_handle_t driver = nullptr;
  uint32_t count = 1;
  check(zeDriverGet(&count, &driver));

  static ze_device_handle_t device = nullptr;
  check(zeDeviceGet(driver, &count, &device));

  ze_command_queue_desc_t queue_desc{};
  queue_desc.version = ZE_COMMAND_QUEUE_DESC_VERSION_CURRENT;
  ze_command_queue_handle_t command_queue = nullptr;
  check(zeCommandQueueCreate(device, &queue_desc, &command_queue));

  ze_command_list_desc_t command_list_desc{};
  ze_command_list_handle_t command_list;
  check(zeCommandListCreate(device, &command_list_desc, &command_list));

  uint8_t *buffer = (uint8_t *)malloc(BUFFERSIZE);
  FILE *filp = fopen("kernel.spv", "rb");
  size_t bytes_read = fread(buffer, sizeof(uint8_t), BUFFERSIZE, filp);

  ze_module_desc_t module_desc = {ZE_MODULE_DESC_VERSION_CURRENT,
                                  ZE_MODULE_FORMAT_IL_SPIRV,
                                  bytes_read,
                                  buffer,
                                  nullptr,
                                  nullptr};
  ze_module_handle_t module;
  check(zeModuleCreate(device, &module_desc, &module, nullptr));

  ze_kernel_desc_t kernel_desc = {ZE_KERNEL_DESC_VERSION_CURRENT,
                                  ZE_KERNEL_FLAG_NONE, "dummy"};
  ze_kernel_handle_t kernel;
  check(zeKernelCreate(module, &kernel_desc, &kernel));

  size_t size = 1;
  uint8_t *out = (uint8_t *)malloc(sizeof(uint8_t));
  memset(out, -1, size);

  ze_device_mem_alloc_desc_t device_desc;
  device_desc.ordinal = 0;
  device_desc.flags = ZE_DEVICE_MEM_ALLOC_FLAG_DEFAULT;
  device_desc.version = ZE_DEVICE_MEM_ALLOC_DESC_VERSION_CURRENT;

  void *mem;
  check(zeDriverAllocDeviceMem(driver, &device_desc, size, 0, device, &mem));

  check(zeKernelSetArgumentValue(kernel, 0, sizeof(int *), &mem));
  ze_group_count_t kernel_args = {1, 1, 1};
  check(zeCommandListAppendLaunchKernel(command_list, kernel, &kernel_args,
                                        nullptr, 0, nullptr));

  check(zeCommandListAppendBarrier(command_list, nullptr, 0, nullptr));

  check(zeCommandListAppendMemoryCopy(command_list, out, mem, size, nullptr));

  check(zeCommandListClose(command_list));

  check(zeCommandQueueExecuteCommandLists(command_queue, 1, &command_list,
                                          nullptr));

  check(zeCommandQueueSynchronize(command_queue, UINT_MAX));

  for (size_t i = 0; i < size; i++) {
    printf("out[%zu] = %d\n", i, out[i]);
    fflush(stdout);
    assert(out[i] == 42);
  }
}
```

</p>
</details>

But that doesn't give the expected result, `*out` is still 0! If I use LLVM and SPIR-V IR that more resembles what a C compiler would generate, i.e. passing a pointer directly, I do get the expected results of 42.

```llvm
define spir_kernel void @dummy(i32 addrspace(1)*) {
top:
  store i32 42, i32 addrspace(1)* %0, align 1
  ret void
}
```

```
119734787 65536 393230 10 0 
2 Capability Addresses 
2 Capability Kernel 
5 ExtInstImport 1 "OpenCL.std"
3 MemoryModel 2 2 
5 EntryPoint 6 6 "dummy"
3 Source 3 100000 
3 Name 8 "top"
4 TypeInt 3 32 0 
4 Constant 3 9 42 
2 TypeVoid 2 
4 TypePointer 4 5 3 
4 TypeFunction 5 2 4 

5 Function 2 6 0 5 
3 FunctionParameter 4 7 

2 Label 8 
5 Store 7 9 2 1 
1 Return 

1 FunctionEnd
```

The same, but from the SPIR-V tools:

```
               OpCapability Addresses
               OpCapability Kernel
          %1 = OpExtInstImport "OpenCL.std"
               OpMemoryModel Physical64 OpenCL
               OpEntryPoint Kernel %6 "dummy"
               OpSource OpenCL_C 100000
               OpName %top "top"
       %uint = OpTypeInt 32 0
    %uint_42 = OpConstant %uint 42
       %void = OpTypeVoid
%_ptr_CrossWorkgroup_uint = OpTypePointer CrossWorkgroup %uint
          %5 = OpTypeFunction %void %_ptr_CrossWorkgroup_uint
          %6 = OpFunction %void None %5
          %7 = OpFunctionParameter %_ptr_CrossWorkgroup_uint
        %top = OpLabel
               OpStore %7 %uint_42 Aligned 1
               OpReturn
               OpFunctionEnd
```

Except for the conversion, these kernels look pretty identical, but I'm not familiar enough with the IR yet to judge.

jchodor, 2020-05-28:
First of all, you have i32 type representing the address parameter even though the target triple is spir64.  This means that only lower 32-bits of the address are valid after casting them to actual pointer type.

The big question is - are there any specific reasons for passing pointers as integers instead of pointer types? With pointer type arguments, driver would get additional implicit hint that parameter represents a buffer and would be able to handle memory residency automatically. Otherwise (when using integer arguments), you need to handle residency explicitly in the host code of your application. Additionally, inttoptr/ptrtoint casts can negatively impact or even block certain compiler optimizations which will lead to degraded performance.

maleadt, 2020-05-28:
> First of all, you have i32 type representing the address parameter even though the target triple is spir64. This means that only lower 32-bits of the address are valid after casting them to actual pointer type.

Ah yes, I wrongly simplified my example. The problem persists passing a `i64` though.

> The big question is - are there any specific reasons for passing pointers as integers instead of pointer types?

It's the language's ABI, so in principle I can't do anything about that. I'll be trying to pass actual pointers where possible, but the general case of passing an integer and casting it to a pointer should still work, right?

> (when using integer arguments), you need to handle residency explicitly in the host code of your application

Can you elaborate on that?

bfliflet, 2020-05-28:
> Can you elaborate on that?

Under currently supported memory models, the GPU can't page fault.  The compute runtime will implicitly page-in memory for kernel arguments that point to buffers and images. When you reference memory in a _clever_ way, the runtime doesn't know to implicitly page in your memory.  We call the process of paging in memory for access by GPU as 'residency'.

maleadt, 2020-05-28:
So even device memory isn't always resident? I found the relevant section, https://spec.oneapi.com/versions/latest/oneL0/core/PROG.html#device-residency (which uses host memory, hence my question), I'll try and see if that helps.

bfliflet, 2020-05-28:
Each application typically assumes it owns the entirety of GPU resources including device memory.  Device memory is shared across all running applications.  This is just like the operating system does with physical DRAM in your computer - paging out to disk when over-subscribed.  In the case of device memory, contents must be preserved when your device memory is needed by another application and typically gets swapped to system memory.  When your application gets its time-slice, the driver stack will migrate it back to device memory prior to execution.  

Establishing residency does take time and we tried to provide the relevant hooks so that applications can explicitly control it.  The link you provided provides additional details.  It is important to note that the residency API doesn't provide exclusive access to memory.  When over-subscribed, resources will still page-swap.  I hope this helps!

maleadt, 2020-05-29:
Thanks. My device does support page faults though, so residency shouldn't matter here, right?

```
julia> using oneAPI.oneL0

julia> drv = first(drivers())
ZeDriver(Ptr{Nothing} @0x00000000016440f0)

julia> dev = first(devices(drv))
ZeDevice(GPU, vendor 0x8086, device 0x1912): Intel(R) Gen9

julia> properties(dev)
(type = oneAPI.oneL0.ZE_DEVICE_TYPE_GPU, vendorId = 0x8086, deviceId = 0x1912, uuid = UUID("00000000-0000-0000-0000-191200008086"), subdeviceId = nothing, coreClockRate = 1150, unifiedMemorySupported = true, eccMemorySupported = false, onDemandPageFaultsSupported = true, maxCommandQueues = 1, numAsyncComputeEngines = 1, numAsyncCopyEngines = 0, maxCommandQueuePriority = 0, numThreadsPerEU = 7, physicalEUSimdWidth = 8, numEUsPerSubslice = 8, numSubslicesPerSlice = 3, numSlices = 1, timerResolution = 83, name = "Intel(R) Gen9")

julia> properties(dev).onDemandPageFaultsSupported
true
```

bfliflet, 2020-05-29:
Gen9 reporting onDemandPageFaultsSupported is a known bug and will be fixed shortly.  Unfortunately, it doesn't support on demand faulting and requires resources resident (implicitly or explicitly) prior to spawning kernels.

It's not clear why zeDeviceMakeMemoryResident isn't resolving your problem.  Can you elaborate on the memory you're trying to reference and the parameters to zeDeviceMakeMemoryResident?

maleadt, 2020-05-30:
> It's not clear why zeDeviceMakeMemoryResident isn't resolving your problem. Can you elaborate on the memory you're trying to reference and the parameters to zeDeviceMakeMemoryResident?

The loader from above, but with a call to `zeDeviceMakeMemoryResident`:

```
  size_t size = 1;
  uint8_t *out = (uint8_t *)malloc(size*sizeof(uint8_t));
  memset(out, -1, size);

  ze_device_mem_alloc_desc_t device_desc;
  device_desc.ordinal = 0;
  device_desc.flags = ZE_DEVICE_MEM_ALLOC_FLAG_DEFAULT;
  device_desc.version = ZE_DEVICE_MEM_ALLOC_DESC_VERSION_CURRENT;

  void *mem;
  check(zeDriverAllocDeviceMem(driver, &device_desc, size, 0, device, &mem));

  check(zeDeviceMakeMemoryResident(device, mem, size*sizeof(uint8_t)));
```

bfliflet, 2020-06-04:
Sorry for the delay in responding to your last update.  Your pattern is the one we want people to use.  I suspect there are issues with zeDeviceMakeMemoryResident.  Currently, that API only is reported as support on Windows.  We're working on Linux support.  Can you try enabling explicit residency for all resources via ZE_KERNEL_FLAG_FORCE_RESIDENCY to see if that helps your problem?  If not, you'll need to design your memory access patterns in such a way that when the driver establishes residency for your kernel arguments it makes this indirect memory resident also.  

Residency is currently determined by finding the base allocation of a pointer. If your indirect memory was part of that base allocation, it will be made resident.  Imagine allocating a large heap with one allocation and sub-managing it yourself.  That may work around the issue of indirect residency until Linux support is added to zeDeviceMakeMemoryResident.

maleadt, 2020-06-05:
> Can you try enabling explicit residency for all resources via ZE_KERNEL_FLAG_FORCE_RESIDENCY to see if that helps your problem?

Nope, doesn't work. I'm on Linux.

> If not, you'll need to design your memory access patterns in such a way that when the driver establishes residency for your kernel arguments it makes this indirect memory resident also.

This isn't an indirect access: all my pointer arguments are integers, so the driver won't be able to establish any residency. It seems like a fairly insane design for the driver to be introspecting kernel arguments types and values to figure out what to make resident?

> It is important to note that the residency API doesn't provide exclusive access to memory. When over-subscribed, resources will still page-swap. I hope this helps!

I hope that's only the case when page faulting is supported? Otherwise I'd need to be making everything resident again every time I launch a kernel, IIUC.

bfliflet, 2020-06-05:
It would appear your program expects all allocated memory to be resident.  Is that correct?  I want to make sure I understand your expectations.

maleadt, 2020-06-05:
> It would appear your program expects all allocated memory to be resident. Is that correct? I want to make sure I understand your expectations.

I'm used to CUDA programming, where yes I'd expect device allocations to be resident all the time, and only unified host/device memory is paged in and out whenever needed. I don't really need it to be that way, but it's not entirely clear to me what the programming model expects / best practices are here.

jandres742, 2020-06-08:
@maleadt : So I tested your sample with latest release, https://github.com/intel/compute-runtime/releases/tag/20.22.16952, and it works for me w/o problems. I took your kernel, generated the SPIRV with ocloc, and got the expected output:

```cpp
kernel void dummy(__global int* ptr) {
    *ptr = 42;
    return;
}
```

```sh
$ LD_LIBRARY_PATH=~/lib/driver-public ./a.out
out[0] = 42
```

I assume you are also using latest build.

could you share your AUB trace file (execute with `SetCommandStreamReceiver=1`). This will give us insight of how the commands are being programmed and offloaded to the GPU, and maybe we can spot some differences between your execution and ours.

jandres742, 2020-06-20:
Issue with onDemandPageSupport fixed in af2fe237b42c27ed18faf2cedaa127ed147bfe88

maleadt, 2020-06-23:
Sorry, have been busy with other work. I'll try to revisit this soon.

maleadt, 2020-07-02:
I'm still seeing the same issue, and put up a full repro here: https://gist.github.com/maleadt/8041818a1eb927826469c09d6b288c44 (just do `make` and `./main`; don't rebuild the kernel sources since I modified the `.ll` to pass the pointer as an integer as detailed in this issue). Tested on both 20.18.16699 (as provided by Arch Linux) and 20.24.17065.

maleadt, 2020-07-15:
I've worked around this from the Julia side, so this is low-priority for me now.

jandres742, 2020-09-12:
@maleadt  I updated your test to support L0 1.0, but dont see any errors. Is there something to do here?

```sh
$ ./test.run
out[0] = 42
```

```cpp
#include <cassert>
#include <climits>
#include <cstdio>
#include <cstdlib>
#include <cstring>
#include <level_zero/ze_api.h>
#include <iostream>
#include <vector>
#include <limits>

#define BUFFERSIZE 1024

#define check(ans)                                                             \
  { do_check((ans), __FILE__, __LINE__); }
void do_check(ze_result_t code, const char *file, int line) {
  if (code != ZE_RESULT_SUCCESS) {
    fprintf(stderr, "Failed: %d at %s %d\n", code, file, line);
    exit(1);
  }
}

int main() {
// Initialize driver
    check(zeInit(ZE_INIT_FLAG_GPU_ONLY));

    // Retrieve driver
    uint32_t driverCount = 0;
    check(zeDriverGet(&driverCount, nullptr));

    ze_driver_handle_t driverHandle;
    check(zeDriverGet(&driverCount, &driverHandle));

    ze_context_desc_t contextDesc = {};
    ze_context_handle_t context;
    check(zeContextCreate(driverHandle, &contextDesc, &context));

    // Retrieve device
    uint32_t deviceCount = 0;
    check(zeDeviceGet(driverHandle, &deviceCount, nullptr));

    ze_device_handle_t device;
    deviceCount = 1;
    check(zeDeviceGet(driverHandle, &deviceCount, &device));

    // Print some properties
    ze_device_properties_t deviceProperties = {};
    check(zeDeviceGetProperties(device, &deviceProperties));

    // Create command queue
    uint32_t numQueueGroups = 0;
    check(zeDeviceGetCommandQueueGroupProperties(device, &numQueueGroups, nullptr));
    if (numQueueGroups == 0) {
        return 1;
    }
    std::vector<ze_command_queue_group_properties_t> queueProperties(numQueueGroups);
    check(zeDeviceGetCommandQueueGroupProperties(device, &numQueueGroups,
                                                        queueProperties.data()));

    ze_command_queue_handle_t command_queue;
    ze_command_queue_desc_t cmdQueueDesc = {};

    for (uint32_t i = 0; i < numQueueGroups; i++) {
        if (queueProperties[i].flags & ZE_COMMAND_QUEUE_GROUP_PROPERTY_FLAG_COMPUTE) {
            cmdQueueDesc.ordinal = i;
        }
    }
    cmdQueueDesc.index = 0;
    cmdQueueDesc.mode = ZE_COMMAND_QUEUE_MODE_ASYNCHRONOUS;
    check(zeCommandQueueCreate(context, device, &cmdQueueDesc, &command_queue));

    // Create command list
    ze_command_list_handle_t command_list;
    ze_command_list_desc_t cmdListDesc = {};
    cmdListDesc.commandQueueGroupOrdinal = cmdQueueDesc.ordinal;
    check(zeCommandListCreate(context, device, &cmdListDesc, &command_list));

  uint8_t *buffer = (uint8_t *)malloc(BUFFERSIZE);
  FILE *filp = fopen("kernel.spv", "rb");
  size_t bytes_read = fread(buffer, sizeof(uint8_t), BUFFERSIZE, filp);

  ze_module_desc_t module_desc = {};
  module_desc.format = ZE_MODULE_FORMAT_IL_SPIRV;
  module_desc.inputSize = bytes_read;
  module_desc.pInputModule = buffer;
  ze_module_handle_t module;
  check(zeModuleCreate(context, device, &module_desc, &module, nullptr));

  ze_kernel_desc_t kernel_desc = {};
  kernel_desc.pKernelName = "dummy";
  ze_kernel_handle_t kernel;
  check(zeKernelCreate(module, &kernel_desc, &kernel));

  size_t size = 1;
  uint8_t *out = (uint8_t *)malloc(sizeof(uint8_t));
  memset(out, -1, size);

  ze_device_mem_alloc_desc_t device_desc;
  device_desc.ordinal = 0;

  void *mem;
  check(zeMemAllocDevice(context, &device_desc, size, 0, device, &mem));

  check(zeKernelSetArgumentValue(kernel, 0, sizeof(int *), &mem));
  ze_group_count_t kernel_args = {1, 1, 1};
  check(zeCommandListAppendLaunchKernel(command_list, kernel, &kernel_args,
                                        nullptr, 0, nullptr));

  check(zeCommandListAppendBarrier(command_list, nullptr, 0, nullptr));

  check(zeCommandListAppendMemoryCopy(command_list, out, mem, size, nullptr, 0, nullptr));

  check(zeCommandListClose(command_list));

  check(zeCommandQueueExecuteCommandLists(command_queue, 1, &command_list,
                                          nullptr));

  check(zeCommandQueueSynchronize(command_queue, std::numeric_limits<uint64_t>::max()));

  for (size_t i = 0; i < size; i++) {
    printf("out[%zu] = %d\n", i, out[i]);
    fflush(stdout);
    assert(out[i] == 42);
  }
}
```

maleadt, 2020-09-14:
Yes, this still reproduces. Demo in Julia below (I don't have the time to make a C reducer once again).

Set-up: 1-element array, kernel that sets value.

```julia
julia> using oneAPI

julia> kernel(ptr) = (unsafe_store!(ptr, 1); return)

julia> a = oneArray([0])
1-element oneArray{Int64,1}:
 0

julia> using .oneL0

julia> drv = first(drivers())
ZeDriver(0x0000000002130770, version 1.0.0)

julia> dev = first(devices(drv))
ZeDevice(GPU, vendor 0x8086, device 0x1912): Intel(R) Gen9

julia> ctx = context()
ZeContext(Ptr{Nothing} @0x0000000001d0ab90, ZeDriver(Ptr{Nothing} @0x0000000002130770))
```

Version 1: call with pointer as integer:

```julia
julia> @device_code_llvm @oneapi kernel(pointer(a))
; CompilerJob of kernel kernel(Ptr{Int64}) for GPUCompiler.SPIRVCompilerTarget
;  @ REPL[3]:1 within `kernel'
define dso_local spir_kernel void @_Z17julia_kernel_30133PtrI5Int64E(i64) local_unnamed_addr {
top:
; ┌ @ pointer.jl:118 within `unsafe_store!' @ pointer.jl:118
   %1 = inttoptr i64 %0 to i64*
   store i64 1, i64* %1, align 1
; └
  ret void
}

julia> synchronize(global_queue(ctx, dev))

julia> a
1-element oneArray{Int64,1}:
 0
```

Doesn't work. Change by passing pointer as an actual pointer (the `LLVMPtr` type has a different ABI):

```julia
julia> @device_code_llvm @oneapi kernel(reinterpret(Core.LLVMPtr{Int,AS.Global}, pointer(a)))
; CompilerJob of kernel kernel(Core.LLVMPtr{Int64,1}) for GPUCompiler.SPIRVCompilerTarget
;  @ REPL[3]:1 within `kernel'
define dso_local spir_kernel void @_Z17julia_kernel_33507LLVMPtrI5Int64Li1EE(i8 addrspace(1)*) local_unnamed_addr {
top:
; ┌ @ /home/tim/Julia/pkg/LLVM/src/interop/pointer.jl:82 within `unsafe_store!' @ /home/tim/Julia/pkg/LLVM/src/interop/pointer.jl:82 @ /home/tim/Julia/pkg/LLVM/src/interop/pointer.jl:82
; │┌ @ /home/tim/Julia/pkg/LLVM/src/interop/pointer.jl:42 within `pointerset'
; ││┌ @ /home/tim/Julia/pkg/LLVM/src/interop/pointer.jl:42 within `macro expansion' @ /home/tim/Julia/pkg/LLVM/src/interop/base.jl:80
     %1 = bitcast i8 addrspace(1)* %0 to i64 addrspace(1)*
     store i64 1, i64 addrspace(1)* %1, align 1
; └└└
  ret void
}

julia> a
1-element oneArray{Int64,1}:
 1
```

jandres742, 2020-09-14:
Thanks @maleadt . Could you summarize what you think the error is? Is it still about residency?

maleadt, 2020-09-14:
> Could you summarize what you think the error is? Is it still about residency?

Hard to tell, but it doesn't look like it. Even if I reverse the order of kernels, i.e. first the one taking an 'actual' pointer (which, I assumes, results in the driver making the memory resident), or if I explicitly use the memory residency APIs, the memory still isn't updated:

```julia
# version of the kernel that takes a value
julia> kernel(ptr, val) = (unsafe_store!(ptr, val); return)

julia> a = oneArray([0])
1-element oneArray{Int64,1}:
 0

# first the kernel that uses actual pointers
julia> @device_code_llvm @oneapi kernel(reinterpret(Core.LLVMPtr{Int,AS.Global}, pointer(a)), 1)
; CompilerJob of kernel kernel(Core.LLVMPtr{Int64,1}, Int64) for GPUCompiler.SPIRVCompilerTarget
;  @ REPL[16]:1 within `kernel'
define dso_local spir_kernel void @_Z17julia_kernel_32617LLVMPtrI5Int64Li1EES0_(i8 addrspace(1)*, i64) local_unnamed_addr {
top:
; ┌ @ /home/tim/Julia/pkg/LLVM/src/interop/pointer.jl:82 within `unsafe_store!' @ /home/tim/Julia/pkg/LLVM/src/interop/pointer.jl:82 @ /home/tim/Julia/pkg/LLVM/src/interop/pointer.jl:82
; │┌ @ /home/tim/Julia/pkg/LLVM/src/interop/pointer.jl:42 within `pointerset'
; ││┌ @ /home/tim/Julia/pkg/LLVM/src/interop/pointer.jl:42 within `macro expansion' @ /home/tim/Julia/pkg/LLVM/src/interop/base.jl:80
     %2 = bitcast i8 addrspace(1)* %0 to i64 addrspace(1)*
     store i64 %1, i64 addrspace(1)* %2, align 1
; └└└
  ret void
}

julia> a
1-element oneArray{Int64,1}:
 1

# then the one with pointer-as-integer
julia> @device_code_llvm @oneapi kernel(pointer(a), 2)
; CompilerJob of kernel kernel(Ptr{Int64}, Int64) for GPUCompiler.SPIRVCompilerTarget
;  @ REPL[16]:1 within `kernel'
define dso_local spir_kernel void @_Z17julia_kernel_33023PtrI5Int64ES0_(i64, i64) local_unnamed_addr {
top:
; ┌ @ pointer.jl:118 within `unsafe_store!' @ pointer.jl:118
   %2 = inttoptr i64 %0 to i64*
   store i64 %1, i64* %2, align 1
; └
  ret void
}

# not updated
julia> a
1-element oneArray{Int64,1}:
 1

# let's try the memory residency APIs, to be sure (they are already called implicitly)
julia> oneL0.make_resident(a.ctx, a.dev, a.buf)

julia> @oneapi kernel(pointer(a), 2)

# and synchronize, to be sure
julia> oneL0.synchronize(global_queue(a.ctx, a.dev))

# nothing...
julia> a
1-element oneArray{Int64,1}:
 1
```

jandres742, 2020-09-14:
@maleadt : If a reproducer is difficult to come by, could you share how to setup the julia enviroment?

maleadt, 2020-09-15:
Actually, this still reproduces with the L0 1.0 loader you posted above:

```
rm -f main && CPATH=~/.julia/artifacts/1bd904b7c25202d0863ecf68a61321bbf5b9f77e/include LIBRARY_PATH=~/.julia/artifacts/8d99145b2a13a9c0f96552a7b54623c57ae98920/lib make && LD_LIBRARY_PATH=~/.julia/artifacts/8d99145b2a13a9c0f96552a7b54623c57ae98920/lib:~/.julia/artifacts/fa0e77a91dff5a9dd3b600b796e31f6bd0c7e93e/lib64:~/.julia/artifacts/8e92378b7aea6d3a6e27c19dc1d614cc9e11dfe6/lib64:~/.julia/artifacts/8a14bac92eca3b88117b0c12b1f4f9ff23df72ff/lib ./main
clang++ -lze_loader main.cpp -o main
out[0] = 2
main: main.cpp:122: int main(): Assertion `out[i] == 42' failed.
zsh: abort      LD_LIBRARY_PATH= ./main
```

Those CPATH/LIBRARY_PATH/LD_LIBRARY_PATH variables might seem messy, but I list them explicitly to show how to manually use the binaries that we generate for oneAPI.jl. You can install them as follow:

```
# download Julia 1.5 from https://julialang.org/downloads/, extract and run the binary

$ julia

# press ]
(@v1.5) pkg> add NEO_jll oneAPI_Level_Zero_Headers_jll oneAPI_Level_Zero_Loader_jll
  Installed oneAPI_Level_Zero_Loader_jll ── v1.0.0+0
  Installed gmmlib_jll ──────────────────── v20.2.4+0
  Installed libigc_jll ──────────────────── v1.0.4756+0
  Installed NEO_jll ─────────────────────── v20.34.17727+1
  Installed oneAPI_Level_Zero_Headers_jll ─ v1.0.0+0

# press BACKSPACE to get to the normal REPL

julia> using NEO_jll, oneAPI_Level_Zero_Headers_jll, oneAPI_Level_Zero_Loader_jll

julia> println("CPATH=$(dirname(dirname(oneAPI_Level_Zero_Headers_jll.ze_api_path)))")
CPATH=~/.julia/artifacts/1bd904b7c25202d0863ecf68a61321bbf5b9f77e/include

julia> println("LIBRARY_PATH=$(dirname(oneAPI_Level_Zero_Loader_jll.libze_loader_path))")
LIBRARY_PATH=~/.julia/artifacts/8d99145b2a13a9c0f96552a7b54623c57ae98920/lib

julia> println("LD_LIBRARY_PATH=$(dirname(NEO_jll.libze_intel_gpu_path)):$(dirname(NEO_jll.libigc_jll.libigc_path)):$(dirname(NEO_jll.gmmlib_jll.libigdgmm_path))")
LD_LIBRARY_PATH=~/.julia/artifacts/fa0e77a91dff5a9dd3b600b796e31f6bd0c7e93e/lib64:~/.julia/artifacts/8e92378b7aea6d3a6e27c19dc1d614cc9e11dfe6/lib64:~/.julia/artifacts/8a14bac92eca3b88117b0c12b1f4f9ff23df72ff/lib
```

As you can see, these hashes are identical, since artifacts (such as the oneAPI/NEO/... binaries) are content-addressed.

Of course, at that point you can also just install oneAPI.jl and use the pure Julia reproducer I mentioned above:

```
$ julia

# press ]
(@v1.5) pkg> add https://github.com/JuliaGPU/oneAPI.jl
  [8f75cd03] + oneAPI v0.1.0 `https://github.com/JuliaGPU/oneAPI.jl#master`

# press BACKSPACE to get to the normal REPL

julia> using oneAPI

julia> kernel(ptr, val) = (unsafe_store!(ptr, val); return)
kernel (generic function with 1 method)

julia> a = oneArray([0])
1-element oneArray{Int64,1}:
 0

julia> @oneapi kernel(reinterpret(Core.LLVMPtr{Int,AS.Global}, pointer(a)), 1)

julia> a
1-element oneArray{Int64,1}:
 1

julia> @oneapi kernel(pointer(a), 2)

julia> a
1-element oneArray{Int64,1}:
 1
```

jandres742, 2020-09-26:
> Actually, this still reproduces with the L0 1.0 loader you posted above:

You mean you see my reproducer failing? Then we have some differences in setup. Can you point to the build release or commit you are using for L0?

maleadt, 2020-09-28:
It's in my comment:

```
  Installed oneAPI_Level_Zero_Loader_jll ── v1.0.0+0
  Installed gmmlib_jll ──────────────────── v20.2.4+0
  Installed libigc_jll ──────────────────── v1.0.4756+0
  Installed NEO_jll ─────────────────────── v20.34.17727+1
  Installed oneAPI_Level_Zero_Headers_jll ─ v1.0.0+
```

I'm only using tagged versions, and the build rules for, e.g., NEO are here: https://github.com/JuliaPackaging/Yggdrasil/blob/master/N/NEO/build_tarballs.jl
You could try downloading those binaries as explained in my previous post to see if it reproduces.

jandres742, 2020-11-08:
@maleadt : Could you confirm whether you are still seeing this problem? As before, I'm not able to reproduce it. I have tested it again with our latest release, https://github.com/intel/compute-runtime/releases/tag/20.44.18297, and have installed the following packages:

```shell
$ ls ./packages-public/
intel-gmmlib_20.3.2_amd64.deb  intel-igc-core_1.0.5353_amd64.deb  intel-igc-opencl_1.0.5353_amd64.deb  intel-level-zero-gpu_1.0.18297_amd64.deb  intel-ocloc_20.44.18297_amd64.deb  intel-opencl_20.44.18297_amd64.deb
```
Now, with the reproducer I posted previously, and your original kernel, posted at the beginning of this thread, I did the following in a Gen9 machine:

```shell
$ rm -rf ./*.bin ./*.spv ./*.gen
$ ocloc -device skl -file ./kernel.cl
Build succeeded.
$ mv ./kernel_Gen9core.spv ./kernel.spv
$ ./test.run
out[0] = 42
```

And results are correct.

maleadt, 2020-11-09:
Yes, this still perfectly reproduces.

```julia
julia> using oneAPI

julia> oneAPI.versioninfo()
Binary dependencies:
- NEO_jll: 20.44.18297+0
- libigc_jll: 1.0.5353+0
- gmmlib_jll: 20.3.2+0
- SPIRV_LLVM_Translator_jll: 9.0.0+1
- SPIRV_Tools_jll: 2020.2.0+1

Toolchain:
- Julia: 1.5.2
- LLVM: 9.0.1

1 driver:
- 00007f96-ffa0-ea10-1645-d40501000000 (v1.0.0, API v1.0.0)

1 device:
- Intel(R) Graphics Gen9 [0x1912]

julia> kernel(ptr, val) = (unsafe_store!(ptr, val); return)
kernel (generic function with 1 method)

julia> a = oneArray([0])
1-element oneArray{Int64,1}:
 0

julia> oneL0.make_resident(context(), device(), oneAPI.allocated[(context(),a.ptr)][1])

julia> @oneapi kernel(reinterpret(Core.LLVMPtr{Int,AS.Global}, pointer(a)), 1)

julia> a
1-element oneArray{Int64,1}:
 1

julia> @oneapi kernel(reinterpret(Core.LLVMPtr{Int,AS.Global}, pointer(a)), 2)

julia> a
1-element oneArray{Int64,1}:
 2

julia> @oneapi kernel(pointer(a), 3)

julia> a
1-element oneArray{Int64,1}:
 2

julia> @oneapi kernel(reinterpret(Core.LLVMPtr{Int,AS.Global}, pointer(a)), 4)

julia> a
1-element oneArray{Int64,1}:
 4
```

As you can see, invoking the kernel with a pointer-as-integer (what's returned by the `pointer` method by default) does not update the array. Casting it to an LLVM pointer works around the issue.

jandres742, 2020-11-09:
@maleadt : Thanks. Could you examine the reproducer I'm using (https://github.com/intel/compute-runtime/issues/320#issuecomment-691418157) and point out to the differences between the two? Maybe there's a difference there and that's why I'm not able to reproduce. Also, please notice I'm using ocloc to compile the kernel. is that what you are using?

maleadt, 2020-11-12:
> Also, please notice I'm using ocloc to compile the kernel. is that what you are using?

Then you're using the wrong SPIR-V code. The issue here is that the code Julia generates doesn't pass a pointer as `OpTypePointer`, but as `OpTypeInt` + `OpConvertUToPtr`; see the SPIR-V code I posted in the first post here. Using that with your C++ code reproduces the failure.

JacekDanecki, 2021-02-25:
Transferring this issue to IGC