intel/compute-runtime #323: 255 PCI lanes when unknown

bgoglin, 2020-06-04:
Hello

Playing with the sysman API on a laptop, I get the following PCI properties for my GPU: Gen=0 Lanes=255 maxBandwidth=0.

I am not sure why the Linux kernel doesn't report correct details for this link, but I have seen this quite often:
```
/sys/bus/pci/devices/0000:00:02.0$ cat max_link_*
Unknown speed
255
```

Anyway, sysman shouldn't return 255 here (it's likely -1 casted in u8). 0 would likely be better. Or document a special value for "unknown" for gen, lanes et bw.

Thanks

matcabral, 2020-06-05:
Hi @bgoglin,
 
You are correct, this is a bug. 255 is the enum for PCIE_LNK_WIDTH_UNKNOWN (https://elixir.bootlin.com/linux/latest/source/include/linux/pci.h#L234). We'll add a fix to detect this, and make with be 0 in when this happens. 

Thank you for reporting this.

matcabral, 2020-06-19:
This is already fixed: https://github.com/intel/compute-runtime/blob/master/level_zero/tools/source/sysman/pci/linux/os_pci_imp.cpp#L55

jandres742, 2020-07-27:
Closing. @bgoglin Feel free to comment if you still see issues.

closed 2020-07-27 by jandres742