intel/compute-runtime #325: I have installed latest Neo debs, but clinfo still only finds CPU device

boxerab, 2020-06-05:
I am on coffee lake with ubuntu 20, kernel 5.6.x.

boxerab, 2020-06-05:
It looks like I have an old driver:  ` 18.1.0.0920`:

```
Number of platforms                               1
  Platform Name                                   Intel(R) CPU Runtime for OpenCL(TM) Applications
  Platform Vendor                                 Intel(R) Corporation
  Platform Version                                OpenCL 2.1 LINUX
  Platform Profile                                FULL_PROFILE
  Platform Extensions                             cl_khr_icd cl_khr_global_int32_base_atomics cl_khr_global_int32_extended_atomics cl_khr_local_int32_base_atomics cl_khr_local_int32_extended_atomics cl_khr_byte_addressable_store cl_khr_depth_images cl_khr_3d_image_writes cl_intel_exec_by_local_thread cl_khr_spir cl_khr_fp64 cl_khr_image2d_from_buffer cl_intel_vec_len_hint 
  Platform Host timer resolution                  1ns
  Platform Extensions function suffix             INTEL

  Platform Name                                   Intel(R) CPU Runtime for OpenCL(TM) Applications
Number of devices                                 1
  Device Name                                     Intel(R) Core(TM) i7-8750H CPU @ 2.20GHz
  Device Vendor                                   Intel(R) Corporation
  Device Vendor ID                                0x8086
  Device Version                                  OpenCL 2.1 (Build 0)
  Driver Version                                  18.1.0.0920
  Device OpenCL C Version                         OpenCL C 2.0 
  Device Type                                     CPU
  Device Profile                                  FULL_PROFILE
  Device Available                                Yes
  Compiler Available                              Yes
  Linker Available                                Yes
  Max compute units                               12
  Max clock frequency                             2200MHz
  Device Partition                                (core)
    Max number of sub-devices                     12
    Supported partition types                     by counts, equally, by names (Intel)
    Supported affinity domains                    (n/a)
  Max work item dimensions                        3
  Max work item sizes                             8192x8192x8192
  Max work group size                             8192
  Preferred work group size multiple              128
```

JacekDanecki, 2020-06-08:
Can you provide file log.txt created by command:
```
strace -f -s 255 -o log.txt clinfo
```

boxerab, 2020-06-08:
[log.txt](https://github.com/intel/compute-runtime/files/4745940/log.txt)

boxerab, 2020-06-08:
Here is my log file:
[log.txt](https://github.com/intel/compute-runtime/files/4746042/log.txt)

JacekDanecki, 2020-06-08:
As I can see Neo was loaded:
```
openat(AT_FDCWD, "/usr/lib/x86_64-linux-gnu/intel-opencl/libigdrcl.so", O_RDONLY|O_CLOEXEC) = 5
```
then gmmlib was loaded from MSDK:
```
openat(AT_FDCWD, "/opt/intel/mediasdk/lib64/libigdgmm.so.11", O_RDONLY|O_CLOEXEC) = 5
```
But IGC libraries were not loaded. It looks like installed gmmlib from MSDK is not compatible with Neo library. Can you install gmmlib from Neo release?
What is the output from commands:
```
ldconfig -p | grep libigdgmm
LD_DEBUG=libs clinfo
```

boxerab, 2020-06-08:
```
$ ldconfig -p | grep libigdgmm

	libigdgmm.so.11 (libc6,x86-64) => /opt/intel/mediasdk/lib64/libigdgmm.so.11
	libigdgmm.so.11 (libc6,x86-64) => /usr/local/lib/libigdgmm.so.11
	libigdgmm.so.11 (libc6,x86-64) => /usr/lib/x86_64-linux-gnu/libigdgmm.so.11
	libigdgmm.so.11 (libc6) => /usr/lib/i386-linux-gnu/libigdgmm.so.11
	libigdgmm.so (libc6,x86-64) => /opt/intel/mediasdk/lib64/libigdgmm.so
```

boxerab, 2020-06-08:
[clinfo.txt](https://github.com/intel/compute-runtime/files/4746195/clinfo.txt)

boxerab, 2020-06-08:
I already have neo gmmlib installed.

boxerab, 2020-06-08:
btw, how can i uninstall media SDK? As I am not currently using it.

JacekDanecki, 2020-06-08:
You can rename directory /opt/intel/mediasdk, or remove/rename file in /etc/ld.so.conf.d directory containing path to mediasdk and run ldconfig as root.
It looks you've 3 gmmlib versions installed: from msdk, Ubuntu, and Neo release.
I don't see output from linker in clinfo.txt file you've provided. Are you using bash shell? Have you redirect output from standard output to clinfo.txt only? Can you run:
```
export LD_DEBUG=libs
 clinfo  > log.txt 2>&1
```
or set LD_DEBUG_OUTPUT to log.txt, see
```
LD_DEBUG=help clinfo
Valid options for the LD_DEBUG environment variable are:

  libs        display library search paths
  reloc       display relocation processing
  files       display progress for input file
  symbols     display symbol table processing
  bindings    display information about symbol binding
  versions    display version dependencies
  scopes      display scope information
  all         all previous options combined
  statistics  display relocation statistics
  unused      determined unused DSOs
  help        display this help message and exit

To direct the debugging output into a file instead of standard output
a filename can be specified using the LD_DEBUG_OUTPUT environment variable.

```

boxerab, 2020-06-08:
Thanks, if I rename the mediasdk folder, then cinfo won't run:

```
Abort was called at 42 line in file:
/build/intel-compute-runtime-kXIZhg/intel-compute-runtime-20.13.16352/shared/source/built_ins/built_ins.cpp
Aborted (core dumped)
```

boxerab, 2020-06-08:
Here is my clinfo with redirected stdout:
[log.txt](https://github.com/intel/compute-runtime/files/4746850/log.txt)

JacekDanecki, 2020-06-08:
You need to run "sudo ldconfig" to update linker cache, but
You've installed Neo 20.13.16352 from Ubuntu 20.04 in /usr/lib/x86_64-linux-gnu/intel-opencl directory and also Neo from github in /usr/local/lib/intel-opencl/libigdrcl.so

Packages in github releases have different layout than packages from Ubuntu, so you can't install them together. 
Please remove Ubuntu 20.04 packages:
-  intel-opencl-icd
- libigc1, libigdfcl1, libopencl-clang10
- libigdgmm11
and then install packages from github release:
- intel-opencl
- intel-igc-core, intel-igc-opencl
- intel-gmmlib

Or you can add [intel-opencl ppa](https://launchpad.net/~intel-opencl/+archive/ubuntu/intel-opencl) and update Neo packages using this repo. It'll remove github packages and update Ubuntu 20.04 packages:
```
apt-get install intel-opencl-icdReading package lists... Done
Building dependency tree       
Reading state information... Done
The following packages were automatically installed and are no longer required:
  libclang-cpp10 libllvm10 libllvmspirvlib10 libopencl-clang10
Use 'apt autoremove' to remove them.
The following additional packages will be installed:
  libigc
The following packages will be REMOVED:
  intel-igc-core intel-igc-opencl intel-level-zero-gpu intel-ocloc libigc1 libigdfcl1
The following NEW packages will be installed:
  libigc
The following packages will be upgraded:
  intel-opencl-icd
1 upgraded, 1 newly installed, 6 to remove and 183 not upgraded.
Need to get 35.0 MB of archives.
After this operation, 239 MB disk space will be freed.
Do you want to continue? [Y/n] 

```

boxerab, 2020-06-08:
Thanks, I installed from ppa and removed media sdk - now recognizing the GPU.
Will the ppa version work with media SDK ?

closed 2020-06-08 by boxerab

darkRaspberry, 2021-02-17:
@JacekDanecki Man you are nice! 
You saved my day!

> Packages in github releases have different layout than packages from Ubuntu, so you can't install them together.

This line solve my almost issue and finally run the DaVinci video editor FInally on my mesa.