intel/compute-runtime #326: Question: possible to test OpenCL 3.0 on Skylake via ForceOCLVersion?

oscarbg, 2020-06-12:
Hi,
just seeing latest release support OpenCL 3.0 on Tigerlake (unreleased hardware) have two questions:

1) can share if Intel is going to enable eventually OpenCL 3.0 on older GPU generations, like Gen9 (skylake)?

I say because at Khronos press release of OpenCL 3.0 it was said all features beyond 1.2 are now optional so really all implementations supporting OpenCL 1.2 should be able to expose OpenCL 3.0 like said Gen9..

2) even better yet.. there is some way to test OpenCL 3.0 right now on Gen9 or similar?
I say because there seems to be a ForceOCLVersion flag that seems to be an enviroment variable so running:
ForceOCLVersion=30 clinfo
should report OpenCL 3.0?
it is said in forums like:
https://software.intel.com/en-us/forums/opencl/topic/843037
but testing:
 ForceOCLVersion=12 clinfo
on current Neo doesn't seem to downgrade OCL version..
it's because requires to build Neo driver with in some debug mode that posted release builds aren't supporting?
also I see in:
https://gitlab.com/UbikBSD/WayKara/clComputeRuntime/commit/878fd43a1a9cbd61ea5ab066b684ee0e0860d781
that maybe should be possible to create a igdrcl.config file with:
ForceOCLVersion = 30
on program path and that should force OpenCL version too..
but also running clinfo with that file I see no changes to reported version..

hope you can solve my doubts..

thanks.. 
Oscar.

fhazubski-Intel, 2020-06-19:
Thank you for your interest in OpenCL 3.0.

1. We intend to switch all of the GPUs to OpenCL 3.0. However this cannot happen until the specification is finalized.

2. As the support for OpenCL 3.0 is taking shape in NEO, `ForceOCLVersion = 30` flag is expected to work with any device. However, as you suggested, this flag is for internal use only. In order to be able to use it you will have to build the driver yourself with build type set to `ReleaseInternal` or `Debug`. For more details please refer to [BUILD.md](https://github.com/intel/compute-runtime/blob/master/BUILD.md).

oscarbg, 2020-06-19:
thanks for info.. greatly appreciated..

closed 2020-06-19 by oscarbg

MichalMrozek, 2020-06-19:
You may also build release build with this CMAKE define to allow registry reads:
RELEASE_WITH_REGKEYS = 1