intel/compute-runtime #327: Valgrind mismatched free/delete/delete[]

boxerab, 2020-06-16:
I have the ppa version of the runtime, so this may be already fixed, but:

```
==34323== Mismatched free() / delete / delete []
==34323==    at 0x483CA3F: free (in /usr/lib/x86_64-linux-gnu/valgrind/vgpreload_memcheck-amd64-linux.so)
==34323==    by 0x1E77B37D: ??? (in /usr/lib/x86_64-linux-gnu/libigc.so.1.0.4062)
==34323==    by 0x1F00B925: ??? (in /usr/lib/x86_64-linux-gnu/libigc.so.1.0.4062)
==34323==    by 0x1F0D80C6: ??? (in /usr/lib/x86_64-linux-gnu/libigc.so.1.0.4062)
==34323==    by 0x1F0D7B52: ??? (in /usr/lib/x86_64-linux-gnu/libigc.so.1.0.4062)
==34323==    by 0x1F0DB319: ??? (in /usr/lib/x86_64-linux-gnu/libigc.so.1.0.4062)
==34323==    by 0x1F0DB700: llvm::ValueMapper::mapValue(llvm::Value const&) (in /usr/lib/x86_64-linux-gnu/libigc.so.1.0.4062)
==34323==    by 0x1EE38C21: llvm::IRMover::move(std::unique_ptr<llvm::Module, std::default_delete<llvm::Module> >, llvm::ArrayRef<llvm::GlobalValue*>, std::function<void (llvm::GlobalValue&, std::function<void (llvm::GlobalValue&)>)>, bool) (in /usr/lib/x86_64-linux-gnu/libigc.so.1.0.4062)
==34323==    by 0x1EE3EA3E: llvm::Linker::linkInModule(std::unique_ptr<llvm::Module, std::default_delete<llvm::Module> >, unsigned int, std::function<void (llvm::Module&, llvm::StringSet<llvm::MallocAllocator> const&)>) (in /usr/lib/x86_64-linux-gnu/libigc.so.1.0.4062)
==34323==    by 0x1E8AEA0E: ??? (in /usr/lib/x86_64-linux-gnu/libigc.so.1.0.4062)
==34323==    by 0x1F533240: llvm::legacy::PassManagerImpl::run(llvm::Module&) (in /usr/lib/x86_64-linux-gnu/libigc.so.1.0.4062)
==34323==    by 0x1E6EE452: ??? (in /usr/lib/x86_64-linux-gnu/libigc.so.1.0.4062)
==34323==  Address 0x17039400 is 0 bytes inside a block of size 8,192 alloc'd
==34323==    at 0x483BE63: operator new(unsigned long) (in /usr/lib/x86_64-linux-gnu/valgrind/vgpreload_memcheck-amd64-linux.so)
==34323==    by 0x1EE375D2: llvm::IRMover::move(std::unique_ptr<llvm::Module, std::default_delete<llvm::Module> >, llvm::ArrayRef<llvm::GlobalValue*>, std::function<void (llvm::GlobalValue&, std::function<void (llvm::GlobalValue&)>)>, bool) (in /usr/lib/x86_64-linux-gnu/libigc.so.1.0.4062)
==34323==    by 0x1EE3EA3E: llvm::Linker::linkInModule(std::unique_ptr<llvm::Module, std::default_delete<llvm::Module> >, unsigned int, std::function<void (llvm::Module&, llvm::StringSet<llvm::MallocAllocator> const&)>) (in /usr/lib/x86_64-linux-gnu/libigc.so.1.0.4062)
==34323==    by 0x1E8AEA0E: ??? (in /usr/lib/x86_64-linux-gnu/libigc.so.1.0.4062)
==34323==    by 0x1F533240: llvm::legacy::PassManagerImpl::run(llvm::Module&) (in /usr/lib/x86_64-linux-gnu/libigc.so.1.0.4062)
==34323==    by 0x1E6EE452: ??? (in /usr/lib/x86_64-linux-gnu/libigc.so.1.0.4062)
==34323==    by 0x1E6EE942: ??? (in /usr/lib/x86_64-linux-gnu/libigc.so.1.0.4062)
==34323==    by 0x1E6D0EAF: ??? (in /usr/lib/x86_64-linux-gnu/libigc.so.1.0.4062)
==34323==    by 0x1E78A2D8: ??? (in /usr/lib/x86_64-linux-gnu/libigc.so.1.0.4062)
==34323==    by 0x1E78C7E2: ??? (in /usr/lib/x86_64-linux-gnu/libigc.so.1.0.4062)
==34323==    by 0x16781E5E: ??? (in /usr/lib/x86_64-linux-gnu/intel-opencl/libigdrcl.so)
==34323==    by 0x1680BC95: ??? (in /usr/lib/x86_64-linux-gnu/intel-opencl/libigdrcl.so)
```

JacekDanecki, 2020-06-16:
Issues reported by valgrind in IGC were fixed o IGC side, see #305.
ppa packages currently use IGC 1.0.4062, so they don't have fix yet.
Can you confirm [IGC commit](https://github.com/intel/intel-graphics-compiler/commit/241078902a30eca9c7ad687f1191e4f3cd1da305) fixes this issue on your side?

boxerab, 2020-06-17:
Thanks, I will just wait for a ppa package update.

closed 2020-06-17 by boxerab