intel/compute-runtime #328: Conditional jump or move depends on uninitialised value(s)

boxerab, 2020-06-16:
```
34323== Conditional jump or move depends on uninitialised value(s)
==34323==    at 0x24E47F0C: Intel::OpenCL::DeviceBackend::Utils::CPUDetect::CPUDetect() (CPUDetect.cpp:149)
==34323==    by 0x24CF66C4: GetInstance (CPUDetect.h:20)
==34323==    by 0x24CF66C4: Intel::OpenCL::DeviceBackend::Utils::GetOrDetectCpuId(std::string const&) (CPUCompiler.cpp:65)
==34323==    by 0x24CF76F2: Intel::OpenCL::DeviceBackend::CPUCompiler::SelectCpu(std::string const&, std::string const&) (CPUCompiler.cpp:258)
==34323==    by 0x24CF82BC: Intel::OpenCL::DeviceBackend::CPUCompiler::CPUCompiler(Intel::OpenCL::DeviceBackend::ICompilerConfig const&) (CPUCompiler.cpp:216)
==34323==    by 0x24CFAC8A: Intel::OpenCL::DeviceBackend::CPUProgramBuilder::CPUProgramBuilder(Intel::OpenCL::DeviceBackend::IAbstractBackendFactory*, Intel::OpenCL::DeviceBackend::ICompilerConfig const&) (CPUProgramBuilder.cpp:48)
==34323==    by 0x24CF6083: Intel::OpenCL::DeviceBackend::CPUCompileService::CPUCompileService(Intel::OpenCL::DeviceBackend::ICompilerConfig const&) (CPUCompileService.cpp:37)
==34323==    by 0x24D1D63A: Intel::OpenCL::DeviceBackend::ServiceFactory::GetCompilationService(Intel::OpenCL::DeviceBackend::ICLDevBackendOptions const*, Intel::OpenCL::DeviceBackend::ICLDevBackendCompilationService**) (ServiceFactory.cpp:99)
==34323==    by 0x162CEC0E: Intel::OpenCL::CPUDevice::ProgramService::Init() (program_service.cpp:126)
==34323==    by 0x162C7F4F: Intel::OpenCL::CPUDevice::CPUDevice::Init() (cpu_device.cpp:333)
==34323==    by 0x162C854C: Intel::OpenCL::CPUDevice::CPUDevice::clDevGetInstance(unsigned int, IOCLFrameworkCallbacks*, IOCLDevLogDescriptor*, cl_dev_err_code&) (cpu_device.cpp:246)
==34323==    by 0x162C85F2: clDevCreateDeviceInstance (cpu_device.cpp:558)
==34323==    by 0x159780E9: Intel::OpenCL::Framework::Device::CreateInstance() (Device.cpp:330)
==34323== 
==34323== Conditional jump or move depends on uninitialised value(s)
==34323==    at 0x24E47F1D: Intel::OpenCL::DeviceBackend::Utils::CPUDetect::CPUDetect() (CPUDetect.cpp:155)
==34323==    by 0x24CF66C4: GetInstance (CPUDetect.h:20)
==34323==    by 0x24CF66C4: Intel::OpenCL::DeviceBackend::Utils::GetOrDetectCpuId(std::string const&) (CPUCompiler.cpp:65)
==34323==    by 0x24CF76F2: Intel::OpenCL::DeviceBackend::CPUCompiler::SelectCpu(std::string const&, std::string const&) (CPUCompiler.cpp:258)
==34323==    by 0x24CF82BC: Intel::OpenCL::DeviceBackend::CPUCompiler::CPUCompiler(Intel::OpenCL::DeviceBackend::ICompilerConfig const&) (CPUCompiler.cpp:216)
==34323==    by 0x24CFAC8A: Intel::OpenCL::DeviceBackend::CPUProgramBuilder::CPUProgramBuilder(Intel::OpenCL::DeviceBackend::IAbstractBackendFactory*, Intel::OpenCL::DeviceBackend::ICompilerConfig const&) (CPUProgramBuilder.cpp:48)
==34323==    by 0x24CF6083: Intel::OpenCL::DeviceBackend::CPUCompileService::CPUCompileService(Intel::OpenCL::DeviceBackend::ICompilerConfig const&) (CPUCompileService.cpp:37)
==34323==    by 0x24D1D63A: Intel::OpenCL::DeviceBackend::ServiceFactory::GetCompilationService(Intel::OpenCL::DeviceBackend::ICLDevBackendOptions const*, Intel::OpenCL::DeviceBackend::ICLDevBackendCompilationService**) (ServiceFactory.cpp:99)
==34323==    by 0x162CEC0E: Intel::OpenCL::CPUDevice::ProgramService::Init() (program_service.cpp:126)
==34323==    by 0x162C7F4F: Intel::OpenCL::CPUDevice::CPUDevice::Init() (cpu_device.cpp:333)
==34323==    by 0x162C854C: Intel::OpenCL::CPUDevice::CPUDevice::clDevGetInstance(unsigned int, IOCLFrameworkCallbacks*, IOCLDevLogDescriptor*, cl_dev_err_code&) (cpu_device.cpp:246)
==34323==    by 0x162C85F2: clDevCreateDeviceInstance (cpu_device.cpp:558)
==34323==    by 0x159780E9: Intel::OpenCL::Framework::Device::CreateInstance() (Device.cpp:330)
==34323== 
==34323== Conditional jump or move depends on uninitialised value(s)
==34323==    at 0x24E47F3E: Intel::OpenCL::DeviceBackend::Utils::CPUDetect::CPUDetect() (CPUDetect.cpp:165)
==34323==    by 0x24CF66C4: GetInstance (CPUDetect.h:20)
==34323==    by 0x24CF66C4: Intel::OpenCL::DeviceBackend::Utils::GetOrDetectCpuId(std::string const&) (CPUCompiler.cpp:65)
==34323==    by 0x24CF76F2: Intel::OpenCL::DeviceBackend::CPUCompiler::SelectCpu(std::string const&, std::string const&) (CPUCompiler.cpp:258)
==34323==    by 0x24CF82BC: Intel::OpenCL::DeviceBackend::CPUCompiler::CPUCompiler(Intel::OpenCL::DeviceBackend::ICompilerConfig const&) (CPUCompiler.cpp:216)
==34323==    by 0x24CFAC8A: Intel::OpenCL::DeviceBackend::CPUProgramBuilder::CPUProgramBuilder(Intel::OpenCL::DeviceBackend::IAbstractBackendFactory*, Intel::OpenCL::DeviceBackend::ICompilerConfig const&) (CPUProgramBuilder.cpp:48)
==34323==    by 0x24CF6083: Intel::OpenCL::DeviceBackend::CPUCompileService::CPUCompileService(Intel::OpenCL::DeviceBackend::ICompilerConfig const&) (CPUCompileService.cpp:37)
==34323==    by 0x24D1D63A: Intel::OpenCL::DeviceBackend::ServiceFactory::GetCompilationService(Intel::OpenCL::DeviceBackend::ICLDevBackendOptions const*, Intel::OpenCL::DeviceBackend::ICLDevBackendCompilationService**) (ServiceFactory.cpp:99)
==34323==    by 0x162CEC0E: Intel::OpenCL::CPUDevice::ProgramService::Init() (program_service.cpp:126)
==34323==    by 0x162C7F4F: Intel::OpenCL::CPUDevice::CPUDevice::Init() (cpu_device.cpp:333)
==34323==    by 0x162C854C: Intel::OpenCL::CPUDevice::CPUDevice::clDevGetInstance(unsigned int, IOCLFrameworkCallbacks*, IOCLDevLogDescriptor*, cl_dev_err_code&) (cpu_device.cpp:246)
==34323==    by 0x162C85F2: clDevCreateDeviceInstance (cpu_device.cpp:558)
==34323==    by 0x159780E9: Intel::OpenCL::Framework::Device::CreateInstance() (Device.cpp:330)
==34323== 
==34323== Conditional jump or move depends on uninitialised value(s)
==34323==    at 0x24E47F4C: Intel::OpenCL::DeviceBackend::Utils::CPUDetect::CPUDetect() (CPUDetect.cpp:170)
==34323==    by 0x24CF66C4: GetInstance (CPUDetect.h:20)
==34323==    by 0x24CF66C4: Intel::OpenCL::DeviceBackend::Utils::GetOrDetectCpuId(std::string const&) (CPUCompiler.cpp:65)
==34323==    by 0x24CF76F2: Intel::OpenCL::DeviceBackend::CPUCompiler::SelectCpu(std::string const&, std::string const&) (CPUCompiler.cpp:258)
==34323==    by 0x24CF82BC: Intel::OpenCL::DeviceBackend::CPUCompiler::CPUCompiler(Intel::OpenCL::DeviceBackend::ICompilerConfig const&) (CPUCompiler.cpp:216)
==34323==    by 0x24CFAC8A: Intel::OpenCL::DeviceBackend::CPUProgramBuilder::CPUProgramBuilder(Intel::OpenCL::DeviceBackend::IAbstractBackendFactory*, Intel::OpenCL::DeviceBackend::ICompilerConfig const&) (CPUProgramBuilder.cpp:48)
==34323==    by 0x24CF6083: Intel::OpenCL::DeviceBackend::CPUCompileService::CPUCompileService(Intel::OpenCL::DeviceBackend::ICompilerConfig const&) (CPUCompileService.cpp:37)
==34323==    by 0x24D1D63A: Intel::OpenCL::DeviceBackend::ServiceFactory::GetCompilationService(Intel::OpenCL::DeviceBackend::ICLDevBackendOptions const*, Intel::OpenCL::DeviceBackend::ICLDevBackendCompilationService**) (ServiceFactory.cpp:99)
==34323==    by 0x162CEC0E: Intel::OpenCL::CPUDevice::ProgramService::Init() (program_service.cpp:126)
==34323==    by 0x162C7F4F: Intel::OpenCL::CPUDevice::CPUDevice::Init() (cpu_device.cpp:333)
==34323==    by 0x162C854C: Intel::OpenCL::CPUDevice::CPUDevice::clDevGetInstance(unsigned int, IOCLFrameworkCallbacks*, IOCLDevLogDescriptor*, cl_dev_err_code&) (cpu_device.cpp:246)
==34323==    by 0x162C85F2: clDevCreateDeviceInstance (cpu_device.cpp:558)
==34323==    by 0x159780E9: Intel::OpenCL::Framework::Device::CreateInstance() (Device.cpp:330)
==34323== 
==34323== Conditional jump or move depends on uninitialised value(s)
==34323==    at 0x24CF7814: Intel::OpenCL::DeviceBackend::CPUCompiler::SelectCpu(std::string const&, std::string const&) (CPUCompiler.cpp:296)
==34323==    by 0x24CF82BC: Intel::OpenCL::DeviceBackend::CPUCompiler::CPUCompiler(Intel::OpenCL::DeviceBackend::ICompilerConfig const&) (CPUCompiler.cpp:216)
==34323==    by 0x24CFAC8A: Intel::OpenCL::DeviceBackend::CPUProgramBuilder::CPUProgramBuilder(Intel::OpenCL::DeviceBackend::IAbstractBackendFactory*, Intel::OpenCL::DeviceBackend::ICompilerConfig const&) (CPUProgramBuilder.cpp:48)
==34323==    by 0x24CF6083: Intel::OpenCL::DeviceBackend::CPUCompileService::CPUCompileService(Intel::OpenCL::DeviceBackend::ICompilerConfig const&) (CPUCompileService.cpp:37)
==34323==    by 0x24D1D63A: Intel::OpenCL::DeviceBackend::ServiceFactory::GetCompilationService(Intel::OpenCL::DeviceBackend::ICLDevBackendOptions const*, Intel::OpenCL::DeviceBackend::ICLDevBackendCompilationService**) (ServiceFac [... cut ...]
==34323==    by 0x1593916B: Intel::OpenCL::Framework::Context::Context(long const*, unsigned int, unsigned int, Intel::OpenCL::Utils::SharedPtr<Intel::OpenCL::Framework::FissionableDevice>*, void (*)(char const*, void const*, unsigned long, void*), void*, int*, ocl_entry_points*, ocl_gpa_data*, Intel::OpenCL::Framework::ContextModule&) (Context.cpp:233)
==34323==    by 0x159132AA: Allocate (Context.h:90)
==34323==    by 0x159132AA: Intel::OpenCL::Framework::ContextModule::CreateContext(long const*, unsigned int, _cl_device_id* const*, void (*)(char const*, void const*, unsigned long, void*), void*, int*) (context_module.cpp:412)
==34323== 
```

JacekDanecki, 2020-06-16:
This issue looks like problem with CPU OpenCL implementation, not Neo, can you create issue in [OpenCL forum](https://software.intel.com/en-us/forums/opencl) intead?

boxerab, 2020-06-17:
Thanks, sorry for the noise :)

closed 2020-06-17 by boxerab