intel/compute-runtime #331: Build on Alpine Linux fails due to missing `-lexecinfo`

krupenik, 2020-06-20:
```
[ 30%] Building CXX object opencl/test/unit_test/offline_compiler/CMakeFiles/ocloc_tests.dir/__/__/__/__/shared/offline_compiler/source/utilities/linux/get_current_dir_linux.cpp.o
[ 30%] Building CXX object opencl/test/unit_test/offline_compiler/CMakeFiles/ocloc_tests.dir/__/__/__/__/shared/source/os_interface/linux/os_thread_linux.cpp.o
[ 30%] Linking CXX executable ../../../../bin/ocloc_tests
/usr/lib/gcc/x86_64-alpine-linux-musl/9.3.0/../../../../x86_64-alpine-linux-musl/bin/ld: CMakeFiles/ocloc_tests.dir/__/__/__/__/shared/offline_compiler/source/utilities/linux/safety_caller_linux.cpp.o: in function `SafetyGuardLinux::sigAction(int, siginfo_t*, void*)':
safety_caller_linux.cpp:(.text._ZN16SafetyGuardLinux9sigActionEiP9siginfo_tPv[_ZN16SafetyGuardLinux9sigActionEiP9siginfo_tPv]+0x46): undefined reference to `backtrace'
/usr/lib/gcc/x86_64-alpine-linux-musl/9.3.0/../../../../x86_64-alpine-linux-musl/bin/ld: safety_caller_linux.cpp:(.text._ZN16SafetyGuardLinux9sigActionEiP9siginfo_tPv[_ZN16SafetyGuardLinux9sigActionEiP9siginfo_tPv]+0x5d): undefined reference to `backtrace_symbols'
collect2: error: ld returned 1 exit status
make[2]: *** [opencl/test/unit_test/offline_compiler/CMakeFiles/ocloc_tests.dir/build.make:795: bin/ocloc_tests] Error 1
make[2]: Leaving directory '/home/dennis/aports/testing/intel-compute-runtime/src/compute-runtime-20.23.16988/build'
make[1]: *** [CMakeFiles/Makefile2:9367: opencl/test/unit_test/offline_compiler/CMakeFiles/ocloc_tests.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
```

PR: https://github.com/intel/compute-runtime/pull/330

Odianosen25, 2020-07-05:
@krupenik,

Please can you help with the cmake command for compiling this on Alpine Linux? I wish to compile it within a docker environment so I can utilize it.

Regards

krupenik, 2020-07-05:
@Odianosen25 Please see https://gitlab.alpinelinux.org/alpine/aports/-/merge_requests/9623

Odianosen25, 2020-07-05:
Hello  @krupenik,

Thanks for this, but I know not how to install it from there. Besides it seems its still in development. I don't mind testing it as I need it asap. I know not how to use GitLab so if wanting to  install it, what's the procedure like?

Thanks

krupenik, 2020-07-07:
@Odianosen25 best way to do it would be to grab the whole branch and add all those aports to your local tree

JablonskiMateusz, 2023-01-13:
It should be fixed by #330  - waiting for PR author

krupenik, 2023-01-20:
Updated #330, but building openvino on Alpine these days is quite tricky due to llvm versions drift

kgibala, 2026-01-09:
Hi @krupenik,

We’d like to know if this issue is still affecting you. If so, please provide an update or any additional information in the mentioned PR #330. Otherwise, we’ll close this issue after 30 days of inactivity. Your feedback is appreciated.