intel/compute-runtime #332: Latest khronos OpenCL-CTS (master) fails in test_api test.

llmike, 2020-07-02:
queue_properties...
Queue property NULL. Testing ...
ERROR: clEnqueueNDRangeKernel failed.! (CL_INVALID_COMMAND_QUEUE from /data/svn/scratch/mgorchak/opencl/OpenCL-CTS/qnx_build/test_api/nto/x86_64/o/../../../../../test_confor
ERROR: enqueue_kernel failed! (CL_INVALID_COMMAND_QUEUE from /data/svn/scratch/mgorchak/opencl/OpenCL-CTS/qnx_build/test_api/nto/x86_64/o/../../../../../test_conformance/api
queue_properties FAILED

They passed NULL as properties to the clCreateCommandQueueWithProperties() call. NULL is supported by specification 2.1 and 1.2 + KHR extension via clCreateCommandQueueWithPropertiesKHR() call.

llmike, 2020-07-02:
It was tested on Intel ApolloLake platform (MRB) which exposes OpenCL 1.2 API.

JacekDanecki, 2020-07-03:
We've executed "test_api queue_properties" based on [commit 105a3aff62799c498f27a644564c82b0f9fb562c](https://github.com/KhronosGroup/OpenCL-CTS/commit/105a3aff62799c498f27a644564c82b0f9fb562c) on all platforms and they passed.
Can you provide more details about reproduction steps?

llmike, 2020-07-07:
Platform is important in this case I think. Did you test it on ApolloLake where OpenCL 1.2 is provided instead of OpenCL 2.1?

JacekDanecki, 2020-07-08:
Yes we run also on APL (device ID: 0x5A84):

``` Initializing random seed to 0.
Requesting GPU device based on command line for platform index 0 and device index 0
Compute Device Name = Intel(R) Gen9 HD Graphics NEO, Compute Device Vendor = Intel(R) Corporation, Compute Device Version = OpenCL 1.2 NEO , CL C Version = OpenCL C 1.2 
Supports single precision denormals: YES
sizeof( void*) = 8  (host)
sizeof( void*) = 4  (device)
queue_properties...
Queue property NULL. Testing ... 
Queue property CL_QUEUE_OUT_OF_ORDER_EXEC_MODE_ENABLE supported. Testing ... 
Queue property CL_QUEUE_PROFILING_ENABLE supported. Testing ... 
Queue property CL_QUEUE_OUT_OF_ORDER_EXEC_MODE_ENABLE & CL_QUEUE_PROFILING_ENABLE supported. Testing ... 
queue_properties passed
PASSED sub-test.
PASSED test.

```

llmike, 2020-07-09:
My apologies, guys! It appears that there was active another layer between before ICD loader and Intel OpenCL implementation. Which comes unnoticed initially and which acts in some strange ways in some cases. This was one of such cases.

closed 2020-07-09 by llmike