intel/compute-runtime #333: Building for Alpine Linux Docker

Odianosen25, 2020-07-05:
Hello,

Please I use OpenCV for a project we working on, and we want to use OpenCL to enhance our code. We want to build the OpenCV library using `WITH_OPENCL=ON` so we can use it as backend, but it doesn't build as we understand we need this OpenCL runtime package. Please how can we build this locally as can you help with the cmake instructions?

Kind Regards

alalek, 2020-07-05:
These lines are enough in OpenCV's CMake output summary to build OpenCV with OpenCL support:

```
--   OpenCL:                        YES (no extra features)
--     Include path:                <...opencv...>/3rdparty/include/opencl/1.2
--     Link libraries:              Dynamic load
```

OpenCV doesn't use or link with any OpenCL SDK (by default). Dynamic loading is used instead.
Refer to related Wiki page: https://github.com/opencv/opencv/wiki/OpenCL-optimizations

Ensure that you have properly configured OpenCL subsystem in your environment by using of `clinfo` tool.

Odianosen25, 2020-07-05:
@alalek,

Thanks for the feedback. I do have those lines as shown below

```
OpenCL:                      YES (no extra features)
Include path:                /tmp/opencv-4.3.0/3rdparty/include/opencl/1.2
Link libraries:              Dynamic load
```
But when I run `cv2.ocl.haveOpenCL()` it returns `False`. So when I try to run a code using UMat, it obviously shows no difference that its of any use. I also installed OpenCV on Ubuntu and I have the same issue. The only environment I get a `True` is in my Windows Laptop .

So I am  trying to figure out why. I got to I am supported to have the Intel runtime in it, which I suspect supported to be in a directory `/opt/intel` but nothing there. So why I asked for help

Kind regards

alalek, 2020-07-05:
Looks good.

> But when I run `cv2.ocl.haveOpenCL()` it returns `False`

This means that OpenCV didn't find OpenCL device (or can't create it).

Next investigation steps:
- run `clinfo` tool (sources: https://github.com/Oblomov/clinfo if your Linux doesn't provide this package)
- build and run `opencv_version --opencl` to query all "visible" OpenCL devices.

BTW, If `clinfo` returns nothing then OpenCL subsystem is not configured properly. There are several issues in this repository how to fix that (search for `lspci`).

Odianosen25, 2020-07-05:
Hello @alalek,

Thanks for taking your time to respond to my query. I ran `clinfo` and got `0` and `opencv_version --opencl` gives me `OpenCL is disabled`.

Will see what I can find then get back.

Kind regards

Odianosen25, 2020-07-05:
@alalek,

I ran `lspci` and all I got was 

```
bash-5.0# lspci
03:00.0 Class 0108: 1987:5008
00:17.0 Class 0106: 8086:9d03
00:1f.2 Class 0580: 8086:9d21
00:1c.0 Class 0604: 8086:9d14
00:08.0 Class 0880: 8086:1911
00:1f.0 Class 0601: 8086:9d4e
02:00.0 Class 0200: 10ec:8168
00:1c.5 Class 0604: 8086:9d15
00:16.0 Class 0780: 8086:9d3a
01:00.0 Class 0280: 8086:3165
00:1f.3 Class 0403: 8086:9d71
00:00.0 Class 0600: 8086:5914
00:1d.0 Class 0604: 8086:9d18
00:14.2 Class 1180: 8086:9d31
00:02.0 Class 0300: 8086:5917
00:14.0 Class 0c03: 8086:9d2f
00:1f.4 Class 0c05: 8086:9d23

``` 
I don't even understand what it all means

alalek, 2020-07-05:
> 00:02.0 Class 0300: 8086:5917

Looks like it is VGA device. So iGPU is not blocked by BIOS.

- `sudo clinfo`
- if doesn't work above, then `strace -s 255 -f -o log.txt clinfo` and attach log here
- `dmesg | grep -E 'drm|i915'`

Odianosen25, 2020-07-05:
Oh thanks @alalek,

First if I may add I am operating from within a docker container, and know not if it affects my results.

[log.txt](https://github.com/intel/compute-runtime/files/4875364/log.txt)

`dmesg | grep -E 'drm|i915'` gave nothing.

Regards

alalek, 2020-07-05:
> log.txt

`ls -al /etc/OpenCL/vendors` ?

If it is empty then NEO OpenCL runtime is not installed.

> `dmesg | grep -E 'drm|i915'` gave nothing.
> docker

Run this on the host system.

> docker container

Docker virtualizes CPU/memory/storage/network by default. For using iGPU extra steps must be performed.

In ubuntu->ubuntu docker configuration it is enough to pass this device: `--device=/dev/dri:/dev/dri` (for docker run/create commands, `docker build` can't not work with devices).

Odianosen25, 2020-07-05:
> ls -al /etc/OpenCL/vendors ?

`ls: /etc/OpenCL/vendors: No such file or directory`

> If it is empty then NEO OpenCL runtime is not installed.

Well yeah and why I was thinking of installing this within my container if I can figure out how. Unless installing it on the host, also makes it pass unto the docker container

I do run my container using host and `--privileged mode` to ensure everything is passed to it. But still nothing

Regards

Odianosen25, 2020-07-06:
@alalek Good morning,

If I may add also, I do use hardware acceleration within my docker container for other video related stuffs using ffmpeg. So reaching the hardware device is not an issue. I just wish I understood why OpenCL refuses to compile in OpenCV, not even the one installed on Ubuntu (though that was via a pre-built binary which I got when I installed the python package), but Windows where I installed the same package works great.

And something interesting I found out

on  my host where I also have OpenCV installed when I run `opencv_version --opencl`. I just get
```
$ opencv_version --opencl
3.2.0
```

While in my  docker container it gives me
```
# opencv_version --opencl
4.3.0
OpenCL is disabled
```

There is no link to  opencl in  the host, though maybe its not just built in it anyway

alalek, 2020-07-06:
> OpenCL refuses to compile in OpenCV

As wrote above:
- OpenCV doesn't need any OpenCL SDK to compile with OpenCL support
- even `clinfo` doesn't work in your environment.

You need to fix OpenCL subsystem first (build/install NEO OpenCL runtime in docker container, configure user rights, etc).

---

> `$ opencv_version --opencl`
> `3.2.0`

3.2.0 doesn't support [`--opencl` in opencv_version](https://github.com/opencv/opencv/commit/01f4a173aba2882eaec6b6249ae2895bc3624dcc#diff-44099e0a3dcf7f9b9abfa9c0dfff7e6eR22) and just ignores this flag.

Odianosen25, 2020-07-06:
Ah ok @alalek thanks for the clarification, so need to figure out how to install the runtime 😬. Don’t know how to do that yet I’m docket, but studying. I have messaged one of the devs here https://github.com/JacekDanecki/neo-hub/issues/5 so hopefully he will respond.

Thanks for the patience and all. Kind regards

Odianosen25, 2020-09-11:
@alalek,

I can confirm the runtime is now available as an Alpine package, and I have been installing it. I have been able to build Intel media SDK with it, and I believe it went fine.

I am looking to close this issue, since by the package being available, I think this can officially be closed. but I got another issue,  and I am not sure where to report it, and will be nice if you could point me in the right direction please.

The other issue I have now, is that Intel OpenCL SDK doesn't seem to be compiled for some reasons, and though I have installed all available OpenCL headers within the Alpine packages (and these where built like since 2016, so very old), when building my OpenCV, it seems to be looking for a file `CL/va_ext.h`. Also `clinfo` gives me no platforms within Alpine.

I had reported it as an OpenCV issue [here](https://github.com/opencv/opencv/issues/18307#issuecomment-690281210), but I am not sure where exactly I am to report this issue, or what I might be doing wrong. Been told to install Intel OpenCL SDK, but there seem to be none for Alpine, nor how to compile it. 

As I was told in the OpenCV issue, I downloaded the Intel OpenCL SDK (only options are for Centos and Ubuntu) and installed it in my Ubuntu server and `clinfo` gave me an output, but not for Alpine after installing the new packages.

Will be very grateful please, if I could be assisted with this. And thanks for the help so far.

Kind regards

Odianosen25, 2020-09-11:
One more thing that has just got to my attention, the OpenCL headers in Alpine has not the file `cl_va_api_media_sharing_intel.h`, which apparently is not needed as replace for `CL/va_ext.h` as indicated in this patch for ffmpeg https://patchwork.ffmpeg.org/project/ffmpeg/patch/1548141337-32109-1-git-send-email-ruiling.song@intel.com/. 

How do I get my hands on it?

Regards

closed 2020-09-11 by Odianosen25