intel/compute-runtime #334: intel-opencl docker images do not contain oneAPI loader

maleadt, 2020-07-07:
https://github.com/intel/compute-runtime/issues/319#issuecomment-635939207 mentioned that oneAPI Level Zero was added to the image, but it only seems like `libze_intel_gpu.so.0.8` is there, and not `libze_loader.so`:

```
# find / | grep libze
/usr/share/doc/libze-intel-gpu
/usr/share/doc/libze-intel-gpu/copyright
/usr/share/doc/libze-intel-gpu/changelog.Debian.gz
/usr/lib/x86_64-linux-gnu/libze_intel_gpu.so.0.8.17111
/usr/lib/x86_64-linux-gnu/libze_intel_gpu.so.0.8
/var/lib/dpkg/info/libze-intel-gpu.list
/var/lib/dpkg/info/libze-intel-gpu.triggers
/var/lib/dpkg/info/libze-intel-gpu.md5sums
/var/lib/dpkg/info/libze-intel-gpu.shlibs
```

JacekDanecki, 2020-07-07:
With [commit](https://github.com/JacekDanecki/neo-hub/commit/e2b716f4231c409cff918ce8975edfb0078a7f80) Level Zero packages were added to docker images.
Example:
```
$ docker run -it --device /dev/dri:/dev/dri --rm docker.io/intelopencl/intel-opencl:ubuntu-18.04-ppa bash
root@91ff598987f8:/# find / | grep libze
/usr/share/doc/libze-intel-gpu
/usr/share/doc/libze-intel-gpu/copyright
/usr/share/doc/libze-intel-gpu/changelog.Debian.gz
/usr/share/doc/libze-loader
/usr/share/doc/libze-loader/copyright
/usr/share/doc/libze-loader/changelog.Debian.gz
/usr/lib/x86_64-linux-gnu/libze_validation_layer.so.0.91
/usr/lib/x86_64-linux-gnu/libze_intel_gpu.so.0.8.17111
/usr/lib/x86_64-linux-gnu/libze_loader.so.0.91
/usr/lib/x86_64-linux-gnu/libze_intel_gpu.so.0.8
/usr/lib/x86_64-linux-gnu/libze_loader.so.0.91.10
/usr/lib/x86_64-linux-gnu/libze_validation_layer.so.0.91.10
/var/lib/dpkg/info/libze-loader.triggers
/var/lib/dpkg/info/libze-loader.md5sums
/var/lib/dpkg/info/libze-loader.shlibs
/var/lib/dpkg/info/libze-intel-gpu.shlibs
/var/lib/dpkg/info/libze-intel-gpu.list
/var/lib/dpkg/info/libze-loader.list
/var/lib/dpkg/info/libze-intel-gpu.md5sums
/var/lib/dpkg/info/libze-intel-gpu.triggers
```

closed 2020-07-07 by JacekDanecki

maleadt, 2020-07-07:
Awesome, thanks!