intel/compute-runtime #335: Passing generic pointer to kernel crashes zeKernelSetArgumentValue

maleadt, 2020-07-07:
I was trying to pass a generic pointer to a SPIR-V kernel (which I learned later is not allowed in OpenCL, not sure about SPIR-V):

```julia
function debug_kernel(ptr)
    i = get_global_id()
    unsafe_store!(ptr, 42, i)
    return
end
```

```llvm
define dso_local spir_kernel void @_Z23julia_debug_kernel_17647LLVMPtrI7Float32Li4EE(i8 addrspace(4)*) local_unnamed_addr {
top:
   %1 = call i64 @_Z13get_global_idj(i32 0)
      %2 = add i64 %1, -1
     %3 = bitcast i8 addrspace(4)* %0 to float addrspace(4)*
     %4 = getelementptr inbounds float, float addrspace(4)* %3, i64 %2
     store float 4.200000e+01, float addrspace(4)* %4, align 1
  ret void
}
```

```spirv
; SPIR-V
; Version: 1.0
; Generator: Khronos LLVM/SPIR-V Translator; 14
; Bound: 22
; Schema: 0
               OpCapability Addresses
               OpCapability Linkage
               OpCapability Kernel
               OpCapability Int64
               OpCapability GenericPointer
               OpCapability Int8
          %1 = OpExtInstImport "OpenCL.std"
               OpMemoryModel Physical64 OpenCL
               OpEntryPoint Kernel %10 "_Z23julia_debug_kernel_17647LLVMPtrI7Float32Li4EE" %__spirv_BuiltInGlobalInvocationId
               OpSource OpenCL_C 200000
               OpName %__spirv_BuiltInGlobalInvocationId "__spirv_BuiltInGlobalInvocationId"
               OpName %top "top"
               OpDecorate %__spirv_BuiltInGlobalInvocationId BuiltIn GlobalInvocationId
               OpDecorate %__spirv_BuiltInGlobalInvocationId Constant
               OpDecorate %__spirv_BuiltInGlobalInvocationId LinkageAttributes "__spirv_BuiltInGlobalInvocationId" Import
      %ulong = OpTypeInt 64 0
      %uchar = OpTypeInt 8 0
%ulong_18446744073709551615 = OpConstant %ulong 18446744073709551615
    %v3ulong = OpTypeVector %ulong 3
%_ptr_Input_v3ulong = OpTypePointer Input %v3ulong
       %void = OpTypeVoid
%_ptr_Generic_uchar = OpTypePointer Generic %uchar
          %9 = OpTypeFunction %void %_ptr_Generic_uchar
      %float = OpTypeFloat 32
%_ptr_Generic_float = OpTypePointer Generic %float
%__spirv_BuiltInGlobalInvocationId = OpVariable %_ptr_Input_v3ulong Input
   %float_42 = OpConstant %float 42
         %10 = OpFunction %void None %9
         %11 = OpFunctionParameter %_ptr_Generic_uchar
        %top = OpLabel
         %13 = OpLoad %v3ulong %__spirv_BuiltInGlobalInvocationId
         %14 = OpCompositeExtract %ulong %13 0
         %16 = OpIAdd %ulong %14 %ulong_18446744073709551615
         %19 = OpBitcast %_ptr_Generic_float %11
         %20 = OpInBoundsPtrAccessChain %_ptr_Generic_float %19 %16
               OpStore %20 %float_42 Aligned 1
               OpReturn
               OpFunctionEnd
```

This crashed the NEO runtime when calling `zeKernelSetArgumentValue`, somewhere in `L0::KernelImp::setArgSampler(unsigned int, unsigned long, void const*)` (a segfault, no abort). That seems wrong, moreso because this pointer isn't texture-related.

jandres742, 2020-07-07:
@maleadt Do you have a backtrace? Also, what do you mean by _generic pointer_?

maleadt, 2020-07-07:
> what do you mean by _generic pointer_?

`OpTypePointer Generic`, coming from `i8 addrspace(4)*` which is the generic address space (in OpenCL semantics).

The backtrace wasn't any more interesting than `zeKernelSetArgumentValue` -> `L0::KernelImp::setArgSampler(unsigned int, unsigned long, void const*)`, but I didn't have a debug build so can't say exactly which operation faulted.

jandres742, 2020-07-10:
@maleadt : could you share the reprocer?

maleadt, 2020-07-14:
Just the simplest thing possible, passing a pointer.

```c++
#include <cassert>
#include <climits>
#include <cstdio>
#include <cstdlib>
#include <cstring>
#include <level_zero/ze_api.h>

#define BUFFERSIZE 1024

#define check(ans)                                                             \
  { do_check((ans), __FILE__, __LINE__); }
void do_check(ze_result_t code, const char *file, int line) {
  if (code != ZE_RESULT_SUCCESS) {
    fprintf(stderr, "Failed: %d at %s %d\n", code, file, line);
    exit(1);
  }
}

int main() {
  check(zeInit(ZE_INIT_FLAG_NONE));

  ze_driver_handle_t driver = nullptr;
  uint32_t count = 1;
  check(zeDriverGet(&count, &driver));

  static ze_device_handle_t device = nullptr;
  check(zeDeviceGet(driver, &count, &device));

  ze_command_queue_desc_t queue_desc{};
  queue_desc.version = ZE_COMMAND_QUEUE_DESC_VERSION_CURRENT;
  ze_command_queue_handle_t command_queue = nullptr;
  check(zeCommandQueueCreate(device, &queue_desc, &command_queue));

  size_t size = 1;

  ze_device_mem_alloc_desc_t device_desc;
  device_desc.ordinal = 0;
  device_desc.flags = ZE_DEVICE_MEM_ALLOC_FLAG_DEFAULT;
  device_desc.version = ZE_DEVICE_MEM_ALLOC_DESC_VERSION_CURRENT;

  // allocate the memory, and make it resident on the device
  void *mem;
  check(zeDriverAllocDeviceMem(driver, &device_desc, size * sizeof(float), 0, device, &mem));
  check(zeDeviceMakeMemoryResident(device, mem, size * sizeof(float)));

  //////////////////////////////////////////////////////////////////////////////////////////

  ze_command_list_desc_t command_list_desc{};
  ze_command_list_handle_t command_list;
  check(zeCommandListCreate(device, &command_list_desc, &command_list));

  uint8_t *buffer = (uint8_t *)malloc(BUFFERSIZE);
  FILE *filp = fopen("kernel.spv", "rb");
  size_t bytes_read = fread(buffer, sizeof(uint8_t), BUFFERSIZE, filp);

  ze_module_constants_t constants = {0, nullptr, nullptr};

  ze_module_desc_t module_desc = {ZE_MODULE_DESC_VERSION_CURRENT,
                                  ZE_MODULE_FORMAT_IL_SPIRV,
                                  bytes_read,
                                  buffer,
                                  "",
                                  &constants};
  ze_module_handle_t module;
  check(zeModuleCreate(device, &module_desc, &module, nullptr));

  ze_kernel_desc_t kernel_desc = {ZE_KERNEL_DESC_VERSION_CURRENT,
                                  ZE_KERNEL_FLAG_NONE, "kernel"};
  ze_kernel_handle_t kernel;
  check(zeKernelCreate(module, &kernel_desc, &kernel));

  check(zeKernelSetArgumentValue(kernel, 0, sizeof(int*), (int *)&mem));

  ze_group_count_t kernel_args = {1, 1, 1};
  check(zeCommandListAppendLaunchKernel(command_list, kernel, &kernel_args,
                                        nullptr, 0, nullptr));

  check(zeCommandListClose(command_list));

  check(zeCommandQueueExecuteCommandLists(command_queue, 1, &command_list,
                                          nullptr));
}
```

Full LLVM module:

```llvm
source_filename = "julia"
target datalayout = "e-i64:64-v16:16-v24:32-v32:32-v48:64-v96:128-v192:256-v256:256-v512:512-v1024:1024"
target triple = "spir64-unknown-unknown"

define dso_local spir_kernel void @kernel(i8 addrspace(4)*) local_unnamed_addr {
top:
  %1 = call i64 @_Z13get_global_idj(i32 0)
  %2 = bitcast i8 addrspace(4)* %0 to float addrspace(4)*
  %3 = getelementptr inbounds float, float addrspace(4)* %2, i64 %1
  store float 4.200000e+01, float addrspace(4)* %3, align 1
  ret void
}

declare i64 @_Z13get_global_idj(i32) local_unnamed_addr

!llvm.module.flags = !{!0, !1}

!0 = !{i32 2, !"Dwarf Version", i32 4}
!1 = !{i32 1, !"Debug Info Version", i32 3}
```

JacekDanecki, 2021-02-11:
Transferring to IGC project.