intel/compute-runtime #336: error: ‘IGFX_DG1’ is not a member of ‘PRODUCT_FAMILY’

jacione, 2020-07-07:
### What Happened:
I followed the instructions given in BUILD.md, and ran into trouble at step 6, when I run `` make -j`nproc` ``. The first time I ran it, it quietly threw this error at 7%:
```
/home/nick/workspace/neo/shared/source/gen12lp/hw_helper_gen12lp.cpp:97:66: error: ‘IGFX_DG1’ is not a member of ‘PRODUCT_FAMILY’
```
but proceeded to continue building. At 37%, it crashed with this output:
```
Makefile:151: recipe for target 'all' failed
make: *** [all] Error 2
```

It looks like there's some data missing, but when I searched the internet for "IGFX_DG1", I came up empty (literally, Google said there's nothing). I assume 'IGFX' stands for something like 'Intel Graphics Effects,' but that's about all I got.

And here's where it gets interesting. Defying the definition of insanity, I ran `` make -j`nproc` `` again, and this time I got to 45%. I tried again and got to 54%, then 60%, then 74%. Each time the same error came up quietly in the middle of the process, and the whole build crashed later. I now get to 74% every time.

### My Question:
I'm not sure if the problem is below the keyboard (e.g. in the hardware or code) or above it (e.g. I'm an idiot or something). Has anyone else encountered this, and is there a proper way to approach this problem?

### Machine Info:
OS: Ubuntu 18.04 (64-bit)
CPU: Core i7-7500U
GPU: HD Graphics 620 (Kaby Lake GT2)

### Other Info:

- In step 2, I installed the prebuilt packages using 

> ```
> sudo add-apt-repository ppa:intel-opencl/intel-opencl
> sudo apt-get update
> sudo apt-get install libigc-dev
> sudo apt-get install libigdgmm-dev
> ```

- Even though it now crashes at the same place each time, I've still noticed that the build processes were slightly different each time, as in the following two (abridged) examples:

> ```
> [ 16%] Built target builtins_Gen9core
> [ 21%] Built target builtins_Gen11lp
> [ 21%] Building CXX object shared/source/CMakeFiles/neo_shared.dir/gen12lp/hw_helper_gen12lp.cpp.o
> [ 21%] Building CXX object shared/source/CMakeFiles/neo_shared.dir/gmm_helper/gmm.cpp.o
> [ 26%] Built target builtins_Gen9lp
> [ 31%] Built target builtins_Gen12LPlp
> [ 32%] Built target scheduler_Gen12LPlp
> [ 32%] Built target scheduler_Gen8core
> [ 32%] Building CXX object shared/source/CMakeFiles/neo_shared.dir/gmm_helper/gmm_helper.cpp.o
> [ 60%] Built target igdrcl_lib_release
> ```
> ```
> [ 16%] Built target builtins_Gen9core
> [ 16%] Building CXX object shared/source/CMakeFiles/neo_shared.dir/gen12lp/hw_helper_gen12lp.cpp.o
> [ 21%] Built target builtins_Gen11lp
> [ 26%] Built target builtins_Gen9lp
> [ 26%] Building CXX object shared/source/CMakeFiles/neo_shared.dir/helpers/timestamp_packet_extra.cpp.o
> [ 27%] Building CXX object shared/source/CMakeFiles/neo_shared.dir/kernel/kernel_descriptor.cpp.o
> [ 55%] Built target igdrcl_lib_release
> ```

JacekDanecki, 2020-07-08:
You need to use [gmmlib 20.2.2](https://github.com/intel/gmmlib/releases/tag/intel-gmmlib-20.2.2) to compile current Neo.
If you want to use gmmlib packages from ppa, you can use repo: https://launchpad.net/~ocl-dev/+archive/ubuntu/intel-opencl, where I've updated gmmlib to 20.2.2.

ppa:intel-opencl/intel-opencl I'll update today to [Neo 20.26.17199](https://github.com/intel/compute-runtime/releases/tag/20.26.17199) containing gmmlib 20.2.2.

jacione, 2020-07-08:
I upgraded to gmmlib 20.2.2.1, but now when I run cmake it says `Gmmlib not found!`.

JacekDanecki, 2020-07-09:
Have you installed devel package? Where did you install gmmlib from? What directory gmmlib was installed to?
Can you provide output from cmake?
What is the output from commands:
```
ldconfig -p | grep gmm
pkg-config --cflags igdgmm

```

jacione, 2020-07-09:
It works! I uninstalled everything, updated to Ubuntu 20.04, and started from scratch.

Here's the play-by-play:
```
sudo apt-get install cmake g++ git pkg-config
```
It said that `libllvm9` was no longer required, and suggested removal.
```
sudo apt autoremove
mkdir workspace
cd workspace/
mkdir build
git clone https://github.com/intel/compute-runtime neo
cd build
cmake -DCMAKE_BUILD_TYPE=Release -DSKIP_UNIT_TESTS=1 ../neo
```
Here it told me (as expected) that I was missing dependencies. I checked the [ppa:intel-opencl/intel-opencl](https://launchpad.net/~intel-opencl/+archive/ubuntu/intel-opencl) repository to make sure that gmmlib was updated to v20.2.2, then continued:
```
sudo add-apt-repository ppa:intel-opencl/intel-opencl
sudo apt-get update
sudo apt install libigc-dev
sudo apt install libigdgmm-dev
```
Here it told me that `libigdgmm-dev` was already installed, and gave me another list of packages that were no longer required: `libclang-cpp10` `libllvmspirvlib10` `libopencl-clang10` but I decided NOT to autoremove them, just in case.
```
cmake -DCMAKE_BUILD_TYPE=Release -DSKIP_UNIT_TESTS=1 ../neo
make -j`nproc`
sudo make install
```
And it looks like it's all installed and ready to go!

JacekDanecki, 2020-07-10:
Closing as issue was resolved.

closed 2020-07-10 by JacekDanecki